dorssel/usbipd-win #21: Contribution Request: Isochronous Support and UI Improvements

NelsonDaniel, 2021-06-01:
Hi, I’m an intern at Microsoft and I am interested in contributing support for isochronous transfer and some UI as part of my summer project. Would such a contribution be welcomed? My team would like to discuss my project with you so you have context on how we intend to use your project. Can you please contact me at t-nelsont@microsoft.com?

dorssel, 2021-06-01:
Hi. All contributions are welcome!

Let me first reply here, so I can put some of my thoughts down for others to read as well.

At first I needed only to support a specific SmartCard reader to be used in a Hyper-V Linux guest. Then, some colleagues of mine wanted support for a USB 3 device. Now, usbipd-win supports all the Bulk USB devices I could get my hands on. And so, a full project takes off...

What is missing (in my opinion), is:

- Isochronous pipes (I have maybe one, a USB microphone) If you have/know/understand the specs and have some devices: great! Go for it!
- Interrupt pipes (I have those, e.g. Yubikey 5, keyboards, mice). This requires asynchronous request/reply, i.e. usbip UNLINK support. I am working on that: not too hard USB-wise.

And about UI. There are may routes you can take, but since you are at MS I am sure it will integrate nicely "Windows style" (perhaps via mmc?). Again: go for it!

Ultimate goal: support all USB devices (except hubs, of course) seamlessly.  And a nice UI to do "windows admin style" management is more than welcome, too.

dorssel, 2021-09-27:
Contributions #27 #30 and #31 added the UI improvements (CLI).
#42 added isochronous support.

Thanks for contributing.

closed 2021-09-27 by dorssel