dorssel/usbipd-win #24: Potential memory corruption bug in DeviceFile.IoControlAsync

MrTrillian, 2021-06-18:
In `DeviceFile.IoControlAsync`, the input and output arrays are pinned using the `lock` statement for the call to `DeviceIoControl`. However, the arrays are then unlocked before the overlapped operation completes. If the GC moves the buffers after the arrays get unlocked, this could result in memory corruption or use-after-free when the win32 code attempts to read or write to the pointers.

To solve this, use an explicit GCHandle of type Pinned, and release it only when the overlapped operation completes.

https://github.com/dorssel/usbipd-win/blob/6db4ea0c74b4b462b9ceac3f31b4579d04fd737a/UsbIpServer/DeviceFile.cs#L66

Found by code inspection

dorssel, 2021-06-19:
The line right above that takes care of that:
https://github.com/dorssel/usbipd-win/blob/6db4ea0c74b4b462b9ceac3f31b4579d04fd737a/UsbIpServer/DeviceFile.cs#L65
`Pack()` pins the objects until the overlapped structure is `Free()`-ed, which happens *after* the overlapped operation completes.
See the note in the remarks at the end of [this documentation](https://docs.microsoft.com/en-us/dotnet/api/system.threading.overlapped.pack?view=net-5.0#System_Threading_Overlapped_Pack_System_Threading_IOCompletionCallback_System_Object_).

closed 2021-06-19 by dorssel

MrTrillian, 2021-06-20:
You are quite right, thanks for pointing out the documentation. That's a surprising .NET API.