dorssel/usbipd-win #53: Service Crashes on starts

scyto, 2021-11-04:
I installed using winget on win11
installed current version

```
Application: usbipd.exe
CoreCLR Version: 5.0.1121.47308
.NET Version: 5.0.11
Description: The process was terminated due to an unhandled exception.
Exception Info: System.Net.Sockets.SocketException (10013): An attempt was made to access a socket in a way forbidden by its access permissions.
   at System.Net.Sockets.Socket.UpdateStatusAfterSocketErrorAndThrowException(SocketError error, String callerName)
   at System.Net.Sockets.Socket.DoBind(EndPoint endPointSnapshot, SocketAddress socketAddress)
   at System.Net.Sockets.Socket.Bind(EndPoint localEP)
   at System.Net.Sockets.TcpListener.Start(Int32 backlog)
   at UsbIpServer.Server.ExecuteAsync(CancellationToken stoppingToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Server.cs:line 59
   at Microsoft.Extensions.Hosting.Internal.Host.StartAsync(CancellationToken cancellationToken)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.Run(IHost host)
   at UsbIpServer.Program.ExecuteServer(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 582
   at UsbIpServer.Program.<>c__DisplayClass19_3.<Main>b__12() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 349
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 555
```

```
Faulting application name: usbipd.exe, version: 1.1.1.143, time stamp: 0x614cbf2c
Faulting module name: KERNELBASE.dll, version: 10.0.22489.1000, time stamp: 0x6d237450
Exception code: 0xe0434352
Fault offset: 0x000000000003f29c
Faulting process id: 0x2fdc
Faulting application start time: 0x01d7d1be5ddd792c
Faulting application path: C:\Program Files\usbipd-win\usbipd.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report Id: 3c4d8cff-cc62-4b39-87be-0609746145e5
Faulting package full name: 
Faulting package-relative application ID: 
```
```
Fault bucket 1761208375485854480, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: usbipd.exe
P2: 1.1.1.143
P3: 614cbf2c
P4: KERNELBASE.dll
P5: 10.0.22489.1000
P6: 6d237450
P7: e0434352
P8: 000000000003f29c
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.06fbc066-e275-44ca-bdfe-7119ed33033e.tmp.mdmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.771e5b3c-2a08-4cd4-8e68-f9e3aef60a6d.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.ccc98cca-1a27-4e49-a700-ede6c0f31f69.tmp.xml
WPR_initiated_DiagTrackMiniLogger_OneTrace_User_Logger_20211011_1 EC_0_inject.etl
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.371f4ae1-3b74-49d0-b02b-e9a6c0572546.tmp.etl
WPR_initiated_DiagTrackMiniLogger_WPR System Collector_inject.etl
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.37268237-3b05-42ec-b4e2-0c1940d04e91.tmp.etl
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.7bbc396f-18a7-4a6a-9999-caf617a12a4b.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.876268b9-ca60-4ca5-9961-4e4d6049f9b5.tmp.txt

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_usbipd.exe_ef47495212974ddfe9a9b8b19fc674f127ed7_cb48227a_3a642198-3cc4-4268-9e95-15687735b76d

Analysis symbol: 
Rechecking for solution: 0
Report Id: 3c4d8cff-cc62-4b39-87be-0609746145e5
Report Status: 268435456
Hashed bucket: 2d0924dc9ddecdf4687111afac46f310
Cab Guid: 0
````

dorssel, 2021-11-04:
Hi, thx for the report!

> An attempt was made to access a socket in a way forbidden by its access permissions.

The server always binds to TCP port 3240 (the default USBIP network port number).
I've not seen this error before, since the service runs with full privileges (LOCAL SYSTEM).
Furthermore, a (standard Windows) firewall rule is created by the installer to allow this.

Could you provide me with the following additional information?
- Could you try running as administrator on the console, as explained in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting ?
- Can you verify a firewall rule exists with the name `usbipd`?
- Are you running with any 3rd party firewall software installed, and if so, could you try turning that off temporarily to see if it makes a difference?

scyto, 2021-11-04:
Hi!
1. (to prove i really did it in an admin command prompt) (the service was already stopped cause it never started)
[image]

2 firewall rule
[image]

3. nope haven't run 3rd party AV in 10 years

a thought... i may have executed the winget command in a terminal as me not as administrator (though i am in the admins group)... will try uninstalling... re-installing....

scyto, 2021-11-04:
nope that didn't fix it.

scyto, 2021-11-04:
turned firewall completely off, that didnt fix

scyto, 2021-11-04:
trace command output...
```
C:\Windows\System32>usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
Unhandled exception. System.Net.Sockets.SocketException (10013): An attempt was made to access a socket in a way forbidden by its access permissions.
   at System.Net.Sockets.Socket.UpdateStatusAfterSocketErrorAndThrowException(SocketError error, String callerName)
   at System.Net.Sockets.Socket.DoBind(EndPoint endPointSnapshot, SocketAddress socketAddress)
   at System.Net.Sockets.Socket.Bind(EndPoint localEP)
   at System.Net.Sockets.TcpListener.Start(Int32 backlog)
   at UsbIpServer.Server.ExecuteAsync(CancellationToken stoppingToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Server.cs:line 59
   at Microsoft.Extensions.Hosting.Internal.Host.StartAsync(CancellationToken cancellationToken)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.Run(IHost host)
   at UsbIpServer.Program.ExecuteServer(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 582
   at UsbIpServer.Program.<>c__DisplayClass19_3.<Main>b__12() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 349
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 555
```

dorssel, 2021-11-04:
Thanks for all the attempts. I know exactly what source line is crashing. However, I have not (yet) been able to reproduce it. Still trying.

It looks like you're using a very recent Windows 11 Insider (22489.1000).
I am able to install and run on that version.
However, unfortunately, in the past we've seen more difficulties with the recent Insider builds.
(please note that Windows 10, or regular Windows 11 will work fine).

All I could find on that specific error is:
https://ardalis.com/attempt-made-to-access-socket/

That article suggests that the port may on the blacklist (although that has never been the case on any of my test machines; in fact, this is the first time I encounter this specific error, it's rather rare). Could you check with:
`netsh interface ipv4 show excludedportrange protocol=tcp`

One more (wild) guess: Have you used/tried any other USBIP related server software on this machine? Maybe that has persistently changed something for port 3240...

Currently, there is an update for Insider to 22494.1000. Perhaps that may give you better results.

dorssel, 2021-11-04:
Great! We found it. Now the big question is: how did all those ports get excluded. Here is my list from the exact same Insider version:

```
Protocol tcp Port Exclusion Ranges

Start Port    End Port
----------    --------
      5357        5357
     49715       49814
     49815       49914
     50000       50059     *

* - Administered port exclusions.
```

I am running:

```
Edition	Windows 11 Pro Insider Preview
Version	Dev
Installed on	‎28-‎10-‎2021
OS build	22489.1000
Experience	Windows Feature Experience Pack 1000.22489.1000.0
```

Could this be Home/Pro related?
Or perhaps you are in a domain, and this is a group policy?

scyto, 2021-11-04:
hmm i found this that implies it can be windows update or installing hyperv that excludes them
https://www.randseq.org/2021/08/stealth-port-exclusion-on-windows-10.html
https://serverfault.com/questions/1045527/how-do-i-find-out-why-certain-ports-are-excluded-and-delete-the-exclusion
I am leary of removing them if hyper-v put them there for a reason... maybe something to ask Craig Lowen?

dorssel, 2021-11-04:
I also have Hyper-V installed (yes ... nested in a VM, that may be different).

scyto, 2021-11-04:
and I am AAD joined i am not classically domain joined so no group policy from AD and no group policy from intune.

scyto, 2021-11-04:
> I also have Hyper-V installed (yes ... nested in a VM, that may be different).

run the netsh command in that nest VM - you should see the exclusions there?

dorssel, 2021-11-04:
I think I found something:

https://superuser.com/questions/1579346/many-excludedportranges-how-to-delete-hyper-v-is-disabled

It looks like your host is NAT-ing for guests, and one of the guests is using port-forwading for exactly that port? Could that be usbipd (maybe even Linux) in a VM?

dorssel, 2021-11-04:
Other people in that same thread are blaming Docker...

scyto, 2021-11-04:
No VMs installed.
WSL2 installed
Docker Desktop installed.......

scyto, 2021-11-04:
according to that thread my dynamic ports is mis-set?

C:\Windows\System32>netsh int ipv4 show dynamicport tcp

Protocol tcp Dynamic Port Range
---------------------------------
Start Port      : 1024
Number of Ports : 64511

dorssel, 2021-11-04:
Good catch!
Solution here: https://github.com/microsoft/WSL/issues/5306
It contains instruction to fix the dynamic port range.

scyto, 2021-11-04:
https://docs.microsoft.com/en-US/troubleshoot/windows-server/networking/default-dynamic-port-range-tcpip-chang

dorssel, 2021-11-04:
Looks like a faulty windows update in the past. One that I never received, hence no problems.
Just deliberately excluded 3240 myself and I have the *exact* same error.

scyto, 2021-11-04:
Cool, i ran all 4 of these from the article above, will report back after the build to build upgrade

```
The following are sample commands:

netsh int ipv4 set dynamicport tcp start=10000 num=1000
netsh int ipv4 set dynamicport udp start=10000 num=1000
netsh int ipv6 set dynamicport tcp start=10000 num=1000
netsh int ipv6 set dynamicport udp start=10000 num=1000

```

dorssel, 2021-11-04:
You may want to improve on that. My clean Windows says:

```
Protocol tcp Dynamic Port Range
---------------------------------
Start Port      : 49152
Number of Ports : 16384
```
for all 4.

Those are the RFC recommended ones too (https://datatracker.ietf.org/doc/html/rfc6056).

scyto, 2021-11-04:
thanks, wonder why the writer of the tech doc chose arbitrary ranges - can i be bothered to open an issue on the tech doc... lol

scyto, 2021-11-04:
for anyone that wants easy cut and paste and arrives here
```
netsh int ipv4 set dynamicport tcp start=49152 num=16384
netsh int ipv4 set dynamicport udp start=49152 num=16384
netsh int ipv6 set dynamicport tcp start=49152 num=16384
netsh int ipv6 set dynamicport udp start=49152 num=16384
```

scyto, 2021-11-04:
I think we can declare this issue resolved

```
STUDY\C:\Users\alex> usbipd server
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\alex
```

closed 2021-11-04 by scyto

scyto, 2021-11-04:
one additional note, the service still wouldn't start
so i did uninstall
reinstall then failed
noted service was not removed but set to disabled / marked for deletion
rebooted
reinstall then succeeded

dorssel, 2021-11-04:
Interesting ... thx for the info! Will try to polish uninstall/reinstall behavior some more. Always tricky with driver/service installs...

scyto, 2021-11-05:
thanks, will do more testing once:
1. I figure out how to get apt on debian from using IPv6 when it shouldn't  (i did figure it out)
2. I figure out why apt on IPv4 isn't working yet pings etc to externall IPv4 work just fine
3. how on earth i disable IPv6 when it is no longer possible to get to the old network adapter control panel (this works for windows Disable-NetAdapterBinding -Name "*" -ComponentID ms_tcpip6 but WSL still thinks its IPv6 - awesome)