dorssel/usbipd-win #54: AddDescriptor crashes adding duplicate key

shanselman, 2021-11-04:
When trying to run `./usbipd wsl list` I get:

```
Unhandled exception. System.AggregateException: One or more errors occurred. (An item with the same key has already been added. Key: [1, UsbIpServer.UsbConfigurationDescriptors+UsbConfiguration])
 ---> System.ArgumentException: An item with the same key has already been added. Key: [1, UsbIpServer.UsbConfigurationDescriptors+UsbConfiguration]
   at System.Collections.Generic.TreeSet`1.AddIfNotPresent(T item)
   at System.Collections.Generic.SortedSet`1.Add(T item)
   at System.Collections.Generic.SortedDictionary`2.Add(TKey key, TValue value)
   at UsbIpServer.UsbConfigurationDescriptors.AddDescriptor(ReadOnlySpan`1 descriptor) in C:\github\usbipd-win\UsbIpServer\UsbConfigurationDescriptors.cs:line 98
   at UsbIpServer.ExportedDevice.GetConfigurationDescriptor(DeviceFile hub, UInt16 connectionIndex, Byte numConfigurations) in C:\github\usbipd-win\UsbIpServer\ExportedDevice.cs:line 130
   at UsbIpServer.ExportedDevice.GetDevice(SafeDeviceInfoSetHandle deviceInfoSet, SP_DEVINFO_DATA devInfoData, CancellationToken cancellationToken) in C:\github\usbipd-win\UsbIpServer\ExportedDevice.cs:line 202
   at UsbIpServer.ExportedDevice.GetAll(CancellationToken cancellationToken) in C:\github\usbipd-win\UsbIpServer\ExportedDevice.cs:line 239
   at UsbIpServer.Program.<>c.<<Main>b__19_18>d.MoveNext() in C:\github\usbipd-win\UsbIpServer\Program.cs:line 387
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.<>c__DisplayClass79_0.<OnExecute>b__0()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
   at UsbIpServer.Program.Main(String[] args) in C:\github\usbipd-win\UsbIpServer\Program.cs:line 555
```

So I added some Console.WriteLine logging and get

```
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Unhandled exception. System.AggregateException: One or more errors occurred. (An item with the same key has already been added. Key: [1, UsbIpServer.UsbConfigurationDescriptors+UsbConfiguration])
```

But I'm confused. Why didn't the second "adding key: 1" fail, and what's config.bConfigurationValue expected to be?

It seems like my machine has some number of devices that you're not expecting to "be the same" when loaded into the hashtable. Two mice? Two webcams? Two identical brand Anker USB Hubs? Yes to all of these things.

Any ideas on the best way to debug?

dorssel, 2021-11-04:
Your seeing many "Key: 1"s, namely one entry for each device. The bug, however, seems to be a device with more than one Configuration descriptor. Those are very rare, so it may very well be I overlooked something there.

If you are familiar with`usbview.exe` (part of Windows SDK), I would love to see the device descriptors (all of them) of that particular device.

shanselman, 2021-11-05:
[ScottUSBViewAll.txt](https://github.com/dorssel/usbipd-win/files/7479767/ScottUSBViewAll.txt)
Does this help?

shanselman, 2021-11-05:
I added more logging...seems like it's my USB Sound board from Mackie Designs, a ProFX6?

```
 USB\VID_051D&PID_0002\3B1613X15938__ - 1309,2,144
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_045E&PID_082C\604863194321 - 1118,2092,256
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0FD9&PID_0066\0002869971000 - 4057,102,0
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_046D&PID_085E\61309117 - 1133,2142,23
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0B05&PID_1867\00000000001A - 2821,6247,512
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0B05&PID_185C\00E04C000001 - 2821,6236,272
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_046D&PID_C52B\8&197761EB&0&1 - 1133,50475,4624
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0B0E&PID_2003\8&197761EB&0&2 - 2830,8195,769
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_1BCF&PID_0005\7&7F4B74D&0&3 - 7119,5,20
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_1058&PID_25F6\575542533238343530323432 - 4184,9718,4105
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_04D8&PID_F848\8&35414E0F&0&4 - 1240,63560,2
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0A73&PID_0023\6&14597C0D&0&1 - 2675,35,1778
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
Unhandled exception. System.AggregateException: One or more errors occurred. (An item with the same key has already been added. Key: [1, UsbIpServer.UsbConfigurationDescriptors+UsbConfiguration])
```
[image]

shanselman, 2021-11-05:
Confirmed, when I turn off the Mackie, I can get a list of USB devices:
```
 .\usbipd wsl list
 USB\VID_051D&PID_0002\3B1613X15938__ - 1309,2,144
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_045E&PID_082C\604863194321 - 1118,2092,256
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0FD9&PID_0066\0002869971000 - 4057,102,0
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_046D&PID_085E\61309117 - 1133,2142,23
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0B05&PID_1867\00000000001A - 2821,6247,512
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0B05&PID_185C\00E04C000001 - 2821,6236,272
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_046D&PID_C52B\8&197761EB&0&1 - 1133,50475,4624
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0B0E&PID_2003\8&197761EB&0&2 - 2830,8195,769
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_1BCF&PID_0005\7&7F4B74D&0&3 - 7119,5,20
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_1058&PID_25F6\575542533238343530323432 - 4184,9718,4105
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_04D8&PID_F848\8&35414E0F&0&4 - 1240,63560,2
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
 USB\VID_0FD9&PID_0060\AL44H1C10981 - 4057,96,256
Adding Key: 1 with Windows.Win32.Devices.Usb.USB_CONFIGURATION_DESCRIPTOR
BUSID  DEVICE                                                        STATE
1-13   Realtek Bluetooth Adapter                                     Not attached
1-14   USB Input Device                                              Not attached
1-20   USB Audio Device, USB Input Device, USB Video Device          Not attached
9-2    American Power Conversion USB UPS                             Not attached
10-2   Logitech BRIO, USB Audio Device                               Not attached
11-3   USB Mass Storage Device                                       Not attached
13-2   USB Input Device                                              Not attached
13-3   USB Input Device                                              Not attached
15-1   Logitech USB Input Device, Logitech Unifying USB receiver...  Not attached
15-2   USB Audio Device, USB Input Device                            Not attached
15-3   USB Input Device                                              Not attached
16-4   USB Input Device                                              Not attached
```

dorssel, 2021-11-05:
Just like I suspected:

```
bNumConfigurations:                0x02
*!*CAUTION:    Most host controllers will only work with one configuration per speed
```

This device has more than 1 configuration, which (as stated before) is very rare.
In this case it even looks more special than that. From the USB specification:

```
Two descriptors allow a high-speed capable device to report configuration information about the other
operating speed. The two descriptors are: the (other_speed) device_qualifier descriptor and the
other_speed_configuration descriptor. These two descriptors are retrieved by the host by using the
GetDescriptor request with the corresponding descriptor type values.

Note: These descriptors are not retrieved unless the host explicitly issues the corresponding GetDescriptor
requests. If these two requests are not issued, the device would simply appear to be a single speed device.
```

I certainly did not account for that. The CAUTION above certainly indicates that this is indeed a multi-speed device, but will report only 1 of its two available configurations unless asked in this special way (which usbipd-win does *not* do).

Without the device available to experiment with, this is not easily fixed. Also because there are a lot of other WARNINGS in the descriptor list. What I *will* do is make sure that such failures to decode will simply filter out the device from the list altogether. But I don't expect that usbipd-win can support this device in the near future (unless you are willing to send it to me).

shanselman, 2021-11-05:
Sounds like a great deal @dorssel and I thank you for your patience and help! Yes, if it was either filtered out or filtered out with a warning that'd be cool. I am more than willing to apply any PRs/patches and test locally if that helps you!

Would be happy to send it to you but it's a pretty obscure high end audio mixer so I suspect it's not worth the hassle.  :)

Thank you for all your great work!

cho45, 2021-11-05:
I have same issue for following device.

(from usbview.exe)
```
       ---===>Device Information<===---
English product name: "xCORE USB Audio 2.0"

ConnectionStatus:
Current Config Value:              0x01  -> Device Bus Speed: High (is not SuperSpeed or higher capable)
Device Address:                    0x0A
Open Pipes:                           2

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0200
bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x20B1 = XMOS Ltd.
idProduct:                       0x000A
bcdDevice:                       0x06C5
iManufacturer:                     0x01
     English (United States)  "XMOS"
iProduct:                          0x03
     English (United States)  "xCORE USB Audio 2.0"
iSerialNumber:                     0x00
bNumConfigurations:                0x02
*!*CAUTION:    Most host controllers will only work with one configuration per speed
```

ebinans, 2021-11-05:
Apparently not that rare and looks like Audio interfaces are the main offenders :)

[BEHRINGER_UMC_202HD_192k.txt](https://github.com/dorssel/usbipd-win/files/7481908/BEHRINGER_UMC_202HD_192k.txt)

ebinans, 2021-11-05:
Looks good :)

```
WARNING: bConfigurationValue 1 is duplicated. ignored.
Present:
BUSID  DEVICE                                                        STATE
1-5    Intel(R) Wireless Bluetooth(R)                                Not shared
1-6    Integrated Camera                                             Not shared
1-8    Synaptics WBDI                                                Not shared
2-2    DisplayLink USB Audio Adapter, DisplayLink USB Device         Not shared
5-4    Remote NDIS Compatible Device                                 Not shared
7-1                                                                  Not shared
7-2    USB Input Device                                              Not shared
7-3    STMicroelectronics STLink Virtual COM Port, STMicroelectr...  Not shared
8-1    Billboard Device, USB Input Device                            Not shared
8-4    USB Input Device                                              Not shared

Persisted:
GUID                                    BUSID  DEVICE

```

PR #61 (2021-11-06): ignore duplicated bConfigurationValue
Fix #54 

I am not sure if this is an accurate change. However, I believe it is better than the current situation where it crashes.
closed 2021-11-06 by dorssel