dorssel/usbipd-win #55: winget installer reboots without prompting...

ethanhs, 2021-11-04:
Hi, I am really excited to see this available for the WSL!

I ran the winget installer to install things on my desktop a bit earlier, and was somewhat surprised when my desktop immediately rebooted itself without prompting me. Fortunately I didn't lose any work I had open, and I know drivers require a reboot before working, but I generally expect to be prompted to to reboot now or later.

I would ask/suggest that you either: 
1. prompt the user to reboot now or later or
2. very clearly document this behavior in the README.

Thanks again for the awesome software!

dorssel, 2021-11-04:
Thanks for the report.

Unfortunately, this is outside the control of the installer.
It is controlled by `winget`, which by default uses silent install (and honors reboot).
This probably can be avoided by using `winget install --interactive`.

Winget issue: https://github.com/microsoft/winget-cli/issues/229

ethanhs, 2021-11-04:
Ah dang ok, thanks for pointing me to that, perhaps then a warning in the README would be helpful. Kinda surprised that issue is still open over a year later, but I guess figuring out how to fix that is tricky.

scyto, 2021-11-05:
never rebooted here, did you have some pending updates or something?

commit 68d1a88d (2021-11-06): Update README.md

Closes #55
closed 2021-11-06 by dorssel