dorssel/usbipd-win #57: Do I need to install these dependecies?

scyto, 2021-11-05:
solved my wsl2 network issues (aka uninstall hyper-v, netst reset, etc - there is a bug MS still have to find..)

Got this far do i need to install all of these?

alexb@STUDY:/mnt/c/WINDOWS/system32$ usbipd wsl list
WARNING: usbipd not found for kernel 5.10.60.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.10.60.1-microsoft-standard-WSL2
    linux-cloud-tools-5.10.60.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2

scyto, 2021-11-05:
meh. instructions didn't say to use sudo
doh

closed 2021-11-05 by scyto

benpicco, 2021-12-01:
Where did you find the `linux-tools-standard-WSL2` pacakege? The Ubuntu repository only contains the packages for the kernel shipped by Ubuntu, not the WSL2 kernel. I tried adding `packages.microsoft.com` but that does not contain it either.

wellcoms-cartier, 2022-01-21:
Please try these commands again but under a Windows CMD in Administrator Mode....