dorssel/usbipd-win #58: Does usbipd-win work with WSL2 preview from store?

scyto, 2021-11-05:
I don't have WSL installed via add/remove features.  
It is installed via the windows store (preview verison of wsl2)

Windows sees WSL as installed
```
STUDY\C:\Users\alex> wsl -l
Windows Subsystem for Linux Distributions:
Ubuntu (Default)
```

but usbipd doesn't
```
STUDY\C:\Users\alex> usbipd wsl attach --busid=6-2
usbipd: error: Windows Subsystem for Linux version 2 is not available. See https://aka.ms/installwsl.
```
(note if it matters wsl was removed from add/remove features AFTER usbipd was installed....)

dorssel, 2021-11-05:
Good find.
Yes and no.

usbipd-win checks for WSL availability by checking if the Windows Subsystem for Linux feature is installed. In your case it is not. This design is from before the Store version became available. I will fix this in the next release...

There is good news, however. As a workaround you can either:

1. *also* install the feature. If you have both, the store version takes precedence, so you will still get the latest/greatest. In this case you just install the feature to convince usbipd-win that you have WSL, even though you will actually be using the store version.
2. do not use the `usbipd wsl` subcommand, but instead use `usbipd bind` and within WSL use `usbip attach`. In this case the check won't happen (and therefore won't fail).

scyto, 2021-11-05:
Thanks using bind on the host works 

On my host we seem to have it listening ok
```
STUDY\C:\Users\alex> netstat -anb | grep 3240
  TCP    0.0.0.0:3240           0.0.0.0:0              LISTENING
  TCP    [::]:3240              [::]:0                 LISTENING
```

we have the device attached (shared)
```
STUDY\C:\Users\alex> usbipd list
Present:
BUSID  DEVICE                                                        STATE
1-1    USB Input Device                                              Not shared
1-5    USB Input Device                                              Not shared
1-6    Billboard Device                                              Not shared
1-11   USB Input Device                                              Not shared
1-14   Intel(R) Wireless Bluetooth(R)                                Not shared
2-1    USB Video Device                                              Not shared
4-1    UM-ONE                                                        Not shared
6-2    Bulk-In, Interface                                            Shared
6-3    Corsair Hydro Series 7289 USB Device                          Not shared
9-1    USB Input Device                                              Not shared
9-2    USB Audio Device, USB Input Device, USB Serial Device         Not shared
9-3    USB Audio Device, USB Input Device, USB Serial Device         Not shared
9-5    SMSC-Microchip WinUSB USB5734 Device                          Not shared
13-3   Logitech BRIO, USB Audio Device                               Not shared

Persisted:
GUID                                    BUSID  DEVICE
```

on the wsl ubuntu instance i get this
```
alexb@STUDY:/mnt/c/WINDOWS/system32$ sudo usbip list -r=192.168.1.137
usbip: error: could not connect to =192.168.1.137:3240: Name or service not known
```
i seem to recall there are issues with WSL connecting to the host from some past issue, but not sure if thats valid any more
i also let it try 127.0.0.1 and that didn't work either

dorssel, 2021-11-05:
> new connection from 192.168.1.137
This tells me that that is the IP address of *WSL* (the server reports the IP address of the client: WSL in this case).

> sudo usbip list -r 192.168.1.137
Here you are asking WSL to connect to itself! You should instead be using the IP address of the host (on the WSL virtual switch). Probably 192.168.1.1?

scyto, 2021-11-05:
random FYI - on todays win11 build running both wsl2 and wsl2 preview results is a seriously borked wsl

fix is to uninstall wsl from features
run wsl --unregister <distro>
re-install distro from store (note doing this from command line may not work)
etc

dorssel, 2021-11-05:
All usbipd-win development is from *before* the store version of WSL2. I am not really surprised that does not work well yet. I'll be working on that...

scyto, 2021-11-05:
>

Not at all.

192.168.1.137 is the IP of my windows host OS.
192.168.239.143 is the IP of my WSL guest.

both are running on the same machine - i suspect this is one of those cases where the traffic never goes on to the wire and is looping back in the stack, as such i think usbipd cant tell what actual source address the traffic has because it is 0.0.0.0?

scyto, 2021-11-05:
> All usbipd-win development is from _before_ the store version of WSL2. I am not really surprised that does not work well yet. I'll be working on that...

oh absolutely, no expectation here, was just warning you about the funkiness incase you install both :-)

dorssel, 2021-11-05:
Oh wow, I never knew WSL2 did such network trickery.

I only have WSL actually installed on Windows 10, where the WSL IP range is 172.27.32/20.
And on Windows 10 `usbipd` does report the correct client (WSL) IP address, and `usbip list --remote` just works.

Must be something special about Windows 11, or Insider, or the store version.

The only thing I can think of now, is that these IP-address-shenanigans may cause the firewall to believe the connection is not from a local subnet. You could try opening up the `usbipd` firewall rule for everyone (to test).

scyto, 2021-11-05:
yeah WSL and hyper-v networking have always been 'dodgy' IMHO

absoutely could be something to do with 11 / insider or store - i will revert to WSL in the OS and dump store version just to remove 1 variable

i already tried turning off the firewall completely....

dorssel, 2021-11-05:
Oh, BTW, `usbipd wsl attach` is actually *identical* to `usbipd bind` followed by `usbip attach` on the client side. Including figuring out the host WSL-switch network address and passing that to `usbip`. `usbipd wsl` is really like macro for those two commands.

scyto, 2021-11-05:
Thanks, good to know.

I think this is a Windows 11 or Windows 11 Insider thing as same issue on the in-built wsl.

```
STUDY\C:\Users\alex> usbipd wsl attach --busid=6-2
[sudo] password for alexb:
usbip: error: Attach Request for 6-2 failed - Request Completed Successfully

usbipd: error: Failed to attach device with ID '6-2'.
```

dorssel, 2021-11-05:
I have another machine with regular Windows 11. Will try WSL on that next. But signing off for now, it's 3:24 AM here...

scyto, 2021-11-05:
good night, i have two hyper-v servers too, might try this using hyper-v at the weekend and give WSL a pause, thanks for being so responsive, nice project!

PR #71 (2021-11-08): Support for WSL from Microsoft Store
Fixes #58
closed 2021-11-08 by dorssel