dorssel/usbipd-win #59: USB Serial device does not present in /dev

adarshmk, 2021-11-05:
'lsusb' lists the device:

```
❯ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

But I don't see a _/dev/ttyUSB*_ or a _/dev/ttyACM*_

dorssel, 2021-11-05:
The `attach` worked OK, so now it is up to the Linux side to query the device properties and start using the correct driver. It may be that your device is somehow incompatible, which should show in the linux logs. Unfortunately, I don't know how to get the WSL syslog; mine is always empty. Maybe `sudo apt install rsyslog` (a guess).

If that fails, maybe you could install a full distro installation in a Hyper-V VM, that will certainly allow you to get the linux USBIP side of logging. Once we get it working in the full VM, you can switch back to WSL.

adarshmk, 2021-11-05:
I detached and attached the same device and captured the syslog:

```
Nov  5 07:23:53 AdiPC rsyslogd: [origin software="rsyslogd" swVersion="8.2001.0" x-pid="1487" x-info="https://www.rsyslog.com"] start
Nov  5 07:24:54 AdiPC kernel: [ 5094.175462] vhci_hcd: connection closed
Nov  5 07:24:54 AdiPC kernel: [ 5094.175537] vhci_hcd: stop threads
Nov  5 07:24:54 AdiPC kernel: [ 5094.175539] vhci_hcd: release socket
Nov  5 07:24:54 AdiPC kernel: [ 5094.175542] vhci_hcd: disconnect device
Nov  5 07:24:54 AdiPC kernel: [ 5094.175602] usb 1-1: USB disconnect, device number 2
Nov  5 07:25:10 AdiPC kernel: [ 5109.891897] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
Nov  5 07:25:10 AdiPC kernel: [ 5109.891899] vhci_hcd vhci_hcd.0: devid(131076) speed(2) speed_str(full-speed)
Nov  5 07:25:10 AdiPC kernel: [ 5109.891915] vhci_hcd vhci_hcd.0: Device attached
Nov  5 07:25:10 AdiPC kernel: [ 5110.177435] vhci_hcd: vhci_device speed not set
Nov  5 07:25:10 AdiPC kernel: [ 5110.247437] usb 1-1: new full-speed USB device number 3 using vhci_hcd
Nov  5 07:25:10 AdiPC kernel: [ 5110.327432] vhci_hcd: vhci_device speed not set
Nov  5 07:25:11 AdiPC kernel: [ 5110.397436] usb 1-1: SetAddress Request (3) to port 0
Nov  5 07:25:11 AdiPC kernel: [ 5110.447429] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
Nov  5 07:25:11 AdiPC kernel: [ 5110.447430] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov  5 07:25:11 AdiPC kernel: [ 5110.447431] usb 1-1: Product: CP2102 USB to UART Bridge Controller
Nov  5 07:25:11 AdiPC kernel: [ 5110.447431] usb 1-1: Manufacturer: Silicon Labs
Nov  5 07:25:11 AdiPC kernel: [ 5110.447432] usb 1-1: SerialNumber: 0001
```

I don't see any error messages, but I also don't see any `udev` messages either. I have tried other USB UART adapters as well (based on the CH340 chip) and same results there as well. I will try it on a full Hyper-V VM as you suggested and see if I can get it working there.

dorssel, 2021-11-05:
Yep, everything looks good, apart from the fact that no driver is actually claiming the device. The fact that there are no errors so far is a very good sign. About 10-20 USB messages must have already passed back and forth for this information to show up. Once you get the driver to load, I have good hopes for this one...

Good luck! And please report back on either success or failure.

adarshmk, 2021-11-05:
Some progress, I have a Black Magic Probe which I attached, and it enumerates fine on ttyACM* using the `cdc_acm` driver:

```
[  909.113922] usb 1-1: New USB device found, idVendor=1d50, idProduct=6018, bcdDevice= 1.00
[  909.113923] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  909.113924] usb 1-1: Product: Black Magic Probe (SWLINK) v1.7.1-259-ge58b7d6
[  909.113925] usb 1-1: Manufacturer: Black Sphere Technologies
[  909.113925] usb 1-1: SerialNumber: 8DB277B2
[  909.137939] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  909.143926] cdc_acm 1-1:1.2: ttyACM1: USB ACM device
```

The other devices (CH340/341, CP210x and Prolific) are still not working. Seems like some config is still missing in the kernel.

adarshmk, 2021-11-05:
It turns out, my recompiled kernel was never being used because the `.wslconfig` file was named incorrectly 😖 

After fixing my stupid mistake, all the USB serial devices worked as expected.

For anyone else who wants to be able to use these USB devices, you will need to compile your kernel with the drivers enabled in the kernel config. The stock kernel from Microsoft does not have them enabled by default (yet!).

I will close this issue. Thanks for the support @dorssel

closed 2021-11-05 by adarshmk