dorssel/usbipd-win #60: Support for Android Debug Bridge (adb)

ln-12, 2021-11-05:
Hey, awesome project! I am using Windows 11 and WSL2 and tried to play around with [ADB](https://developer.android.com/studio/command-line/adb). Unfortunately, it does not seem to work at the moment. My current output is the following:
```
user@desktop:/mnt/c/Users/user$ lsusb
   Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
   Bus 001 Device 002: ID 2a70:4ee7 OnePlus AC2003
   Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
user@deskzop:/mnt/c/Users/user$ adb devices -l
   List of devices attached
   * daemon not running; starting now at tcp:5037
   * daemon started successfully

user@deskzop:/mnt/c/Users/user$
```
As you can see, my phone is available in WSL, but ADB does not see it. Do you have any idea if it might be a configuration issue or is it simply not currently supported?

cjshearer, 2021-11-05:
I'll try this myself in a bit to see if it works for me. I've been handling this use case by running ADB in windows with `socat` to forward the USB traffic to wsl for a while, but I'd love for a more seamless experience.

cjshearer, 2021-11-05:
Got it working. I had the wrong kernel version and needed to setup the udev rules:

1. Follow the instructions [WSL support wiki](https://github.com/dorssel/usbipd-win/wiki/WSL-support):
    - before attaching your android device to WSL, be sure it is in USB debugging mode
    - if your kernel version is older than the one mentioned in the WSL support wiki, you can update by first shutting down WSL with `wsl.exe --shutdown` and update it with `wsl.exe --update`
    - don't worry about setting the udev rules just yet, we'll get to that in step 2

Your device should now be visible in WSL and should be in debug mode before continuing:
```
user@desktop:~/$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 18d1:4ee2 Google Inc. Nexus Device (debug)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

2 . Follow [these instructions ](https://android.stackexchange.com/a/144967/237358) to configure your udev rules for the device.

If I'm not missing anything, ADB should now be able to see your device:

```
user@desktop:~/$ adb devices
List of devices attached
09021FDD4003NW  device
```

ln-12, 2021-11-07:
Thanks for your response, @cjshearer. Unfortunately, it won't work for me just yet. I set up everything like this:
```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2a70:4ee7 OnePlus AC2003
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ cat /etc/udev/rules.d/51-android.rules
SUBSYSTEM=="usb", ATTR{idVendor}=="2a70", ATTR{idProduct}=="4ee7", MODE="0666", GROUP="plugdev", SYMLINK+="android%n"

$ cat ~/.android/adb_usb.ini
0x2a70

$ groups
user adm dialout cdrom floppy sudo audio dip video plugdev netdev libvirt docker

$ ls -hasl /dev
total 4.0K
   0 drwxr-xr-x  9 root root     3.1K Nov  7 13:40 .
4.0K drwxr-xr-x 19 root root     4.0K Nov  7 13:40 ..
   0 crw-rw-rw-  1 root root  10, 235 Nov  7 13:40 autofs
   0 drwxr-xr-x  2 root root       40 Nov  7 13:40 block
   0 drwxr-xr-x  2 root root      140 Nov  7 13:40 bsg
   0 crw-rw-rw-  1 root root  10, 234 Nov  7 13:40 btrfs-control
   0 drwxr-xr-x  3 root root       60 Nov  7 13:40 bus
   0 crw-rw-rw-  1 root root   5,   1 Nov  7 13:40 console
   ...
```

As you can see, there is no `/dev/androidN` device, so it's no surprise that `adb` can't find it. Do you have any idea what I am missing now?

cjshearer, 2021-11-07:
I tried to connect my device again this morning and ran into the same problem you did. I spent a couple hours trying to find the minimal set of instructions to connect my device (after having already followed the [WSL Support instructions](https://github.com/dorssel/usbipd-win/wiki/WSL-support)). Also, I switched to using these [udev rules for android](https://github.com/M0Rf30/android-udev-rules), although it did work for me before then. 

I find I have to do the following for every boot of WSL (also, after every change to the udev rules):
1. Restart udev:
   1. `sudo service udev restart`
   2. `sudo udevadm control --reload`
2. Attach, or reattach the device to WSL (file transfer mode, usb debugging)

After which, the device is visible to ADB. Having done this once, I can unplug/plug my device back in and (after reattaching to WSL) it is still visible to ADB, so long as I don't restart WSL.

ln-12, 2021-11-12:
@cjshearer Thanks for sharing! It works for me now too, althoug sometimes I have to run ` usbipd wsl attach --busid <ID>` multiple times to actually attach the device.

Now the question is who is responsible for making the process a bit more user friendly. Is it related to usbipd, WSL or ADB?

mangkoran, 2021-11-17:
Hello. I already follow the instruction to add `udev` rules, but `adb devices` still show nothing. My phone showed in `lsusb` tho:
```
Bus 001 Device 003: ID 18d1:4ee2 Google Inc. Nexus/Pixel Device (MTP + debug)
```

mangkoran, 2021-11-17:
After further investigation, turns out I need to run `adb start-server` as `sudo`. Only after that I can see my device under `adb devices`. So it might be `udev` rules problem for me. Aside from that, I can say that I have successfully connected my device to WSL2 `adb`. Good job devs!

mangkoran, 2021-11-17:
I don't know if this is under the scope of this project, but I put it here otherwise: My device only detected under `adb devices` only if I run `adb start-server` as `sudo`
```
~
> adb devices
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached

~
> adb kill-server

~
> sudo adb devices
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached
5950907f        device

~
> adb devices
List of devices attached
5950907f        device
```

dorssel, 2022-02-11:
WSL documentation has been updated with `udev` details. Closing.

closed 2022-02-11 by dorssel