dorssel/usbipd-win #62: WSL2 usbip - secure_path update not taken

athario, 2021-11-05:
Updated secure_path with sudo visudo but can't run usbip without the full filepath
I have tried wsl --shutdown and restarted the instance.

Update present in /etc/sudoers:

$ sudo cat /etc/sudoers
(file is stock other than this update)
...
Defaults        secure_path="/usr/lib/linux-tools/5.4.0-77-generic:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin"
...

$ usbip
WARNING: usbip not found for kernel 5.10.60.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.10.60.1-microsoft-standard-WSL2
    linux-cloud-tools-5.10.60.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2

$ /usr/lib/linux-tools/5.4.0-77-generic/usbip
usage: usbip [--debug] [--log] [--tcp-port PORT] [version]
             [help] <command> <args>

  attach     Attach a remote USB device
  detach     Detach a remote USB device
  list       List exportable or local USB devices
  bind       Bind device to usbip-host.ko
  unbind     Unbind device from usbip-host.ko
  port       Show imported USB devices

$ uname -r
5.10.60.1-microsoft-standard-WSL2

I feel like I am missing something ridiculously simple!  Any advice would be greatly appreciated!

benmcmorran, 2021-11-05:
Updating `secure_path` will only put `usbip` on your path as root, so `sudo usbip` should work. For non-root access you'll need to either alias `usbip` or edit `$PATH` in your `.bashrc`. I'll update the wiki article to clarify.

athario, 2021-11-05:
Thank you @benmcmorran.

Stange one, after running sudo usbip on the wsl, which worked, the PowerShell sprang into life and allowed me to mount a device!  Before that it was returning the same error.

benmcmorran, 2021-11-05:
Glad you got it working! I added a few lines on [the wiki page](https://github.com/dorssel/usbipd-win/wiki/WSL-support#wsl-setup) about non-root access and will close this issue.

closed 2021-11-05 by benmcmorran