dorssel/usbipd-win #63: usbip: error: Attach Request for [bus-id-here] failed - Request Failed

calipzo, 2021-11-05:
When I try to connect the device, it says the following:

`PS C:\windows\system32> usbipd wsl attach --busid=6-1
[sudo] password for calipso:
usbip: error: Attach Request for 6-1 failed - Request Failed
usbipd: error: Failed to attach device with ID '6-1'.
PS C:\windows\system32>`
[image]

With extended error output the messages are as follows:
`dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\windows\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[0]
      new connection from 172.21.14.67
dbug: UsbIpServer.ConnectedClient[0]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client: System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 144
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 155
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 161
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 158
dbug: UsbIpServer.Server[0]
      connection closed`

[image]

Any ideas?

dorssel, 2021-11-05:
This is unfortunately the part where usbipd-win has the least amount of control. At this point, the VBoxUSBMon driver is supposed to tell the USB-hub that  the device has "changed". It does this by overriding the device ID, such that Windows believes a different device is now plugged in, upon which it will use the VBoxUSB stub driver.

VBoxUSBMon does this by simulating an unplug/replug sequence. I've seen exactly this error before with integrated devices that are marked as "non-pluggable". For that case, usbipd-win throws in an additional RESET_PORT to the hub.

Apparently, your device is not being correctly filtered by VBoxUSBMon, hence Windows will not substitute the driver. And at that point usbipd-win reports: device not found. This means it is going wrong at the "hub" level, not at the "device" level.

You can try any of these things:

- connect it to a different port, preferably even a different port version (i.e. use a USB 2.0 port instead of USB 3.0 or vice versa)
- connect it to a different hub, possibly even add an additional hub between the current port and the device
- try disabling power saving features for the device (via the energy settings in the device properties page of the device manager)

At the time of "attaching" a device with usbipd-win, you will hear a Windows audio signal (the same you here when you unplug a device). Does that happen for you? It will give some more information about the stage that goes wrong... You may also open device manager and select the device. When you try to "attach", does the device (briefly) disappear, or does nothing happen?

pouyaemami, 2021-11-10:
Looks like I am having the same issue. For me the device does briefly disappear and I do get a Windows notification for a newly connected device. I also tried disabling the power saving feature for all my USB hubs and tried a couple of different ports on my machine just to be sure (I even tried different cables) but none of that seemed to help. Is there anything else that can be done?

dorssel, 2021-11-13:
> Is there anything else that can be done?

`usbipd-win` relies on the VBoxUSBMon driver to capture the device from the Windows host and replace the host driver with a stub driver. Unfortunately, there is no way (that I know of) to diagnose why this fails for some devices.

If you want to pursue this further, you could try if a full installation of VirtualBox is in fact capable of forwarding the device to its guests. If that works it is an indication that it should be fixable in `usbipd-win` too. However, these days VirtualBox does not play very well with Hyper-V and WSL. Experimenting with this will be time consuming. I cannot help there, since I don't have any such device and cannot reproduce the problem.

Until then, we'll have to mark this device as incompatible. Can you provide make/model and the Vendor/Product ID (0123:4567)?

Eric-N-Be, 2021-11-14:
hi,
i have the same problem with my usb serial converter to read data into smart meter

powershell running in admin mode to execute usbipd command
windows 10 (10.0.19043)
WSL 2 : 5.10.60.1-microsoft-standard-WSL2
Debian GNU/Linux 11 (bullseye)
usbipd-win 1.1.1

`usbipd wsl list  
BUSID  DEVICE                                                        STATE  
1-1    USB Serial Converter                                          Not attached  
1-2    Périphérique d'entrée USB                                     Not attached  
1-4    Intel(R) Wireless Bluetooth(R)                                Not attached  
1-12   Périphérique vidéo USB                                        Not attached  

usbipd wsl attach -b 1-1  
[sudo] password for xxxxxxx:  
sudo: usbip: command not found  
usbipd: error: Failed to attach device with ID '1-1'.`  

`Get-PnpDevice -FriendlyName "USB Serial Converter" | fl *

Class                       : USB
FriendlyName                : USB Serial Converter
InstanceId                  : USB\VID_0403&PID_6001\A15VXDQP
Problem                     : CM_PROB_NONE
ConfigManagerErrorCode      : CM_PROB_NONE
ProblemDescription          :
Caption                     : USB Serial Converter
Description                 : USB Serial Converter
InstallDate                 :
Name                        : USB Serial Converter
Status                      : OK
Availability                :
ConfigManagerUserConfig     : False
CreationClassName           : Win32_PnPEntity
DeviceID                    : USB\VID_0403&PID_6001\A15VXDQP
ErrorCleared                :
ErrorDescription            :
LastErrorCode               :
PNPDeviceID                 : USB\VID_0403&PID_6001\A15VXDQP
PowerManagementCapabilities :
PowerManagementSupported    :
StatusInfo                  :
SystemCreationClassName     : Win32_ComputerSystem
SystemName                  : xxxxxxx
ClassGuid                   : {36fc9e60-c465-11cf-8056-444553540000}
CompatibleID                : {USB\Class_FF&SubClass_FF&Prot_FF, USB\Class_FF&SubClass_FF, USB\Class_FF}
HardwareID                  : {USB\VID_0403&PID_6001&REV_0600, USB\VID_0403&PID_6001}
Manufacturer                : FTDI
PNPClass                    : USB
Present                     : True
Service                     : FTDIBUS
PSComputerName              :
CimClass                    : ROOT/cimv2 : Win32_PnPEntity
CimInstanceProperties       : {Caption, Description, InstallDate, Name...}
CimSystemProperties         : Microsoft.Management.Infrastructure.CimSystemProperties`

thanks

dorssel, 2021-11-14:
@Eric-N-Be 
Your problem is different. It says:

```
sudo: usbip: command not found
```

This means that the USBIP client on the Linux end is not available. Please follow the instructions on how to setup the client side: https://github.com/dorssel/usbipd-win/wiki/WSL-support

dorssel, 2021-11-14:
@calipzo 
@pouyaemami 
@willzyx-hub
I have updated the heuristic strategy on port power cycling in case VBoxUsbMon fails to grab the device. This has helped me in the past with some integrated devices. Hopefully this helps with your problems too.

Could you please test the build artifact of PR #93, it is at https://github.com/dorssel/usbipd-win/actions/runs/1459452495.
Note that the MSI is unsigned, but the drivers are.

willzyx-hub, 2021-11-15:
> @calipzo @pouyaemami @willzyx-hub I have updated the heuristic strategy on port power cycling in case VBoxUsbMon fails to grab the device. This has helped me in the past with some integrated [...]

Unfortunately it doesn't either fix my problem 

the error still same

C:\Windows\system32>usbipd wsl attach -b 5-2
[sudo] password for willy:
usbip: error: Attach Request for 5-2 failed - Request Failed

usbipd: error: Failed to attach device with ID '5-2'.

calipzo, 2021-11-15:
Confirm, the error still same

Eric-N-Be, 2021-11-15:
> @Eric-N-Be Your problem is different. It says:
> 
> ```
> sudo: usbip: command not found
> ```
> 
> This means that the USBIP client on the Linux end is not available. Please follow the instructions on how to setup the client side: https://github.com/dorssel/usbipd-win/wiki/WSL-support

thanks for your help ..

I have a debian distro, then a need to change a little your procedure.
i don't find any linux-tools-5.4.0-77-generic on debian 11 then i just install :
-  sudo apt-get update 
-  sudo apt-get install usbutils (for lsusb)  
-  sudo apt-get install hwdata  
-  sudo apt-get install usbip  

i don't have at this time change the /etc/sudoers

At this time it's working for a simple mass storage ... i will test later with "USB Serial Converter"
 :~# lsusb
Bus 002 Device 002: ID 0951:1666 Kingston Technology DataTraveler 100 G3/G4/SE9 G2/50

dorssel, 2021-11-15:
@calipzo
@pouyaemami
@willzyx-hub
And yet another attempt. This time a quite drastic disable/re-enable cycle for the device if immediate attach doesn't work. Could you give this one a try?

Build artifact at:
https://github.com/dorssel/usbipd-win/actions/runs/1464274520
Note that the MSI is unsigned, but the drivers are.

willzyx-hub, 2021-11-15:
> @calipzo @pouyaemami @willzyx-hub And yet another attempt. This time a quite drastic disable/re-enable cycle for the device if immediate attach doesn't work. Could you give this one a try?
>
> Build [...]

well no not working either

C:\Windows\system32>usbipd wsl attach -b 3-3
[sudo] password for willy:
usbip: error: Attach Request for 3-3 failed - Request Failed

usbipd: error: Failed to attach device with ID '3-3'.

pouyaemami, 2021-11-16:
@dorssel thanks for looking into this. Here is some more info:

> Can you provide make/model and the Vendor/Product ID (0123:4567)?

```
Class                       : USBDevice
FriendlyName                : OnePlus
InstanceId                  : USB\VID_2A70&PID_4EE7\F21DF099
Problem                     : CM_PROB_NONE
ConfigManagerErrorCode      : CM_PROB_NONE
ProblemDescription          :
Caption                     : OnePlus
Description                 : ADB Device
InstallDate                 :
Name                        : OnePlus
Status                      : OK
Availability                :
ConfigManagerUserConfig     : False
CreationClassName           : Win32_PnPEntity
DeviceID                    : USB\VID_2A70&PID_4EE7\F21DF099
ErrorCleared                :
ErrorDescription            :
LastErrorCode               :
PNPDeviceID                 : USB\VID_2A70&PID_4EE7\F21DF099
PowerManagementCapabilities :
PowerManagementSupported    :
StatusInfo                  :
SystemCreationClassName     : Win32_ComputerSystem
SystemName                  : ATHENA
ClassGuid                   : {88bae032-5a81-49f0-bc3d-a4ff138216d6}
CompatibleID                : {USB\Class_FF&SubClass_42&Prot_01, USB\Class_FF&SubClass_42, USB\Class_FF}
HardwareID                  : {USB\VID_2A70&PID_4EE7&REV_0404, USB\VID_2A70&PID_4EE7}
Manufacturer                : WinUsb Device
PNPClass                    : USBDevice
Present                     : True
Service                     : WINUSB
PSComputerName              :
CimClass                    : ROOT/cimv2:Win32_PnPEntity
CimInstanceProperties       : {Caption, Description, InstallDate, Name...}
CimSystemProperties         : Microsoft.Management.Infrastructure.CimSystemProperties
```
My device list:
```
PS C:\Users\Pouya Emami> usbipd list
Present:
BUSID  DEVICE                                                        STATE
1-8    Integrated Camera                                             Not shared
1-9    Synaptics WBDI                                                Not shared
1-10                                                                 Not shared
1-12   Integrated IR Camera                                          Not shared
1-14   Intel(R) Wireless Bluetooth(R)                                Not shared
1-19   Realtek USB 3.0 Card Reader                                   Not shared
11-4   USB Mass Storage Device                                       Not shared
11-5   Billboard Device                                              Not shared
13-3   ADB Device                                                    Not shared
13-4   USB Audio Device, USB Video Device                            Not shared
14-4   ThinkPad Thunderbolt 3 Dock USB Audio, USB Input Device       Not shared
```
Trying to attach:
```
PS C:\windows\system32> usbipd wsl attach -b 13-3
[sudo] password for pouya:
usbip: error: Attach Request for 13-3 failed - Request Failed

usbipd: error: Failed to attach device with ID '13-3'.
```

<details>
  <summary>Enabling trace logs</summary>

```
PS C:\windows\system32> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\windows\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.24.27.25
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 148
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 200
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 206
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 133
dbug: UsbIpServer.Server[1000]
      connection closed
```
</details>

Is there anything that needs to be on the Linux side after installing the new MSI?

dorssel, 2021-11-16:
@pouyaemami 
Thanks for reporting back. I have a few more follow up questions:

- Interesting (but not really problematic) is device 1-10 (see #69). What type of device is this? USB Audio?
- The error is still `System.IO.FileNotFoundException`, but I see no troublesome devices. In fact, many other people have successfully attached ADB devices. I think the problem is more with the USB root/USB hub. Maybe something is special there. Could you list the details of the USB Root and the USB hub(s) that device 13-3 (ADB) is connected to? And what drivers they use?
- Just to verify if the enable-cycle was actually hit in the code: how much time was there between the attach command and the failure?

dorssel, 2021-11-16:
1. Yes, I know, that's 13-3. But I am interested in 1-10, since that shows an empty device descriptor in `usbipd list`
2. I mean the USB root and USB hub *on the Windows side*. In order to grab the device on Windows, the VBoxUsbMon driver needs to fake the reported device ID and it does that by hooking into the hub the device is connected to. So, the problem  could be some weird compatibility thing with the hub (or the root) and its Windows driver.
3. That sounds correct, that means that the latest addition of a disable/re-enable was indeed attempted. (unfortunately, as that means it did not solve the problem...)

willzyx-hub, 2021-11-16:
> 1. Yes, I know, that's 13-3. But I am interested in 1-10, since that shows an empty device descriptor in `usbipd list`
> 2. I mean the USB root and USB hub _on the Windows side_. In order to grab the [...]

ok i will look into driver and will correct it if i see clumsiness

pouyaemami, 2021-11-16:
@dorsse
Here is an export of USBView on my machine: [USBViewAll.txt](https://github.com/dorssel/usbipd-win/files/7546485/USBViewAll.txt)

1. I am not sure what the device is so I hope the above export clears that up. Please let me know if there's anything else I can do to get you more details.
2. I think the details for the root hub of this device should also be available in the export.
3. It took about 10-12 seconds before failing.

Something to note: I have a ThinkPad Thunderbolt 3 Dock that I've connected to my machine and I've connected my Android device to it. I also tried connecting directly to my laptop but the issues still persists. I also tried this without the dock connected and still no success.

dorssel, 2021-11-16:
1. Device 1-10 appears to be a Wacom Device. It should show up in the Device Manager as a "USB Composite Device" with one or more "HID class" devices below it. My suspicion why the device has an empty descriptor was incorrect. This device looks perfectly normal in your USBView export, with exactly one configuration. I am interested in how it looks in Device Manager, when you select View -> Devices by connection. Can you make a screenshot of the unfolded device tree of that device?
2. The driver versions are not in the export. But what strikes me is that all your root hubs report:  
   ```
   Host Controller Power State Mappings
   System State		Host Controller		Root Hub	USB wakeup	Powered
   S? (unmapped)   	S? (unmapped)   	S? (unmapped)   		
   S? (unmapped)   	S? (unmapped)   	S? (unmapped)   		
   S? (unmapped)   	S? (unmapped)   	S? (unmapped)   		
   S? (unmapped)   	S? (unmapped)   	S? (unmapped)   		
   S? (unmapped)   	S? (unmapped)   	S? (unmapped)   		
   Last Sleep State	S? (unmapped)   
   RootHub
   ```
   All my root hubs have only a few (unmapped) entries, but most are mapped, for example (my machine):
   ```
   Host Controller Power State Mappings
   System State		Host Controller		Root Hub	USB wakeup	Powered
   S0 (working)    	D0              	S? (unmapped)   		
   S1 (sleep)      	D? (unspecified)	D3              		
   S2 (sleep)      	D? (unspecified)	D3              		
   S3 (sleep)      	D3              	D2              Yes		Yes
   S4 (Hibernate)  	D3              	D2              Yes		Yes
   Last Sleep State	S3 (sleep)      
   ```
   Maybe it helps if you try updating the drivers for your USB root hubs, or (re)install Intel Chipset Software. It is as if your root hubs know nothing about power management. And one of the things required for VBoxUsbMon is that the hub is capable of power cycling the ports.
3. 10-12 seconds is to be expected with my latest attempt, which clearly did not solve the problem.

pouyaemami, 2021-11-16:
@dorssel 
1. Here's a snippet of my USB controllers and devices:
[image]
2. How do I get this report in the console?

Let me see what I can do regarding the drivers and I'll get back to you.

pouyaemami, 2021-11-16:
Just tried uninstalling all 3 USB Root Hub devices and rebooting to see if that made a difference but it didn't work. Here are the details about the drivers: 
[image]
All 3 are the same and they all have the `Allow the computer to turn off this device to save power` checkbox ticked under the Power Management tab.

dorssel, 2021-11-16:
@pouyaemami 
Thanks, the drivers for that hub are fine. But with the device tree I meant "View -> Device by connection". For example, on my machine:
[image]
Please provide the driver details of everything from the Root Controller to the last hub, so in my case:
- Intel(R) 8 Series/C220 Series USB EHCI ...
- USB-hoofdhub (that's Dutch for "USB root hub")
- Generic USB Hub

And then also the details for your 1-10 device, which is some Wacom Device. It may be listed as "USB Composite Device". For example in my case:
[image]
In your case it should say: `Location: Port_#0010.Hub_#0001` (i.e. 1-10). I'm interested in the device tree of that device. For example in my case:
- Logitech Unifying USB receiver
- Logitech USB Input Device
- USB-invoerapparaat

pouyaemami, 2021-11-16:
@dorssel

Here are all my root hubs:
[image]
[image]
[image]

Do you want driver details for all of these devices?

Here's the info on the 1-10 device:
[image]
[image]

dorssel, 2021-11-16:
Ah, "Lenovo Pen Device". Lets examine that first. Can you run the following in a CMD prompt and report the output:
```
wmic path Win32_PnPEntity where "Description = 'Lenovo Pen Device'" get *
```

pouyaemami, 2021-11-17:
@dorssel Here's what I get when I run the command you provided (I had to change the `Description` to `Wacom Device` as there was no such device with a description that matches `Lenovo Pen Device`):
```
PS C:\Users\Pouya Emami> wmic path Win32_PnPEntity where "Description = 'Wacom Device'" get *
Availability  Caption            ClassGuid                               CompatibleID                                                                                                ConfigManagerErrorCode  ConfigManagerUserConfig  CreationClassName  Description   DeviceID                               ErrorCleared  ErrorDescription  HardwareID                                                   InstallDate  LastErrorCode  Manufacturer      Name               PNPClass  PNPDeviceID                            PowerManagementCapabilities  PowerManagementSupported  Present  Service          Status  StatusInfo  SystemCreationClassName  SystemName
              Lenovo Pen Device  {745a17a0-74d3-11d0-b6fe-00a0c90f57da}  {"USB\DevClass_00&SubClass_00&Prot_00", "USB\DevClass_00&SubClass_00", "USB\DevClass_00", "USB\COMPOSITE"}  0                       FALSE                    Win32_PnPEntity    Wacom Device  USB\VID_056A&PID_5193\5&380BB8FC&1&10                                  {"USB\VID_056A&PID_5193&REV_0002", "USB\VID_056A&PID_5193"}                              Wacom Technology  Lenovo Pen Device  HIDClass  USB\VID_056A&PID_5193\5&380BB8FC&1&10                                                         TRUE     WacHIDRouterISD  OK                  Win32_ComputerSystem     ATHENA
```

I also ran `Get-PnpDevice` to get a better output:

```
PS C:\Users\Pouya Emami> Get-PnpDevice -FriendlyName "Lenovo Pen Device" | fl *

Class                       : HIDClass
FriendlyName                : Lenovo Pen Device
InstanceId                  : USB\VID_056A&PID_5193\5&380BB8FC&0&10
Problem                     : CM_PROB_PHANTOM
ConfigManagerErrorCode      : CM_PROB_PHANTOM
ProblemDescription          :
Caption                     : Lenovo Pen Device
Description                 : Wacom Device
InstallDate                 :
Name                        : Lenovo Pen Device
Status                      : Unknown
Availability                :
ConfigManagerUserConfig     : False
CreationClassName           : Win32_PnPEntity
DeviceID                    : USB\VID_056A&PID_5193\5&380BB8FC&0&10
ErrorCleared                :
ErrorDescription            :
LastErrorCode               :
PNPDeviceID                 : USB\VID_056A&PID_5193\5&380BB8FC&0&10
PowerManagementCapabilities :
PowerManagementSupported    :
StatusInfo                  :
SystemCreationClassName     : Win32_ComputerSystem
SystemName                  : ATHENA
ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
CompatibleID                : {USB\DevClass_00&SubClass_00&Prot_00, USB\DevClass_00&SubClass_00, USB\DevClass_00, USB\COMPOSITE}
HardwareID                  : {USB\VID_056A&PID_5193&REV_0002, USB\VID_056A&PID_5193}
Manufacturer                : Wacom Technology
PNPClass                    : HIDClass
Present                     : False
Service                     : WacHIDRouterISD
PSComputerName              :
CimClass                    : ROOT/cimv2:Win32_PnPEntity
CimInstanceProperties       : {Caption, Description, InstallDate, Name...}
CimSystemProperties         : Microsoft.Management.Infrastructure.CimSystemProperties

Class                       : HIDClass
FriendlyName                : Lenovo Pen Device
InstanceId                  : USB\VID_056A&PID_5193\5&380BB8FC&1&10
Problem                     : CM_PROB_NONE
ConfigManagerErrorCode      : CM_PROB_NONE
ProblemDescription          :
Caption                     : Lenovo Pen Device
Description                 : Wacom Device
InstallDate                 :
Name                        : Lenovo Pen Device
Status                      : OK
Availability                :
ConfigManagerUserConfig     : False
CreationClassName           : Win32_PnPEntity
DeviceID                    : USB\VID_056A&PID_5193\5&380BB8FC&1&10
ErrorCleared                :
ErrorDescription            :
LastErrorCode               :
PNPDeviceID                 : USB\VID_056A&PID_5193\5&380BB8FC&1&10
PowerManagementCapabilities :
PowerManagementSupported    :
StatusInfo                  :
SystemCreationClassName     : Win32_ComputerSystem
SystemName                  : ATHENA
ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
CompatibleID                : {USB\DevClass_00&SubClass_00&Prot_00, USB\DevClass_00&SubClass_00, USB\DevClass_00, USB\COMPOSITE}
HardwareID                  : {USB\VID_056A&PID_5193&REV_0002, USB\VID_056A&PID_5193}
Manufacturer                : Wacom Technology
PNPClass                    : HIDClass
Present                     : True
Service                     : WacHIDRouterISD
PSComputerName              :
CimClass                    : ROOT/cimv2:Win32_PnPEntity
CimInstanceProperties       : {Caption, Description, InstallDate, Name...}
CimSystemProperties         : Microsoft.Management.Infrastructure.CimSystemProperties
```

It's weird that I get two devices when I run the command using the `FriendlyName`. One has the `Status: OK` while the other has `Status: Unknown`

I only get one device when I run the command using `InstanceId`: `Get-PnpDevice -InstanceId 'USB\VID_056A&PID_5193\5&380BB8FC&1&10' | fl *`

willzyx-hub, 2021-11-17:
> Ah, "Lenovo Pen Device". Lets examine that first. Can you run the following in a CMD prompt and report the output:
> 
> ```
> wmic path Win32_PnPEntity where "Description = 'Lenovo Pen Device'" get *
> ```

ok i have checked all the driver and all the usb port it work without any issue, And i Have reinstalled new Windows 11 and will tell you if Reinstalling windows solve the Problem

dorssel, 2021-11-17:
@pouyaemami 
Thanks for that list. It looks normal. The reason the PowerShell command gives two entries: at some point Windows enumerated the device again, and the first entry is marked as `Present: False`. Only the second one is actual.

This is what I expected, the device itself is USB\COMPOSITE. Now it is time to list its children (PowerShell this time):
```
Get-PnpDeviceProperty -InstanceId 'USB\VID_056A&PID_5193\5&380BB8FC&1&10' -KeyName DEVPKEY_Device_Children | Get-PnpDevice -Instance {$_.Data}
```

dorssel, 2021-11-17:
@willzyx-hub 
I'm afraid we have not found the root cause for the `System.IO.FileNotFoundException` error, yet.
The other issue discussed here is about devices that have no description (it's a different issue). @pouyaemami is making progress there, but is not related to your issue.

dorssel, 2021-11-17:
@calipzo
@pouyaemami
@willzyx-hub
I found this:
https://www.virtualbox.org/ticket/16032
Are any of you using USBPcap or similar software? The symptoms are identical.

calipzo, 2021-11-17:
No, I don't use USBPcap at the moment (I thought of that as soon as I got the glitch).

But now I remembered that I once installed UsbDk. And - bingo! After uninstalling, it worked as it should!

Thanks!
[image]

dorssel, 2021-11-17:
@calipzo 
Ah! Progress. Thanks for reporting back!

willzyx-hub, 2021-11-17:
> @calipzo @pouyaemami @willzyx-hub I found this: https://www.virtualbox.org/ticket/16032 Are any of you using USBPcap or similar software? The symptoms are identical.
> 
> Also found this: [desowin/usbpcap#9](https://github.com/desowin/usbpcap/issues/9)
> 
> Can you check your registry to see if any UpperFilters are installed at: `HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}`

okay i will try it will report back if it's still same problem

dorssel, 2021-11-17:
@calipzo 
It appears to be a known issue, see https://github.com/daynix/UsbDk/issues/40

willzyx-hub, 2021-11-17:
> @calipzo It appears to be a known issue, see [daynix/UsbDk#40](https://github.com/daynix/UsbDk/issues/40)

i also didn't use usbpcab at this moment

dorssel, 2021-11-17:
@willzyx-hub 
What's the output of the following command:
```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

willzyx-hub, 2021-11-17:
> @willzyx-hub What's the output of the following command:
> 
> ```
> reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
> ```

wait let me install wsl and usbipd back since i reinstalled official windows 11 instead of unofficial one

willzyx-hub, 2021-11-17:
> @willzyx-hub What's the output of the following command:
> 
> ```
> reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
> ```

now i can't reinstall it microsoft store gives error while installing it  0x80070005 should i go back to unofficial windows 11 ?

dorssel, 2021-11-17:
No, I don't think so. I also have official Windows 11 + WSL Store Preview.

willzyx-hub, 2021-11-17:
> No, I don't think so. I also have official Windows 11 + WSL Store Preview.

What should i do now :(

dorssel, 2021-11-18:
@willzyx-hub 
Sorry, can't help you there. What I understand is, that you've reinstalled Windows 11 and now the store fails... That's a Windows issue. All I know is that 0x80070005 is just "failure", not a very specific error.

calipzo, 2021-11-18:
@willzyx-hub
You can try to install WSL2 using the command line:
https://www.windowscentral.com/how-install-wsl2-windows-10

willzyx-hub, 2021-11-18:
> @willzyx-hub Sorry, can't help you there. What I understand is, that you've reinstalled Windows 11 and now the store fails... That's a Windows issue. All I know is that 0x80070005 is just "failure", not a very specific error.

now after reinstalling Microsoft store it's working

willzyx-hub, 2021-11-18:
> @willzyx-hub Sorry, can't help you there. What I understand is, that you've reinstalled Windows 11 and now the store fails... That's a Windows issue. All I know is that 0x80070005 is just "failure", not a very specific error.

now it shows this error

C:\Windows\system32>usbipd wsl attach -b 3-3
[sudo] password for willy:
sudo: usbip: command not found
usbipd: error: Failed to attach device with ID '3-3'.

willzyx-hub, 2021-11-18:
> @willzyx-hub Sorry, can't help you there. What I understand is, that you've reinstalled Windows 11 and now the store fails... That's a Windows issue. All I know is that 0x80070005 is just "failure", not a very specific error.

sudo apt install linux-tools-5.4.0-77-generic hwdata
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-tools-5.4.0-77-generic
E: Couldn't find any package by glob 'linux-tools-5.4.0-77-generic'

now it shows this

willzyx-hub, 2021-11-18:
> @willzyx-hub Sorry, can't help you there. What I understand is, that you've reinstalled Windows 11 and now the store fails... That's a Windows issue. All I know is that 0x80070005 is just "failure", not a very specific error.

UPDATE !!!

When i use official windows 11

it shows no error
C:\Windows\system32>usbipd wsl attach -b 3-3
[sudo] password for willy:

willy@Willy:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2717:ff48 Xiaomi Inc. Mi/Redmi series (MTP + ADB)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

but no ADB Not working as it didn't detect my phone

willy@Willy:~$ adb devices
List of devices attached

dorssel, 2021-11-18:
@willzyx-hub 
That's great news. Your device is now attached, but the default WSL kernel may not have the right drivers for ADB, or it is `udev` related.
Fortunately, this is now well understood, see https://github.com/dorssel/usbipd-win/issues/60

willzyx-hub, 2021-11-18:
> @willzyx-hub That's great news. Your device is now attached, but the default WSL kernel may not have the right drivers for ADB, or it is `udev` related. Fortunately, this is now well understood, see #60

That's error that says failed to attach it's basically Driver related issue... after i update my driver to WinUSB the error will gone...

pouyaemami, 2021-11-18:
@dorssel

I ran the command you requested:

> `Get-PnpDeviceProperty -InstanceId 'USB\VID_056A&PID_5193\5&380BB8FC&1&10' -KeyName DEVPKEY_Device_Children | Get-PnpDevice -Instance {$_.Data} | fl *`

<details>
<summary>Here is the list of attached devices</summary>

```
Class                       : HIDClass
FriendlyName                : HID-compliant touch screen
InstanceId                  : HID\VID_056A&PID_5193&COL01\6&14BD95C8&0&0000
Problem                     : CM_PROB_NONE
ConfigManagerErrorCode      : CM_PROB_NONE
ProblemDescription          :
Caption                     : HID-compliant touch screen
Description                 : HID-compliant touch screen
InstallDate                 :
Name                        : HID-compliant touch screen
Status                      : OK
Availability                :
ConfigManagerUserConfig     : False
CreationClassName           : Win32_PnPEntity
DeviceID                    : HID\VID_056A&PID_5193&COL01\6&14BD95C8&0&0000
ErrorCleared                :
ErrorDescription            :
LastErrorCode               :
PNPDeviceID                 : HID\VID_056A&PID_5193&COL01\6&14BD95C8&0&0000
PowerManagementCapabilities :
PowerManagementSupported    :
StatusInfo                  :
SystemCreationClassName     : Win32_ComputerSystem
SystemName                  : ATHENA
ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
CompatibleID                :
HardwareID                  : {HID\VID_056A&PID_5193&REV_0002&Col01, HID\VID_056A&PID_5193&Col01, HID\VID_056A&UP:000D_U:0004, HID_DEVICE_UP:000D_U:0004...}
Manufacturer                : (Standard system devices)
PNPClass                    : HIDClass
Present                     : True
Service                     :
PSComputerName              :
CimClass                    : ROOT/cimv2:Win32_PnPEntity
CimInstanceProperties       : {Caption, Description, InstallDate, Name...}
CimSystemProperties         : Microsoft.Management.Infrastructure.CimSystemProperties

Class                       : Mouse
FriendlyName                : HID-compliant mouse
InstanceId                  : HID\VID_056A&PID_5193&COL06\6&14BD95C8&0&0005
Problem                     : CM_PROB_NONE
ConfigManagerErrorCode      : CM_PROB_NONE
ProblemDescription          :
Caption                     : HID-compliant mouse
Description                 : HID-compliant mouse
InstallDate                 :
Name                        : HID-compliant mouse
Status                      : OK
Availability                :
ConfigManagerUserConfig     : False
CreationClassName           : Win32_PnPEntity
DeviceID                    : HID\VID_056A&PID_5193&COL06\6&14BD95C8&0&0005
ErrorCleared                :
ErrorDescription            :
LastErrorCode               :
PNPDeviceID                 : HID\VID_056A&PID_5193&COL06\6&14BD95C8&0&0005
PowerManagementCapabilities :
PowerManagementSupported    :
StatusInfo                  :
SystemCreationClassName     : Win32_ComputerSystem
SystemName                  : ATHENA
[... 206 lines cut ...]
DeviceID                    : HID\VID_056A&PID_5193&COL05\6&14BD95C8&0&0004
ErrorCleared                :
ErrorDescription            :
LastErrorCode               :
PNPDeviceID                 : HID\VID_056A&PID_5193&COL05\6&14BD95C8&0&0004
PowerManagementCapabilities :
PowerManagementSupported    :
StatusInfo                  :
SystemCreationClassName     : Win32_ComputerSystem
SystemName                  : ATHENA
ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
CompatibleID                :
HardwareID                  : {HID\VID_056A&PID_5193&REV_0002&Col05, HID\VID_056A&PID_5193&Col05, HID\VID_056A&UP:FF11_U:0002, HID_DEVICE_UP:FF11_U:0002...}
Manufacturer                : (Standard system devices)
PNPClass                    : HIDClass
Present                     : True
Service                     :
PSComputerName              :
CimClass                    : ROOT/cimv2:Win32_PnPEntity
CimInstanceProperties       : {Caption, Description, InstallDate, Name...}
CimSystemProperties         : Microsoft.Management.Infrastructure.CimSystemProperties
```
</details>

I don't have USBPcap installed or anything like it (to my knowledge).

dorssel, 2021-11-18:
@pouyaemami 
Thanks for the list. I will copy it to issue #69. At least I understand that issue now, see explanation there. Thanks again for helping out!

About the original issue: `System.IO.FileNotFoundException`. Lets first (re)confirm you actually have the same issue. Please run the server from the console with full logging while trying to attach. See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting. Assuming you indeed get the `System.IO.FileNotFoundException` error, please post the output of the following command
```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

pouyaemami, 2021-11-18:
@dorssel Here's the output you requested:
```
PS C:\Users\Pouya Emami> reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}
    UpperFilters    REG_MULTI_SZ    vl810filter
```

I am not clear on why I needed to run the server in the console as there were not additional logs given. Should I also try to run the attach command?

dorssel, 2021-11-18:
Yes, run on the console with full logging, and while logging run the attach command.
But your previous output is a clear indicator! `vl810filter`, what is that?
The list of UpperFilters is expected to be empty (or even non-existent). Other users have solved this problem by uninstalling other software (USBPcap, UsbDK) that installed such a filter. Google says: 'VIA Labs VL810 Superspeed USB Hub Controller Filter Driver'
Could that be? I think it might be incompatible with `usbipd-win`.

dorssel, 2021-11-18:
@pouyaemami 
see https://forums.nomachine.com/topic/continuous-usb-issues-windows-10-x64#post-15879

pouyaemami, 2021-11-18:
@dorssel that's interesting! I just did a quick Google search and found [this](https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/Detected-unrecognized-USB-driver-Driver-vl810filter-on-X1-Extreme-with-Thunderbolt-Dock/m-p/4485223?page=1#5051831). Turns out that I do have something similar but disabling the service doesn't seem to help. Perhaps I need to uninstall it and try again. I will keep you posted on this!

dorssel, 2021-11-18:
@pouyaemami 
I would just remove that UpperFilters entry from the register. Reboot. And try attach.

pouyaemami, 2021-11-18:
@dorssel so I just rebooted after removing the registry entry. Ran the attach command and boom! Works like a charm :) Thank you so much for all the help and getting things working! If you ever need me to test things out for you, please let me know. It's the least I can do.

dorssel, 2021-11-18:
@pouyaemami 
Thank you too! You kept on delivering with all my information requests. That helped track 2 issues down!

Here is what I will do:
1) I will add a registry check for UpperFilters and report a warning in `usbipd` if one is detected. Something like "warning: possible incompatible USB filter software detected"
2) mark this issue as documentation, for others to find. Eventually I will write it on the Wiki (once I have time)
3) Improve the device description listing, now that I know that USB\COMPOSITE devices can have children none of which have a USB marker

Ultimate goal: get every USB device connected to every WSL installation out there... (ok, ambitious, I admit)

NikLeberg, 2021-11-27:
Hi there and thank you @dorssel for this great tool!

I'm one of those that is getting this FileNotFoundException. Sometimes I can attach my USB device (STLink Debugger), somtimes I cannot. Removing the "UpperFilters" (USBPcap was installed) did not work. But what I found out is, that if I have a dongle (simple hdmi and USB adapter) attached to my laptop (via USB-C). I can't attach. If i remove the dongle, I can attach!

I don't know if this helps in any way. Let me know if I should provide more information for investigation.

dorssel, 2021-11-28:
@NikLeberg 
That's interesting new information, thanks!

I wonder if the dongle, which isn't really a USB device but is attached to a hub that *could* take a USB device, is throwing device enumeration off? Could you run `usbview.exe` (from Windows SDK) with and without the dongle attached, and report if there is a difference between them?

If the problem is with VBoxUsbMon (which is what I suspect), this is a nasty one to track down. I am working on a more aggressive device grabber, see https://github.com/dorssel/usbipd-win/actions/runs/1464274520. Although that build is a bit old you may want to give it a try.

NikLeberg, 2021-11-28:
I don't really know how to read the `usbview.exe` output. I've attached the info it generates:
[USBView_with_dongle.txt](https://github.com/dorssel/usbipd-win/files/7613068/USBView_with_dongle.txt)
[USBView_without_dongle.txt](https://github.com/dorssel/usbipd-win/files/7613069/USBView_without_dongle.txt)

Sadly with the more aggressive device grabber doesnt work either. But the error changes! (It stills works if I disconnect the dongle though.)

```
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.InvalidOperationException: Sequence contains no elements
         at System.Linq.ThrowHelper.ThrowNoElementsException()
         at System.Linq.Enumerable.Single[TSource](IEnumerable`1 source)
         at UsbIpServer.VBoxUsbMon.CycleEnable(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 174
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 215
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 133
```

dorssel, 2021-12-01:
@NikLeberg 
I have a new attempt on a more aggressive device grabber in PR #134. Could you test the build artifact? It is at https://github.com/dorssel/usbipd-win/actions/runs/1527927708.
Please test with and without the dongle, and check if the `FileNotFound` exception turns up. Thanks!

NikLeberg, 2021-12-02:
Well now it gets a bit funny @dorssel ...

I installed the artifact of PR #134
Without the dongle is works always.
With the dongle it works ~50% of the time. Sadly I can not make out a "pattern". Sometimes it works on first try (after service startup e.g. `usbipd server Logging:LogLevel:Default=Trace`) sometimes only after the fifth.

But I could make some observations:
- the device name listed with `usbipd wsl list` changed from `STMicroelectronics STLink dongle` to `STM32 STLink`
- the windows connect / disconnect sounds were a bit different at times. If i got it to attach (with dongle) it played both sounds if I detached. If I attached it (with no dongle) and then detached it, it played only the disconnect sound. 🤷

The error in the tracelog was always the following:

```
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 295
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 134
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 144
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 150
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 233
```

After that I tried a few times more with the current version `1.2.0+51.Branch.master.Sha.a1d6f0c6ff27baae9aab9f279a61d196dad3d0b9`
Somehow it now also works with that version after a few tries. I dont know if something changed somewhere or if I didnt test it that often then (sorry).

If I can provide some additional logging / debugging I'm more than happy to assist!

dorssel, 2021-12-02:
@NikLeberg 
Thanks for all the info. The whole device detection/enumeration changed on the master branch. That may have helped. It is also the reason why the list name changed. That is to be expected.

How is the timing? How much time do you guestimate for successful attach without dongle, with dongle, and failure with dongle?

The "double sound" thing is interesting as well. Do you get any disconnect/connect sounds when it *fails*?

[6 more replies]

PR #134 (2021-12-04): Add disable/enable during attach
Fixes #63 
Fixes #103
closed 2021-12-04 by dorssel