dorssel/usbipd-win #64: usb device crashes when accessing it from WSL

knmcguire, 2021-11-05:
Hi! First of all, awesome work! Been looking for a USB port from WSL2 in windows for quite a while now :) And now I saw [microsoft' dev post](https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/) i was more than happy to give it a go.

So let me explain my situation. I want to link this radio device,[ the crazyradio PA](https://www.bitcraze.io/products/crazyradio-pa/) from bitcraze to control and flash a tiny drone called [the crazyflie](https://www.bitcraze.io/products/crazyflie-2-1/). The PA is installed by [zadig with libusb](https://www.bitcraze.io/documentation/repository/crazyradio-firmware/master/building/usbwindows/). 

Unfortunately, if I try to access it through python on the WSL2 side, the **usb device crashes**, eventhough I can access it from windows with no problem.

After installing everything according [to the wiki](https://github.com/dorssel/usbipd-win/wiki/WSL-support) also see it in the list of usbipd on the windows side and once I attach it I see:
Windows: usbipd list
```
BUSID  DEVICE                                                        STATE
1-1    Crazyradio PA USB Dongle                                      Attached
1-2    Jabra Link 370 (Interface 0), USB Input Device                Not shared
1-6    Synaptics WBDI                                                Not shared
1-7    Intel(R) Wireless Bluetooth(R)                                Not shared
1-8    Integrated Camera                                             Not shared
1-9    Alcor Micro USB Smart Card Reader                             Not shared

```
lbusb on ubuntu:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 038: ID 1915:7777 Nordic Semiconductor ASA Crazyradio PA USB Dongle
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
usbip port on ubuntu:
```
Imported USB devices
====================
libusbip: error: fopen
libusbip: error: read_record
Port 00: <Port in Use> at Full Speed(12Mbps)
       Nordic Semiconductor ASA : unknown product (1915:7777)
       1-1 -> unknown host, remote port and remote busid
           -> remote bus/dev 001/001
```
After trying to access it through pyusb, the following happens:

```
kim@DESKTOP~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
kim@DESKTOP:~$ usbip port --trace
usbip: error: open vhci_driver
usbip: error: list imported device
```

So there is something going wrong still! Maybe the libusbip error for `fopen ` and `read_record `has something to do with it?

Here some more info about my system:

**Host**:
 Windows 10 Pro
 Version 10.0.19042 Build 19042

**WSL2:**
 Ubuntu 20.04
 Kernel version 5.10.60.1
 Udev rules. `SUBSYSTEM=="usb", ATTRS{idVendor}=="1915", ATTRS{idProduct}=="7777", MODE="0664", GROUP="plugdev"`

dorssel, 2021-11-06:
You got very far, up to the point where >10 USB messages have successfully been passed back-and-forth. The "setup" stage for a device (configuration) is usually the most sensitive. When that goes wrong, Linux will usually just give up and close the USBIP connection. This is why you no longer see the device after the error, it got detached. This may also happen if usbipd-win had an internal error.

We need some debug logging to find out which side has the first fatal error:

- on Linux (WSL): syslog has logging for USB errors
- on the Windows side, check https://github.com/dorssel/usbipd-win/wiki/Troubleshooting on how to get debug logging

dorssel, 2021-11-07:
Added to [Tested Devices](https://github.com/dorssel/usbipd-win/wiki/Tested-Devices) as not working

knmcguire, 2021-11-07:
Alright, I wasn't able to get a syslog, but I did do a dmesg -wH, which might be useful.

I'm now showing the debug valures right from the moment that the device gets attached to WSL. Hope that this will show enough for you to to get an idea?

Output of windows uspib server with debug on:

```
dbug: UsbIpServer.Server[0]
      new connection from 172.17.15.110
dbug: UsbIpServer.ConnectedClient[0]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.17.15.110 claimed device at 1-1 (USB\VID_1915&PID_7777\8A35BE1269).
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=76, flags=512, length=64, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 256 0 64
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=76, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 18, requested: 64
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=77, flags=512, length=18, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 256 0 18
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=77, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 18, requested: 18
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=78, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 1536 0 10
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=78, pending count = 1
dbug: UsbIpServer.AttachedClient[0]
      USBSUP_XFER_STALL -> EPIPE -> 0
trce: UsbIpServer.AttachedClient[0]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=79, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 1536 0 10
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=79, pending count = 1
dbug: UsbIpServer.AttachedClient[0]
      USBSUP_XFER_STALL -> EPIPE -> 0
trce: UsbIpServer.AttachedClient[0]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=80, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 1536 0 10
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=80, pending count = 1
dbug: UsbIpServer.AttachedClient[0]
      USBSUP_XFER_STALL -> EPIPE -> 0
trce: UsbIpServer.AttachedClient[0]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=81, flags=512, length=9, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 512 0 9
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=81, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 9, requested: 9
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=82, flags=512, length=32, ep=0
trce: UsbIpServer.AttachedClient[0]
[... 104 lines cut ...]
[  +0.220018] vhci_hcd: vhci_device speed not set
[  +0.000028] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[  +0.079975] vhci_hcd: vhci_device speed not set
[  +0.219917] vhci_hcd: vhci_device speed not set
[  +0.220043] vhci_hcd: vhci_device speed not set
[  +0.219863] vhci_hcd: vhci_device speed not set
[  +0.220212] vhci_hcd: vhci_device speed not set
[  +0.000029] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[  +0.000018] usb usb1-port1: attempt power cycle
[  +0.119890] vhci_hcd: vhci_device speed not set
[  +0.220035] vhci_hcd: vhci_device speed not set
[  +0.219919] vhci_hcd: vhci_device speed not set
[  +0.220142] vhci_hcd: vhci_device speed not set
[  +0.219890] vhci_hcd: vhci_device speed not set
[  +0.000029] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[  +0.079927] vhci_hcd: vhci_device speed not set
[  +0.220065] vhci_hcd: vhci_device speed not set
[  +0.219955] vhci_hcd: vhci_device speed not set
[  +0.220045] vhci_hcd: vhci_device speed not set
[  +0.220045] vhci_hcd: vhci_device speed not set
[  +0.000028] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[  +0.000018] usb usb1-port1: unable to enumerate USB device
[  +8.803420] WSL2: Performing memory compaction.
[Nov 7 20:57] WSL2: Performing memory compaction.
```

dorssel, 2021-11-07:
Hi, I fixed a bug that may just solve your issue. Would you be willing to try the MSI from this  PR: https://github.com/dorssel/usbipd-win/actions/runs/1432468492?

You would have to uninstall usbipd-win, and then install this MSI. It is not signed yet (but the drivers are).

closed 2021-11-08 by dorssel

knmcguire, 2021-11-08:
Thanks for looking into it!

I tried the new build and unfortunately I still see the same errors with this device, with no noticeable change in the logging. Anything else I should log in order to get more data?

knmcguire, 2022-05-17:
I've dropped this for a little while now but I now have an update on this now that I have a new computer

 With Windows 11, WSL2 with Ubuntu 20.04 and usbipd 2.3, it appears to be working! @dorssel you can update that in [Tested Devices](https://github.com/dorssel/usbipd-win/wiki/Tested-Devices) for the nordic device.