dorssel/usbipd-win #65: Any possibility of attaching usbip devices to Windows?

yiays, 2021-11-05:
This project is great, but I'm hoping to use it backwards - connecting a USB device from a physical linux computer to a Windows Machine. There have been other implementations of USBIP for Windows that enabled this, however none of them are very well maintained or work without test signing.
I'm sure it's a long shot, but is this at all possible?

dorssel, 2021-11-06:
Short answer: hate to disappoint you, but not with usbipd-win.

Long answer:

Apart from the actual USBIP protocol, server and client are completely dissimilar. If I would ever go for a Windows client, it would be a separate project. But the problem (as always) is: drivers. It cannot be done without a kernel mode driver, which requires either WHQL or Microsoft attestation signatures. For the server side, I just leveraged the VirtualBox drivers, which are already properly signed. But I do not know of any open-source signed client drivers. There are some commercial solution available.

On the up-side, writing the driver is somewhat easier these days now that Microsoft has a USB simulation framework: https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-emulated-device--ude--architecture. But that still requires kernel mode coding, and therefore hardware Authenticode signatures.

closed 2021-11-06 by dorssel

e-t-l, 2022-03-04:
@dorssel Thanks for this explanation, as I had a similar question. If I can ask one follow-up: Is USBIPD-win _only_ for connecting a Windows-attached USB device virtually to a WSL instance, or can it be used to connect to any other USBIP client? (Specifically, I'm hoping to connect to the client from [https://github.com/cezanne/usbip-win](https://github.com/cezanne/usbip-win), which is supposedly an effective Windows USBIP client.)

dorssel, 2022-03-04:
Yes, it should work with any USBIP client, although I myself have tested it only with the default Linux client. Other people have reported success with Windows clients as well.

luc-git, 2024-02-09:
> Short answer: hate to disappoint you, but not with usbipd-win.
>
> Long answer:
>
> Apart from the actual USBIP protocol, server and client are completely dissimilar. If I would ever go for a Windows [...]

Couldn't be doable with libusb instead of a driver?