dorssel/usbipd-win #66: Unable to list usb devices to attach to WSL

LeoBound, 2021-11-06:
Hi,

I'm running the Windows 11 WSL2 preview version (kernel 5.10.60.1-microsoft-standard-WSL2) and usbipd-win 1.1.1.

I've followed [this](https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/) guide and am unable to run `usbipd wsl list`.

I get the following error:
```
Unhandled exception. System.AggregateException: One or more errors occurred. (An item with the same key has already been added. Key: [1, UsbIpServer.UsbConfigurationDescriptors+UsbConfiguration])
 ---> System.ArgumentException: An item with the same key has already been added. Key: [1, UsbIpServer.UsbConfigurationDescriptors+UsbConfiguration]
   at System.Collections.Generic.TreeSet`1.AddIfNotPresent(T item)
   at System.Collections.Generic.SortedSet`1.Add(T item)
   at System.Collections.Generic.SortedDictionary`2.Add(TKey key, TValue value)
   at UsbIpServer.UsbConfigurationDescriptors.AddDescriptor(ReadOnlySpan`1 descriptor) in D:\a\usbipd-win\usbipd-win\UsbIpServer\UsbConfigurationDescriptors.cs:line 96
   at UsbIpServer.ExportedDevice.GetConfigurationDescriptor(DeviceFile hub, UInt16 connectionIndex, Byte numConfigurations)
   at UsbIpServer.ExportedDevice.GetDevice(SafeDeviceInfoSetHandle deviceInfoSet, SP_DEVINFO_DATA devInfoData, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 202
   at UsbIpServer.ExportedDevice.GetAll(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 239
   at UsbIpServer.Program.<>c.<<Main>b__19_18>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 387
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.<>c__DisplayClass79_0.<OnExecute>b__0()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 555
```

Any thoughts?

Thanks

dorssel, 2021-11-06:
Duplicates #54.

There is a PR that will solve this. For now, you will have to unplug the incompatible device (which is probably not the device you want to attach, so that's good). Best guess is: some USB audio device...

LeoBound, 2021-11-06:
Ah my apologies for missing the other issue. You're correct, it was my audio interface causing it! 

Thanks

closed 2021-11-06 by dorssel