dorssel/usbipd-win #67: Failed attaching devices on Ubuntu 21.10 and later (Impish Indri and Jammy Jellyfish)

akuropka, 2021-11-07:
**Environment:**
Microsoft Windows 11 22000.282
Ubuntu-CommPrev (updated to 21.10 and 22.04 respectively)
usbipd-win 1.1.1
WSL version 2
Kernel 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
linux-tools-5.13.0-20

**Listing devices**
> C:\WINDOWS\system32>usbipd wsl list
> BUSID  DEVICE                                                        STATE
> 1-4    Logitech USB Input Device, USB-Eingabegerät                   Not attached
> 1-6    HP HD Camera                                                  Not attached
> 1-7    Intel(R) Wireless Bluetooth(R)                                Not attached
> 1-8    Synaptics FP Sensors (WBF) (PID=003f)                         Not attached
> 1-15   USB-Massenspeichergerät                                       Not attached

**Attaching USB-stick**
> C:\WINDOWS\system32>usbipd wsl attach --busid 1-15
> [sudo] password for andi:
> WARNING: usbip not found for kernel 5.10.60.1-microsoft
> 
>   You may need to install the following packages for this specific kernel:
>     linux-tools-5.10.60.1-microsoft-standard-WSL2+
>     linux-cloud-tools-5.10.60.1-microsoft-standard-WSL2+
> 
>   You may also want to install one of the following packages to keep up to date:
>     linux-tools-standard-WSL2+
>     linux-cloud-tools-standard-WSL2+
> usbipd: error: Failed to attach device with ID '1-15'.

**Notes:**
Followed the given instructions: https://github.com/dorssel/usbipd-win/wiki/WSL-support
Linux tools v5.4.0-77 are not available in the repo, just v5.13.0-20. (https://packages.ubuntu.com/impish/linux-tools-generic)
I didn't check if this error occurs ob Ubuntu 21.04 or lower.

**Fix:**
1. Install Linux tools with sudo apt install linux-tools-generic
2. Build USBIP as described and copy the libusbip.so.0. An own kernel is not necessary.

dorssel, 2021-11-07:
Thanks for the report. I'll leave this open for others to find.

I will change the title, as your post actually contains the correct workaround. In fact, this is not fixable by usbipd-win. It really ought to be fixed by WSL exposing a proper `usbip` client package for their kernel.

mangkoran, 2022-01-11:
Sorry but I am still confused.

> Build USBIP as described and copy in addition the libusbip.so.0.
In [the wiki](https://github.com/dorssel/usbipd-win/wiki/WSL-support), there is not mentioned on how to build USBIP. Its just installing USBIP, edit visudo and stuffs.

akuropka, 2022-01-11:
I can find it in the wiki...

[image]

mangkoran, 2022-01-11:
Guess I'm blind. Thank you!

dorssel, 2022-02-11:
WSL documentation has been updated with a comment on getting `usbip` client side to work on different distros. Closing.

closed 2022-02-11 by dorssel