dorssel/usbipd-win #69: Device description empty

dorssel, 2021-11-07:
Looks good :)

```
WARNING: bConfigurationValue 1 is duplicated. ignored.
Present:
BUSID  DEVICE                                                        STATE
1-5    Intel(R) Wireless Bluetooth(R)                                Not shared
1-6    Integrated Camera                                             Not shared
1-8    Synaptics WBDI                                                Not shared
2-2    DisplayLink USB Audio Adapter, DisplayLink USB Device         Not shared
5-4    Remote NDIS Compatible Device                                 Not shared
7-1                                                                  Not shared
7-2    USB Input Device                                              Not shared
7-3    STMicroelectronics STLink Virtual COM Port, STMicroelectr...  Not shared
8-1    Billboard Device, USB Input Device                            Not shared
8-4    USB Input Device                                              Not shared

Persisted:
GUID                                    BUSID  DEVICE

```

_Originally posted by @ebinans in https://github.com/dorssel/usbipd-win/issues/54#issuecomment-961823001_

dorssel, 2021-11-07:
@ebinans

I noticed that your 7-1 device has no description at all.
My guess is: it's a composite device without specific devices underneath.
Can you check what the device manager tree says about this one? Maybe post a screenshot of that?

Thx.

ebinans, 2021-11-07:
Sorry, I did not pay attention to that. It's the audio device that triggered issue #54.

[image]

dorssel, 2021-11-07:
Thx.

This means that it is just one more issue with devices that have more than one configuration (mostly audio devices, it seems).

dorssel, 2021-11-13:
Newest guess of root cause: it seems that the empty line is for a device with more than one configuration.

dorssel, 2021-11-17:
Newest guess of root cause: the main device is USB\COMPOSITE, but its children are somehow not enumerated.

Note to self: some USB\COMPOSITE have a descriptive FriendlyName, that could be used and should not be filtered out.

dorssel, 2021-11-18:
See user report https://github.com/dorssel/usbipd-win/issues/63#issuecomment-973042293

Root cause seems to be: device is `USB\COMPOSITE`, so currently its description is filtered out and instead a list of child descriptions is used. However, none of the children have a DeviceID that starts with `USB\`, so all of those get filtered out too. Result: empty descriptor.

Noteworthy:
1. In this particular case, the top-level `USB\COMPOSITE` device does have a descriptive `FriendlyName` that could be used
2. The child devices do have very descriptive names, so we could use them even though their DeviceID does not start with `USB\`

PR #120 (2021-11-30): Refactor SetupAPI to Configuration Manager
- Modern universal API
- Faster device enumeration
- Improved device description

Fixes #69
commit eddadb1e (2021-11-30): Refactor SetupAPI to Configuration Manager

- Modern universal API
- Faster device enumeration
- Impoved device description

Fixes #69
closed 2021-11-30 by dorssel