dorssel/usbipd-win #70: USB storage not accessible

akuropka, 2021-11-07:


dorssel, 2021-11-07:
Thx for the report and the dump.

After device recognition (which ends at timestamp 23.506548), thing fail (or rather: stall). This is right after a SEND request, probably to access actual disk content for the first time. After that it is just repeated port resets.

What I notice is: super_speed. That means either USB 3.0 or 3.1. I must admit, I only tested that with my single USB 3 device I have available (a Kingston USB flash drive). So it could very well be that not all of the USB 3 standard is covered. What does not help either is that the USB 2.0 standard is available online, but the USB 3 standard is not...

If you want your device working ASAP, you could try plugging it in a USB 2.0 port, or add a USB 2.0 hub between your USB 3 port and the device.

To debug the USB 3.0 side of things: I am interested in a TRACE log of usbipd-win, as explained in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

akuropka, 2021-11-07:
I tested the USB 2 port as well but the result is the same. I will do some further testing with other devices...

Attached the trace log for reference.
[usbipd_trace.txt](https://github.com/dorssel/usbipd-win/files/7492821/usbipd_trace.txt)

dorssel, 2021-11-07:
Thx.

USBSUP_XFER_UNDERRUN -> EREMOTEIO -> 0

That's the error. I will look into it to find out when/why/how such an error can occur.

akuropka, 2021-11-07:
Thank you very much... I appreciate your support!

dorssel, 2021-11-07:
Hi, I fixed a bug that may just solve your issue. Would you be willing to try the MSI from this  PR: https://github.com/dorssel/usbipd-win/actions/runs/1432468492?

You would have to uninstall usbipd-win, and then install this MSI. It is not signed yet (but the drivers are).

akuropka, 2021-11-08:
Hi @dorssel,
the fix works like a charm. 👍
Everything perfect... great perfomance and runs stable from what I can judge for the moment.

One sidenote for your [wsl-wiki](https://github.com/dorssel/usbipd-win/wiki/WSL-support) since you suggest enabling some features in the kernel and this here is about storage:
By [default](https://github.com/microsoft/WSL2-Linux-Kernel/blob/linux-msft-wsl-5.10.y/Microsoft/config-wsl) `CONFIG_USB_STORAGE` is not set which will result in USB storage devices won't be accessible in Linux.
The same applies for `CONFIG_EXFAT_FS` and `CONFIG_NTFS_FS` which may be useful if someone wants to access exFAT or NTFS drives.

Have a good day! 👋

dorssel, 2021-11-08:
Forwarding the suggestion to the WSL team: @benmcmorran.

Please note that you now have a non-release version running that reports itself as 1.1.2.
Don't forget to do another uninstall/install cycle once the official release comes out...

closed 2021-11-08 by dorssel

akuropka, 2021-11-08:
Btw upgrade did work without uninstall as well... just fyi (forgot to uninstall first on the second device).

dorssel, 2021-11-08:
Yeah, I always forget that a PR build is already on the *next* version. When you go to release 1.1.2, you will have to uninstall or else the installer thinks you are trying to go from 1.1.2 to 1.1.2 (which won't work).

PR #72 (2021-11-08): Fix error return
regression bug introduced by
https://github.com/dorssel/usbipd-win/commit/62ae32296e5d34e316c5e6c7481872d1e5846931

Fixes #70