dorssel/usbipd-win #77: Not able to bind an USB Device

dorssel, 2021-11-08:
### Discussed in https://github.com/dorssel/usbipd-win/discussions/76

<div type='discussions-op-text'>

<sup>Originally posted by **tunguskar** November  8, 2021</sup>
I followed the instructions. After all i Started powerShell on Windows side and typed: (No Admin Rights)

`usbipd wsl list`
All the devices were output. When I now want to bind a device (Hardware Debugger) I get the following error:

`usbipd: error: Access denied.`

When I start  the power shell with admin privileges, I get  the following:

`PS C:\WINDOWS\system32> usbipd wsl list
Unhandled exception. System.AggregateException: One or more errors occurred. (WSL failed to list distributions: )
 ---> UsbIpServer.UnexpectedResultException: WSL failed to list distributions:
   at UsbIpServer.WslDistributions.CreateAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\WslDistributions.cs:line 90
   at UsbIpServer.Program.<>c.<<Main>b__19_18>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 386
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.<>c__DisplayClass79_0.<OnExecute>b__0()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 555`</div>

cjshearer, 2021-11-08:
Got the same thing this morning. I had to uninstall/reinstall usbipd-win.

```
PS C:\Users\Cody> usbipd.exe wsl list
Unhandled exception. System.AggregateException: One or more errors occurred. (expected USB_INTERFACE_DESCRIPTOR_TYPE)
 ---> System.ArgumentException: expected USB_INTERFACE_DESCRIPTOR_TYPE
   at UsbIpServer.UsbConfigurationDescriptors.AddDescriptor(ReadOnlySpan`1 descriptor) in D:\a\usbipd-win\usbipd-win\UsbIpServer\UsbConfigurationDescriptors.cs:line 115
   at UsbIpServer.ExportedDevice.GetConfigurationDescriptor(DeviceFile hub, UInt16 connectionIndex, Byte numConfigurations)
   at UsbIpServer.ExportedDevice.GetDevice(SafeDeviceInfoSetHandle deviceInfoSet, SP_DEVINFO_DATA devInfoData, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 202
   at UsbIpServer.ExportedDevice.GetAll(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 239
   at UsbIpServer.Program.<>c.<<Main>b__19_18>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 387
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.<>c__DisplayClass79_0.<OnExecute>b__0()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 555
```

dorssel, 2021-11-08:
This may be the same issue as #61.

That has been fixed in master, but was not released yet. Could you try the latest master build? The MSI is at https://github.com/dorssel/usbipd-win/actions/runs/1434867413. The MSI is unsigned, but the drivers are. You should be able to just update the current installation.

cjshearer, 2021-11-08:
Just tried it - no luck. I'll make a separate issue later this evening. I've got a workaround for now

tunguskar, 2021-11-09:
I tried the patch, now usbipd does not recognise my wsl2 system:

On Windows powershell without Admin rights:

Ubuntu is running:
`PS F:\> wsl.exe --list --verbose
  NAME      STATE           VERSION
* Debian    Stopped         2
  Ubuntu    Running         2`

But there is no detection:
`PS F:\> usbipd wsl list
usbipd: error: Windows Subsystem for Linux version 2 is not available. See https://aka.ms/installwsl.`

tunguskar, 2021-11-09:
I think this hast to do with #78

dorssel, 2021-11-13:
@tunguskar
Your issue is different. It seems to be the same as #58. Can you confirm you are running WSL from the Store preview?
If so, it was fixed by PR #71; you may want to try the MSI from that build or later.

dorssel, 2021-11-14:
@tunguskar 
I've improved the WSL checks in PR #92.
This may (or may not...) solve your problem too.
Could you please test the build artifact at: https://github.com/dorssel/usbipd-win/actions/runs/1459278946 ?
Note that the MSI is unsigned, but the drivers are.

tunguskar, 2021-11-14:
I will give it a try thanks!

tunguskar, 2021-11-15:
Unfortunately it is still the same: 
```bash
PS C:\Users\baerch> wsl.exe --list --verbose
  NAME      STATE           VERSION
* Debian    Stopped         2
  Ubuntu    Running         2
PS C:\Users\baerch> usbipd wsl list
usbipd: error: Windows Subsystem for Linux version 2 is not available. See https://aka.ms/installwsl.
PS C:\Users\baerch>
```

tunguskar, 2021-11-15:
How can I find out if it has something to do with the firewall? Deactivating or changing rules is at the moment no option.

dorssel, 2021-11-15:
@tunguskar
It's not the firewall, the error is from a very early check that happens before any network communication.
Could you run the following command on a CMD prompt (not a PowerShell) and post the output:

```
"%SystemRoot%\System32\wsl.exe" --status && echo True || echo False
```

tunguskar, 2021-11-17:
@dorssel where shall I put this command? CMD is clear but I get this error message:
```bash
C:\Users\xxx>"%SystemRoot%\System32\wsl.exe" --status && echo True || echo False
Ungültige Befehlszeilenoption: --status
```

Also under:
```bash
C:\Windows\System32>wsl.exe --status && echo True || echo False
Ungültige Befehlszeilenoption: --status
```

tunguskar, 2021-11-17:
In english it is: Invalid Command Line option: --status

dorssel, 2021-11-17:
Kein Problem, ich verstehe diese deutschen Fehlermeldungen.

What is amazing is that you appear to have WSL 2 working, but that your `wsl.exe` does not know about the `--status` option. This is what usbipd-win uses to detect whether WSL is installed at all.

@benmcmorran: what is the most portable way to detect WSL, given all the versions of windows (10/11, different builds), and windows feature vs. Store Preview?
https://docs.microsoft.com/en-us/windows/wsl/basic-commands suggests that `--status` should always be available.

@tunguskar: What is your exact Windows version + build number. Did you install WSL from the Windows features, or are you using the store preview?

benmcmorran, 2021-11-18:
That's a good question. I'll ask around internally to see if there are cases where `--status` is known to fail and reply back here.

tunguskar, 2021-11-18:
Windows Version: 1909 (Build 18363.1854)
I Installed the BASE Debian system via an offline installer and the Ubuntu System via the software center of our company.

dorssel, 2021-11-18:
@benmcmorran 
User reports:

>Windows Version: 1909 (Build 18363.1854)

That's older than I'm using (21H1). Could it be that `wsl.exe` did not support `--status` back then?

benmcmorran, 2021-11-22:
I got some more information on WSL detection. Apparently WSL on older versions of Windows 10 does not support the `--status` flag (although this support may be coming). The recommendation in these cases is to instead check for the presence of the “lxssmanager” or “lxss” services.

dorssel, 2021-11-22:
@benmcmorran 
How about, for all `usbipd wsl` subcommands, instead of having both `CheckWslInstalled()` (which does `wsl --status`) and later doing `WslDistributions.CreateAsync()` (which does `wsl --list --all --verbose`) we do: always run `WslDistributions.CreateAsync()`, which sets a flag `WslInstalled` in the returned struct based on the return value of `wsl --list --all --verbose`. Then we only rely on that one `wsl` invocation? Although I am not sure if `--list` returns success when WSL is actually installed without any distributions...

benmcmorran, 2021-11-23:
That makes sense to me. We need an installed distribution to do anything, so the results of `--list` are a better fit for this use case anyway.

dorssel, 2021-11-24:
@tunguskar 
@benmcmorran 
I found out that the `--status` option was introduced in Windows 10 2004 (10.0.19041). Version 1909 is actually end-of-life since last May (meaning: you are currently not receiving security updates). In fact, 2004 will be EOL next December 14. Time to upgrade.

I *will* fix this, but it's pretty low on the priority list. Don't hold your breath...

dorssel, 2021-11-24:
@tunguskar 
In the mean time, you can use `usbipd-win` without the `wsl` subcommand by using `usbipd bind` and then manually attach from within WSL with `usbip attach`. In fact, `usbipd wsl attach` is really a macro that does exactly that,

tunguskar, 2021-11-24:
@dorssel thanks for the hints

dorssel, 2021-11-30:
@tunguskar 
Could you please test the build artifact of #132, it is at https://github.com/dorssel/usbipd-win/actions/runs/1523022320.
It should resolve your WSL detection problem. Note that a lot of other stuff changed as well, and this is not ready for release, so even if it works, you may want to switch back to the released version after the test.

closed 2021-12-06 by dorssel

gyengeb, 2022-01-03:
> I tried the patch, now usbipd does not recognise my wsl2 system:
> 
> On Windows powershell without Admin rights:
> 
> Ubuntu is running: [image]
> 
> But there is no detection: [image]

Make one of the distro to default, it helped me.

gyengeb, 2022-01-03:
[image]

After the "usbipd wsl attach --busid <busid>" command I got this message and the SD card opens in Windows, but in wsl I can not see it.
Anyone has met this issue?

dorssel, 2022-01-04:
@gyengeb 

> usbip: error: open vhci_driver

This means your WSL kernel is too old and does not support USBIP yet. You need to update the kernel, see https://github.com/dorssel/usbipd-win/wiki/WSL-support.

gyengeb, 2022-01-04:
@dorssel , thanks, I was able to attach.
But I am not able to see the sd card in /dev, do you have any idea?

dorssel, 2022-01-04:
@gyengeb 
Your device type (storage) is not supported by the default WSL kernel. That only supports usb-to-serial devices. This means you will have to compile your own WSL kernel. Instructions are in the wiki.

PR #132 (2021-12-06): Improve WSL sanity checks
Fixes #77