dorssel/usbipd-win #78: usbipd: error: Do not recognise my running WSL WM

SeniorKullken, 2021-11-08:
When following this instructions: https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/
I can not get the usbipd to recognise the WSL2 VM. Even though I have a WSL WC open / running.
See my Windows Terminal [tab1] below.

Windows Terminal [tab1]/ PowerShell (with Administrator rights)

```
C:\...> usbipd wsl attach --busid 1-1
    usbipd: error: Windows Subsystem for Linux version 2 is not available. 
C:\...> 
C:\...> usbipd --version
    usbipd-win
    1.1.1+17.Branch.master.Sha.b4799323ae62a2c58c736e2c16c0cc57e08cc08f
C:\...> 
```

Please, I need some advise - On what side should I start looking?

1. Windows 11 side?
   * Windows version 21H2 / 22000.282 ?
   * WSL version 0.48.2.0 ?
     </br>
1. WSL 🐧 "Ubuntu-20.04" side?
   * Kernelversion: 5.10.60.1 ?
   * /usr/lib/linux-tools/5.4.0-77-generic ?
    </br>
1. ...or maybe due to Swedish settings in Win11. 
   * Like decimal delimiter = "," (not "." as expected)
   * ...or our strange characters  "ÅÄÖåäö" here and there 🤣 

____________
Some background/additional information.

Windows Terminal [tab2] / WSL Ubuntu-20.04
```
~$ uname -a
Linux Yoda 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
~$ 
```

Added `Defaults secure_path="/usr/lib/linux-tools/5.4.0-77-generic:/usr/local/sbin:..."` to `/etc/sudoers`

Windows Terminal[tab3] / PowerShell
```
C:\...> wsl --version --verbose  
WSL-version: 0.48.2.0 
Kernelversion: 5.10.60.1 
WSLg-version: 1.0.29 
Windows-version: 10.0.22000.282 
C:\...> 
C:\...> usbipd list
Present:
BUSID  DEVICE                                                        STATE
1-1    RP2 Boot, USB-masslagringsenhet                               Not shared  <= My target USB-device
1-4    Synaptics UWP WBDI SGX                                        Not shared
1-7    Integrated Camera                                             Not shared
1-8    Okänd USB-enhet (begäran om enhetsbeskrivning misslyckades)   Not shared
1-10   Intel(R) Wireless Bluetooth(R)                                Not shared
1-13   Okänd USB-enhet (adress kunde inte anges)                     Not shared
Persisted:
GUID                                    BUSID  DEVICE
C:\...> 
```

Information from Win11 Settings/About
```
Utgåva	        Windows 11 Pro   ⚠️  Running Win11 with Swedish settings!!
Version	        21H2
Installerad	‎2021-‎08-‎11
OS-version	22000.282
Gränssnitt	Gränssnittspaket för Windows-funktioner 1000.22000.318.0
```

benmcmorran, 2021-11-08:
Did you install WSL through the store instead of by enabling the Windows Feature? #71 improved the WSL detection logic. Can you try installing [the MSI from the latest CI build](https://github.com/dorssel/usbipd-win/actions/runs/1434867413) and see if it resolves your issue?

SeniorKullken, 2021-11-09:
@benmcmorran 
> Did you install WSL through the store instead of by enabling the Windows Feature?

Yes - I install "Windows Subsystem for Linux **Preview**" through the Microsoft Store.

❓Question: In the settings dialog box "*Activate/Deactivate Windows-features*".
  Should I activate`[x]`  the `[  ] Windows-subsystems for Linux` check-box? 
  Its is not activated today❗

> Can you try installing the MSI from the latest CI build and see if it resolves your issue?

Sure - With the latest `usbipd-win_1.1.2.msi`, the problem seems to have changed shape.
But I still not able to attach --busid😟
```
C:\...> usbipd.exe --version
    usbipd-win
    1.1.2+13.Branch.master.Sha.8a8b604de077f03064ff3583b2994c4cdde1dfb2
C:\...> wsl.exe --list
    Windows Subsystem for Linux Distributions:
    Ubuntu-20.04 (Default)
C:\...> 
```

# Outcome with the new usbipd 1.1.2+13 version
One step forward - Now I came all the way to the [sudo] login prompt.</br>
```
C:\...> usbipd.exe wsl attach --busid 1-1
    [sudo] password for sk:
    ...🥱after a long time (1 min)...
    usbip: error: tcp connect
    usbipd: error: Failed to attach device with ID '1-1'.
C:\...>  
```
But then, 🥱after a long time (1 min) it says "usbip: error: tcp connect"

dorssel, 2021-11-09:
The long stall may indicate the firewall is blocking the communication from the client to the host. I noticed that the WSL network connection is seen as "public" by the host. Could you try temporarily turning off the host firewall?

SeniorKullken, 2021-11-10:
@dorssel 🤓👍
It seems to work - You seem to be absolutely right about the firewall!

# Prerequisites
Windows Terminal [tab1]/ PowerShell (_with Administrator rights_)
```
C:\...>  usbipd list
    Present:
    BUSID  DEVICE                                                        STATE
    1-1    RP2 Boot, USB-masslagringsenhet                               Shared
    1-4    Synaptics UWP WBDI SGX                                        Not shared
    1-7    Integrated Camera                                             Not shared
    1-8    Okänd USB-enhet (begäran om enhetsbeskrivning misslyckades)   Not shared
    1-10   Intel(R) Wireless Bluetooth(R)                                Not shared
    1-13   Okänd USB-enhet (adress kunde inte anges)                     Not shared

    Persisted:
    GUID                                    BUSID  DEVICE
C:\...> 
```

Windows Terminal [tab2] / WSL Ubuntu-20.04
```
~$ lsusb
     Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
     Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
~$
```

# Changes
Deactivated my firewall on my Win11.
|  McAfee LiveSafe  |
| ----- |
|  [☑️] Active → [🟦] Active  |

# Outcome after deactivating my McAfee-firewall
Windows Terminal [tab1]/ PowerShell (_with Administrator rights_)
```
C:\...>  usbipd list
    Present:
    BUSID  DEVICE                                                        STATE
    1-1    RP2 Boot, USB-masslagringsenhet                               Attached  🎯
    1-4    Synaptics UWP WBDI SGX                                        Not shared
    1-7    Integrated Camera                                             Not shared
    1-8    Okänd USB-enhet (begäran om enhetsbeskrivning misslyckades)   Not shared
    1-10   Intel(R) Wireless Bluetooth(R)                                Not shared
    1-13   Okänd USB-enhet (adress kunde inte anges)                     Not shared

    Persisted:
    GUID                                    BUSID  DEVICE
C:\...> 
```
Windows Terminal [tab2] / WSL Ubuntu-20.04
```
~$ lsusb
     Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
     Bus 001 Device 002: ID 2e8a:0003 Raspberry Pi RP2 Boot   🎯Great - Here is my device✌️
     Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
~$
```

A good start👍
But I have to reactivate the firewall again.

dorssel, 2021-11-10:
Glad you have it working.

Usbipd-win installs a firewall rule for the default Windows firewall. Some third party firewalls honor those rules as well, but others apparently do not.

I will consider the original issue as closed, but I'm adding the documentation tag. Once the readme has been updated I will close this issue.

SeniorKullken, 2021-11-11:
@dorssel 
> The long stall may indicate the firewall is blocking the communication from the client to the host. I noticed that the WSL network connection is seen as "public" by the host. Could you try temporarily turning off the host firewall?

Please - Do you have any idea/hint how I can get around the problem with "WSL public virtual networks that is blocked by my Firewall"?
- Microsoft - WSL issue?
- Microsoft - Firewall configuration Win11? 
- McAfee - Firewall configuration issue?

dorssel, 2021-11-11:
Given the fact that you had it working with McAfee firewall disabled: that is where your issue is.
Sorry, I am not familiar with McAfee products. You will have to somehow configure it to allow port 3240 incoming TCP connections. It should be in the manual of that product, or else contact McAfee for support.

commit 55ecd498 (2021-11-12): Update README

Fixes #78
closed 2021-11-12 by dorssel