dorssel/usbipd-win #79: usbip win does not attach to devices listed

mspieth, 2021-11-09:
usbip win from cezanne 0.3.5 (latest) does not attach to a shared device.

This is due to the buffer not being padded with 0. 
The solution is to find the first 0 and get the substring to that point. 

git diff of my solution follows

```
diff --git a/UsbIpServer/ConnectedClient.cs b/UsbIpServer/ConnectedClient.cs
index 6a9fbd4..0878ed5 100644
--- a/UsbIpServer/ConnectedClient.cs
+++ b/UsbIpServer/ConnectedClient.cs
@@ -102,7 +102,13 @@ namespace UsbIpServer
             {
                 var buf = new byte[SYSFS_BUS_ID_SIZE];
                 await Stream.ReadExactlyAsync(buf, cancellationToken);
-                if (!BusId.TryParse(Encoding.UTF8.GetString(buf).TrimEnd('\0'), out busId))
+                var s = Encoding.UTF8.GetString(buf);
+                var end = s.IndexOf("\0");
+                if (end >= 0)
+                {
+                    s = s.Substring(0, end);
+                }
+                if (!BusId.TryParse(s, out busId))
                 {
                     await SendOpCodeAsync(OpCode.OP_REP_IMPORT, Status.ST_NODEV);
                     return;

```

Tested working.

benmcmorran, 2021-11-09:
@dorssel If I understand this issue correctly, it's not WSL related. It's saying that connecting this usbipd-win server to the [usbip-win client from cezanne](https://github.com/cezanne/usbip-win) does not work correctly, which would be a purely Windows to Windows scenario. @mspieth Is that correct?

dorssel, 2021-11-09:
Yep, I misplaced the label.

mspieth, 2021-11-09:
> @dorssel If I understand this issue correctly, it's not WSL related. It's saying that connecting this usbipd-win server to the [usbip-win client from cezanne](https://github.com/cezanne/usbip-win) [...]

correct. windows to windows. Tested inside a VirtualBox to Host and to another windows machine.

dorssel, 2021-11-13:
@mspieth 
I've just created PR #88. Could you please verify that this fixes the problem?
MSI is at build artifact: https://github.com/dorssel/usbipd-win/actions/runs/1455571106
Please note that the MSI is not signed, but the drivers are.

mspieth, 2021-11-13:
I have tested the installer and it appears to work. much more elegant solution.
Had to uninstall the previous install (mine) first though. I did not have to reboot so the drivers must have carried over.
The installer did not install the firewall rule though. this may be expected. I had the same issue with my msi installer.

dorssel, 2021-11-13:
Thx for testing. I'll merge the change.

About the firewall rule: did the released version 1.1.1 install the rule correctly? If you fully uninstall (verifying that everything is really gone, maybe even reboot for good measure) and then install the build artifact, the rule is not installed? That would be a regression bug... If that is the case, could you please file a new issue, since merging the change will close this issue, and it seems to be unrelated anyway.

mspieth, 2021-11-13:
I didn't check for removal of the rule when the program was uninstalled. I had to add one manually and that fixed it.
I will double check this probably on monday.
It would be nice if it could do an update install but this may be an issue due to not being signed the same way (my dev build vs yours vs release).
yes, release 1.1.1 installed the rule correctly.

closed 2021-11-13 by dorssel

e-t-l, 2022-03-04:
@mspieth Just to clarify, did a PR get merged with this USBIPD-win project so that it now can successfully connect to cezanne's Windows client? I'm trying to do the same thing

mspieth, 2022-03-04:
1. Yes it did get merged.
2. Yes. this is what I do. Good for remote jtag debugging. Good for virtualbox where the USB device doesnt work properly without extensions but you don't want to use extensions due to oracle cost issues.
3. This is not a client. Only a server. You can use any other client AFAIK barring bugs :-)

PR #88 (2021-11-13): Fix parsing of attach BusId
- do not assume NUL padding

Fixes #79
commit 9f95bb3a (2021-11-13): Fix parsing of attach BusId

- do not assume NUL padding

Fixes #79