dorssel/usbipd-win #82: Unhandled exception - usbipd wsl list

siddirp, 2021-11-10:
Can't see list of USB devices. I'm using Windows 11 insider preview

C:\Windows\System32>usbipd wsl list
Unhandled exception. System.AggregateException: One or more errors occurred. (DEVPKEY_Device_Address (3) does not match DEVPKEY_Device_LocationInfo (2))
 ---> System.NotSupportedException: DEVPKEY_Device_Address (3) does not match DEVPKEY_Device_LocationInfo (2)
   at UsbIpServer.ExportedDevice.GetDevice(SafeDeviceInfoSetHandle deviceInfoSet, SP_DEVINFO_DATA devInfoData, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 158
   at UsbIpServer.ExportedDevice.GetAll(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 239
   at UsbIpServer.Program.<>c.<<Main>b__19_18>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 387
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.<>c__DisplayClass79_0.<OnExecute>b__0()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 555

dorssel, 2021-11-12:
This is an intriguing case. Could you please do the following?

Lookup each USB device in Device Manager. Best select "View -> Devices by connection". Look at the nodes just below the USB HUB devices. I'm interested only in the direct children of the HUBs, not the devices below any "USB Composite Device".

For each such device, open the property page. At the first page it should say `Location: Port_#0008.Hub_#0005` (but with different numbers, of course). Next, open the tab "Details" and find the value for "Address". This should match the Port number, so in my example that would be `00000008`.

There must be one that has a mismatch.
(my best guess is: one of the port numbers is >= 10 and I made a hex/dec mistake...)

siddirp, 2021-11-12:
I tried to follow your instructions precisely but looking up the devices did not succeed as you described. In the first view of the Device Manager you can see the standard view. It shows the USB devices information. As a note, the one I care about is the AT91 USB to Serial Converter (COM4).
[image]

You instructed me to go to use "View -> Devices by connection, but none of the USB devices appear there.
[image]

The device properties for that USB to Serial Converter:
[image]

Details page with Address:
[image]

Lastly, I only have 3 devices connected to my computer: 
Atmel ICE -https://www.microchipdirect.com/product/search/all/ATATMEL-ICE
A USB to serial converter which I think is a USB-CDC device
A dock that I only use for charging: https://www.walmart.com/ip/Dell-Business-Thunderbolt-Dock-TB16-with-180W-Adapter/942686762?wmlspartner=wlpa&selectedSellerId=101098116&&adid=22222222227085023730&wl0=&wl1=g&wl2=c&wl3=197086867353&wl4=aud-1308651922056:pla-315896260293&wl5=9029858&wl6=&wl7=&wl8=&wl9=pla&wl10=503100124&wl11=online&wl12=942686762&veh=sem&gclid=CjwKCAiAvriMBhAuEiwA8Cs5lWsaaKkrg8U1H-iBfqmJzPoqLfowQskA1zEJT6X6tOInZRDmP1FUuhoCWnIQAvD_BwE&gclsrc=aw.ds

siddirp, 2021-11-12:
@dorssel While working through your instructions I decided to unplug all devices and test them one at a time. It turns out that everything works except for the USB Hub. When it is disconnected I can see the other devices. I'd like to use the HUB because it also lets me connect multiple displays.

dorssel, 2021-11-12:
Thanks. You are nearly there. When you select "Devices by connection", you will have to find *all* USB HUBs (usually more than one) somewhere in the tree. Then it is easy to find all devices as direct children. Note that some devices may be unexpected to you, such as SD-card readers, bluetooth, etc. They may not seem to be "physically connected" to your computer, but internally they do count as USB devices. Also, there may be HUBs connected to HUBs (even internally only). I need the devices on *all* hubs.

For example, here is my computer (it's in Dutch):

[image]

I have 6 devices:
- Samengesteld USB-apparaat (English: Composite Device), there is a webcam device below it, but that does not matter at this point
- Qualcomm Atheros AR3012 Bluetooth 4.0, with several devices below it that do not matter at this point
- 2 more Samengesteld USB-apparaat (English: Composite Device)
- USB-apparaat voor massaopslag (English: USB mass storage device)
- USB-invoerapparaat (English: USB input device)

Also note that some are connected to a root HUB, but most are connected via another "Generic HUB". You will have to find all HUBs this way.

dorssel, 2021-11-12:
> @dorssel While working through your instructions I decided to unplug all devices and test them one at a time. It turns out that everything works except for the USB Hub. When it is disconnected I can see the other devices. I'd like to use the HUB because it also lets me connect multiple displays.

Just noticed this extra bit of information. Of course, the devices of interest are the ones connected to this external hub. Please try to find it in the device tree and list all its children (with the Location and Address information as before).

siddirp, 2021-11-12:
I understand. Thank you.
Here are my devices:
Qualcomm QCA61x4A Bluetooth - Location: Port_#0007.Hub_#0001 Address: 00000007
USB Composite Device (Webcam) - Location: Port_#0005.Hub_#0001 Address: 00000005
USB Composite Device (Fingerprint Reader) - Location: Port_#0010.Hub_#0001 Address: 0000000A
USB Composite Device ( Atmel Ice) - Location: Port_#0001.Hub_#0002 Address: 00000002
USB Input Device: Location: 0039.0000.0000.001.000.000.000.000.000 Address: 00000001
[image]
Generic SuperSpeed USB Hub: Location: Port_#0001.Hub_#0001 Address: 00000001
Realtek USB GbE Family Controller #2  - Location: Port_#0002.Hub_#0002 Address: 00000003

Generic USB Hub: Location: Port_#0003.Hub_#0001  Address: 00000002
USB Composite Device (Realtek audio) - Location: Port_#0005.Hub_#0003 Address: 00000004
[image]

My full tree in case you see something I missed:
[image]
[image]
[image]
[image]

dorssel, 2021-11-13:
Thx. We're getting there!

I invoked my `mspaint` skills and stitched together your device tree:
[image]

For reference, the red arrows indicate the USB hubs. I don't need details on those.
The direct (non-hub) children of the hubs are indicated by the green arrows; there are 7.
Please provide the *exact* format of Location and Address of those 7 devices, so not the abbreviated version. For example, not just Port_#1.Hub_#1, but the full `Location: Port_#0001.Hub_#0001`. The reason is that `usbipd-win` does some strict parsing on that.

For example, you wrote:

> USB Composite Device ( Atmel Ice) - Location: Port_#1.Hub_#2 Address: 00000002

It looks wrong to me, because I expect the Address (2) to be the same as the Port (1), so this device should not work. But I think you made a typo here...

siddirp, 2021-11-15:
Here they are again, double and triple checked. Note, I am using a different monitor this time that creates COM6, but all the other devices are the same:

Qualcomm QCA61x4A Bluetooth - Location: Port_#0007.Hub_#0001 Address: 00000007
AT91 USB to Serial Converter (COM4) - Location: Port_#0001.Hub_#0001 Address: 00000001
USB Composite Device (Webcam) - Location: Port_#0005.Hub_#0001 Address: 00000005
USB Composite Device (Fingerprint Reader) - Location: Port_#0010.Hub_#0001 Address: 0000000A
USB Composite Device (USB Serial Device (COM6)) - Location: Port_#0004.Hub_#0003 Address: 00000004
USB Composite Device (External Webcam) - Location: Port_#0002.Hub_#0003 Address: 00000002
USB Composite Device ( Atmel Ice) - Location: Port_#0001.Hub_#0002 Address: 00000001

dorssel, 2021-11-15:
Can you confirm that all the 'Location: Port_#7.Hub_#1' are really `Location: Port_#0007.Hub_#0001` ?  I mean, with the extra zeros? Otherwise you would have had a different error, I believe.

If so, then all these devices are OK. All 'Address' (hexadecimal) are the same as the Port (decimal). This should not have caused the exception you reported earlier. Are you sure this configuration caused the error?

I am missing the devices from ASMedia (dock?): Realtek network controller + Realtek Audio. Could you connect those and give the same information for them?

siddirp, 2021-11-22:
I'm back to the office with the original dock that caused the problem. I copied and pasted all of the location and addresses, 0s and all.
Let me give you another look:
[image]
[image]

[image]
[image]

[image]
[image]

[image]
[image]

[image]
[image]

[image]
[image]

siddirp, 2021-11-22:
One more note. As you can see the location and address don't match on the last 2 devices. They both are under the same ASMedia USB3.0 Host Controller: 

[image]

dorssel, 2021-11-22:
Excellent! This is what is causing the problem. And now we can try to fix it. The "port number" is used in grabbing the device, and I really don't know if the driver filters on the number that is given by "Port_#" or by "Address". You are the first (and as far as I know, only) person where the two numbers don't match. I assumed they are always the same, hence the check.

I will make two quick hacks: one to use "Port_#" and one that uses "Address". Will you try them to see which is the one that works?

dorssel, 2021-11-22:
@siddirp 
Could you please test both the PR artifacts from #108 and #109?
You may have to uninstall first. Note that the MSIs are unsigned, but the drivers in them are signed.
Please test if you can bind + attach both the Realtek Network Controller and the Realtek Audio from the ASMedia dock.

silabs-JimL, 2021-11-24:
@dorssel Thanks. 

`PS C:\WINDOWS\system32> usbipd wsl list
Unhandled exception. System.AggregateException: One or more errors occurred. (DEVPKEY_Device_Address (4) does not match DEVPKEY_Device_LocationInfo (2))
 ---> System.NotSupportedException: DEVPKEY_Device_Address (4) does not match DEVPKEY_Device_LocationInfo (2)
   at UsbIpServer.ExportedDevice.GetDevice(SafeDeviceInfoSetHandle deviceInfoSet, SP_DEVINFO_DATA devInfoData, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 158
   at UsbIpServer.ExportedDevice.GetAll(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 239
   at UsbIpServer.Program.<>c.<<Main>b__20_18>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 388
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.<>c__DisplayClass79_0.<OnExecute>b__0()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 569`

According to the steps above,  seems it's caused by the realtek gigabit ethernet adapter on the dock.
[image]

I'll try the two PR.

silabs-JimL, 2021-11-24:
@dorssel May I know how to build the msi installer? I'm using vs2022 community, but the installer project shown as incompatible.

dorssel, 2021-11-24:
@silabs-JimL
@siddirp 
No need to build it yourself. Just take the artifacts as published by the PR builds:

https://github.com/dorssel/usbipd-win/actions/runs/1492177129
and
https://github.com/dorssel/usbipd-win/actions/runs/1492196602

If you do want to build yourselves (e.g. if you don't trust unsigned installers), just follow the build steps from the CI:
https://github.com/dorssel/usbipd-win/blob/master/.github/workflows/build.yml

silabs-JimL, 2021-11-25:
@dorssel Thanks. That's really convenient.
I tested with this buid https://github.com/dorssel/usbipd-win/actions/runs/1492196602
It solved my issue.
Thanks a lot.

dorssel, 2021-11-25:
@silabs-JimL 
Did you only test if listing the devices worked? Because I think both PRs will do that. Have you tried actually attaching the problematic devices? (Realtek Audio and Realtek Network) Because the problem wasn't really displaying them in the list, but actually using those devices...

silabs-JimL, 2021-11-25:
@dorssel I changed to https://github.com/dorssel/usbipd-win/actions/runs/1492177129 because I found some problem when using the build https://github.com/dorssel/usbipd-win/actions/runs/1492196602. When I plug a new USB device to the hub connected t the dock, it's not listed.
With the build 1492177129, I didn't see this issue.

For the problematic device (Realtek Audio and Realtek Network), I tried to attach it to WSL2, but seems it's not attached.
`
PS C:\WINDOWS\system32> usbipd wsl attach --busid 2-2
[sudo] password for jim:
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-3    USB Input Device                                              Not attached
1-6    Realtek DMFT - IR, USB Video Device, WinUsb Device            Not attached
1-10   Intel(R) Wireless Bluetooth(R)                                Not attached
2-2    Realtek USB GbE Family Controller                             Not attached
3-5    Realtek USB Audio                                             Not attached
7-3    USB Input Device                                              Not attached
8-1    J-Link driver, JLink CDC UART Port                            Not attached
PS C:\WINDOWS\system32>
PS C:\WINDOWS\system32> usbipd wsl attach --busid 2-2
[sudo] password for jim:
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-3    USB Input Device                                              Not attached
1-6    Realtek DMFT - IR, USB Video Device, WinUsb Device            Not attached
1-10   Intel(R) Wireless Bluetooth(R)                                Not attached
2-2    Realtek USB GbE Family Controller                             Not attached
3-5    Realtek USB Audio                                             Not attached
7-3    USB Input Device                                              Not attached
8-1    J-Link driver, JLink CDC UART Port                            Not attached
`

I'm not sure where is the problem, but actually I don't need this device on the client side.

dorssel, 2021-11-25:
@silabs-JimL 
Thanks for testing! I am glad that you have your device working.

But I also want to solve the real underlying problem: the mismatch between Address and Port_#. That's why I also want those dock-devices to work. You may not need them, but other users may. Are you willing to test some more?

You say you had problems with  https://github.com/dorssel/usbipd-win/actions/runs/1492196602. Can you describe those problems?

And during the (failed) attach of 2-2 using the other build https://github.com/dorssel/usbipd-win/actions/runs/1492177129: can you provide the logs from `usbipd-win` as described in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting?

silabs-JimL, 2021-11-25:
@dorssel Sure. I will test it. Will reply to you later.

silabs-JimL, 2021-11-25:
Below is the log when I attach the realtek ethernet adapter on the dock.

```
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-3    USB Input Device                                              Not attached
1-6    Realtek DMFT - IR, USB Video Device, WinUsb Device            Not attached
1-10   Intel(R) Wireless Bluetooth(R)                                Not attached
2-2    Realtek USB GbE Family Controller                             Not attached
3-5    Realtek USB Audio                                             Not attached
7-3    USB Input Device                                              Not attached
8-1    J-Link driver, JLink CDC UART Port                            Not attached
PS C:\WINDOWS\system32> usbipd wsl attach --busid 2-2
[sudo] password for jim:
```

```
PS C:\WINDOWS\system32>
PS C:\WINDOWS\system32> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.22.49.99
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.22.49.99 claimed device at 2-2 (USB\VID_0BDA&PID_8153\000002000000).
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=1, flags=512, length=8, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 8
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=1, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 8, requested: 8
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=2, flags=0, length=0, ep=0
trce: UsbIpServer.AttachedClient[1001]
      0 49 40 0 0
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=2, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=3, flags=512, length=18, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 18
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=3, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 18
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=4, flags=512, length=5, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 3840 0 5
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=4, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 5, requested: 5
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=5, flags=512, length=22, ep=0
trce: UsbIpServer.AttachedClient[1001]
[... 60 lines cut ...]
      actual: 16, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=13, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 774 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=13, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 26, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=14, flags=0, length=0, ep=0
dbug: UsbIpServer.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 2
info: UsbIpServer.ConnectedClient[2]
      Client 172.22.49.99 released device at 2-2 (USB\VID_0BDA&PID_8153\000002000000).
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values. (Parameter 'configurationValue')
         at UsbIpServer.UsbConfigurationDescriptors.SetConfiguration(Byte configurationValue) in D:\a\usbipd-win\usbipd-win\UsbIpServer\UsbConfigurationDescriptors.cs:line 156
         at UsbIpServer.AttachedClient.HandleSubmitAsync(UsbIpHeaderBasic basic, UsbIpHeaderCmdSubmit submit, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\AttachedClient.cs:line 292
         at UsbIpServer.AttachedClient.RunAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\AttachedClient.cs:line 477
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 184
dbug: UsbIpServer.Server[1000]
      connection closed
```

dorssel, 2021-11-25:
@silabs-JimL 
Thanks! This definitively determines that the Port_# is the only number of interest, and that Address is to be ignored altogether.
Not only does the device show up in the list (as 2-2, instead of 2-4 if Address is used), but it is also claimable by filtering on port "2" (and not "4")! I will fix this. Thanks again, this information has been very useful!

Now we found a new issue:

> SET_CONFIGURATION: 2

I've had some problems in the past with devices that have multiple configurations. But this is the first time I see a device that actually is configured to use anything else than configuration 1. And (of course...) it fails. It is a totally unrelated issue, however, so I will move this to a new issue. I hope you are willing to help out with this one as well?

silabs-JimL, 2021-11-25:
@dorssel Sure. Please ping me on the new issue.

andrewloz, 2021-11-27:
1492177129 fixed The same exception for me.  Using a Dell dock with attached devices.  At least I can list the devices now, have yet to try attaching them.

andrewloz, 2021-11-27:
But I still cannot attach a device.  Bearing in mind I'm not usually a Windows user, so am likely to be making rookie mistakes.  Here's the transcript if it helps:
PS C:\Windows\system32> usbipd wsl list
                                                                                 BUSID  DEVICE                                                        STATE                                              
2-2    Realtek USB GbE Family Controller                             Not attached                                       
2-5    USB Video Device                                              Not attached                                       
2-9    Goodix Moc Fingerprint                                        Not attached                                       
2-14   Intel(R) Wireless Bluetooth(R)                                Not attached                                       
3-4    USB Attached SCSI (UAS) Mass Storage Device                   Not attached                                       
3-5    Realtek USB Audio                                             Not attached                                       
5-2    Logitech USB Input Device, USB Input Device                   Not attached                                       
PS C:\Windows\system32> usbipd wsl attach --busid 3-4                                                                   
[sudo] password for xxxx:                                                                                            
usbip: error: tcp connect                                                                                               
usbipd: error: Failed to attach device with ID '3-4'.

andrewloz, 2021-11-27:
1492196602 results in an error when listing:
Unhandled exception. System.AggregateException: One or more errors occurred. (An item with the same key has already been added. Key: [3-4, UsbIpServer.ExportedDevice])
Hmm... bus id 3-4 again!  This device is connected to the dock.

dorssel, 2021-11-28:
@andrewloz 
Could you please test
https://github.com/dorssel/usbipd-win/actions/runs/1511319336

While you attach, run the server with full logging as in
https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

Please post the entire log. Thanks!

andrewloz, 2021-11-29:
Stopped the existing service and ran from powershell like so:
 .\usbipd.exe server Logging:LogLevel:Default=Trace                                      
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]                                                                
           Hosting starting                                                                                                  
info: Microsoft.Hosting.Lifetime[0]                                                                                 
          Application started. Press Ctrl+C to shut down.                                                                  
 info: Microsoft.Hosting.Lifetime[0]                                                                                
           Hosting environment: Production                                                                                   
info: Microsoft.Hosting.Lifetime[0]                                                                                
           Content root path: C:\Program Files\usbipd-win                                                                    
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]                                                                
           Hosting started                 
From another PS window did as attached below:
[image]
And it failed to attach as you will see.  There were no extra logs from the server when attaching.  Any firewall rules required, maybe?

As previously indicated my windows experience is very limited, so I may be missing something obvious.

andrewloz, 2021-11-29:
The above was using the indicated MSI 1511319336 in your previous comment

dorssel, 2021-11-29:
@andrewloz 
Yes, this looks like a firewall issue. Are you using a 3rd party firewall? For the default Windows firewall, an exception for TCP port 3240 is created by the installer.

andrewloz, 2021-11-29:
Standard firewall so far as I know, it's a new Dell Inspiron 16+

andrewloz, 2021-11-29:
But perhaps there is another one, checking now.

dorssel, 2021-11-29:
@andrewloz 
Did you uninstall your previous installation of `usbipd-win` and then installed the PR? Maybe Windows installer got confused and messed up the firewall rule. You can try a "repair" of `usbipd-win` by running the installer again (the same that is already installed). It should put the firewall rule back in place.

andrewloz, 2021-11-29:
Opened tcp port 3240 in mcAfee firewall - I hate 3rd party hidden apps!  When I tried after that I got a ping, and now the device is listed as attached to wsl.  usbip list --remote=172.31.240.1 on wsl now shows 
Exportable USB devices
======================
 - 172.31.240.1
        4-4: LaCie, Ltd : unknown product (059f:1094)
           : USB\VID_059F&PID_1094\MSFT300000NL35HYST
           : (Defined at Interface level) (00/00/00)
           :  0 - Mass Storage / SCSI / Bulk-Only (08/06/50)
Which is promising!  Any idea how I access the disk on wsl?

andrewloz, 2021-11-29:
Ahh, I see that wsl does not have any udev rules set up.  Will have to work on that.

andrewloz, 2021-11-29:
But it looks like your part of the solution is working now, thanks very much for the assistance.

dorssel, 2021-11-29:
@andrewloz 
Glad you can now attach your device. As far as I know, for mass storage devices under WSL the problem is that the default WSL kernel doesn't have drivers for them. You may have to compile your own, see https://github.com/dorssel/usbipd-win/wiki/WSL-support.

The original issue reported here will be closed once #120 has been merged (this includes the fix from the testing PR #108).

dorssel, 2021-11-29:
Fixed by #119

closed 2021-11-29 by dorssel

siddirp, 2021-12-09:
> @silabs-JimL @siddirp No need to build it yourself. Just take the artifacts as published by the PR builds:
>
> https://github.com/dorssel/usbipd-win/actions/runs/1492177129 and [...]

Sorry for the late reply. I just tested the first artifact (ttps://github.com/dorssel/usbipd-win/actions/runs/1492177129) and it worked! I didn't try the second. I tried to install it on top of the first (ran the .msi of the second after the first) and it said it was already installed.

dorssel, 2021-12-09:
Glad you have it working.
By now, version 1.3.0 has been released. I guess you want to update to a released version. It contains this fix.