dorssel/usbipd-win #83: Exception on usbipd wsl list

RafaelGCPP, 2021-11-10:
When I issue `usbipd wsl list`, I get an exception:

    ❯ usbipd list
    Unhandled exception. System.AggregateException: One or more errors occurred. (An item with the same key has already been added. Key: [1, UsbIpServer.UsbConfigurationDescriptors+UsbConfiguration])
     ---> System.ArgumentException: An item with the same key has already been added. Key: [1, UsbIpServer.UsbConfigurationDescriptors+UsbConfiguration]
       at System.Collections.Generic.TreeSet`1.AddIfNotPresent(T item)
       at System.Collections.Generic.SortedSet`1.Add(T item)
       at System.Collections.Generic.SortedDictionary`2.Add(TKey key, TValue value)
       at UsbIpServer.UsbConfigurationDescriptors.AddDescriptor(ReadOnlySpan`1 descriptor) in D:\a\usbipd-win\usbipd-win\UsbIpServer\UsbConfigurationDescriptors.cs:line 96
       at UsbIpServer.ExportedDevice.GetConfigurationDescriptor(DeviceFile hub, UInt16 connectionIndex, Byte numConfigurations)
       at UsbIpServer.ExportedDevice.GetDevice(SafeDeviceInfoSetHandle deviceInfoSet, SP_DEVINFO_DATA devInfoData, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 202
       at UsbIpServer.ExportedDevice.GetAll(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 239
       at UsbIpServer.Program.<>c.<<Main>b__19_9>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 234
       --- End of inner exception stack trace ---
       at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
       at System.Threading.Tasks.Task`1.get_Result()
       at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.<>c__DisplayClass79_0.<OnExecute>b__0()
       at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
       at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 555

Running Windows 11 Pro Insider Preview
Build 22494

benmcmorran, 2021-11-11:
This is likely a duplicate of #54, which was fixed by #73. The fix has not yet been released. Can you try installing [the MSI from the latest CI build](https://github.com/dorssel/usbipd-win/actions/runs/1434867413) and see if it resolves your issue?

RafaelGCPP, 2021-11-13:
The MSI from the CI is a fix indeed! I believe you can close as duplicate. Here's the output:

    BUSID  DEVICE                                                        STATE
    1-4    USB Audio Device, USB Video Device                            Not attached
    1-9                                                                  Not attached
    1-12   Intel(R) Wireless Bluetooth(R)                                Not attached
    2-1    USB Input Device                                              Not attached
    3-1    Silicon Labs CP210x USB to UART Bridge                        Not attached
    3-4    USB Audio Device, USB Input Device                            Not attached

I was using the 1.1.1 release MSI.

dorssel, 2021-11-13:
Duplicate #54 

See also #69

closed 2021-11-13 by dorssel