dorssel/usbipd-win #84: IP address issue

willzyx-hub, 2021-11-11:
Hello guys when i use usbipd i got an error which says this
C:\Windows\system32>usbipd wsl attach --busid 5-2
usbipd: error: The local IP address for the WSL virtual switch could not be found. Ensure WSL is running. 

does anyone know how to fix that ??? any support will be appreciated

dorssel, 2021-11-11:
WSL uses a virtual switch to communicate between the host and the WSL instances. This switch is "up" only when at least one instance is running. Just open a shell on the instance and the error should be gone.

If the instance was actually running and this error still occurs, then you've found a bug. Please report back if this is the case.

willzyx-hub, 2021-11-11:
> WSL uses a virtual switch to communicate between the host and the WSL instances. This switch is "up" only when at least one instance is running. Just open a shell on the instance and the error [...]

Just run a Windows Powershell and repeat it again ?

sigprof, 2021-11-11:
What distro are you using inside WSL? Apparently `usbipd` tries to run `hostname --all-ip-addresses` in the WSL instance to determine the IP address used by that instance:
https://github.com/dorssel/usbipd-win/blob/1b9d9a3cf53b01d7195f1a82fc26aedf95d1ef8c/UsbIpServer/WslDistributions.cs#L120
However, the package containing the `hostname` utility might not be installed (e.g., Arch Linux has that tool in the `inetutils` package), and, even worse, not all versions of `hostname` support the `--all-ip-addresses` option. E.g., the `hostname` utility in Debian or Ubuntu distros supports `--all-ip-addresses`, but the Arch Linux version (which comes from the GNU inetutils package, so it's a completely different implementation) does not support that option.

I'm not sure what could be a better option to get the list of IP addresses from the WSL instance (one possible option is `ip -j address list`, but the corresponding `iproute2` package might not be installed too, and I'm not sure how common is the `-j` option support; although parsing the `ip -o address list` output could also be an option).

dorssel, 2021-11-11:
The error is even before that. `usbipd-win` first tries to get the host IP address to the WSL instances. It does that by finding the IPv4 address of the switch. Please list the names of your interfaces, for example:

```
C:\>netsh interface show interface

Admin State    State          Type             Interface Name
-------------------------------------------------------------------------
Enabled        Connected      Dedicated        Ethernet
Enabled        Connected      Dedicated        vEthernet (External Switch)
Enabled        Connected      Dedicated        vEthernet (Internal Switch)
Enabled        Disconnected   Dedicated        Wi-Fi
Enabled        Connected      Dedicated        vEthernet (Default Switch)
Enabled        Connected      Dedicated        vEthernet (WSL)
```

`usbipd-win` looks for one that has `WSL` in the name. It is there only if at least one WSL instance is r.unning

sigprof, 2021-11-11:
Sorry, I misread the error message; I reported my issue separately.

willzyx-hub, 2021-11-11:
> The error is even before that. `usbipd-win` first tries to get the host IP address to the WSL instances. It does that by finding the IPv4 address of the switch. Please list the names of your [...]

here with wsl running

C:\Windows\system32>netsh interface show interface

Admin State    State          Type             Interface Name
-------------------------------------------------------------------------
Enabled        Disconnected   Dedicated        Ethernet
Enabled        Connected      Dedicated        Wi-Fi

dorssel, 2021-11-12:
Are you absolutely sure your WSL *instance* is running. I mean: have you started a Linux prompt? Because I don't understand how you could possibly be interacting with the WSL instance without having a virtual network connection to it...

willzyx-hub, 2021-11-12:
> Are you absolutely sure your WSL _instance_ is running. I mean: have you started a Linux prompt? Because I don't understand how you could possibly be interacting with the WSL instance without having a virtual network connection to it...

Yes the wsl (im using ubuntu) it's running in background and it's ready to be used.

dorssel, 2021-11-12:
@benmcmorran 
Ok, I'm out of ideas. Ben?

willzyx-hub, 2021-11-12:
> @benmcmorran Ok, I'm out of ideas. Ben?

Is anything wrong in my Windows ? Or something like that ?

benmcmorran, 2021-11-13:
I haven't seen this before. I have a few questions that may help troubleshoot:

- What is the output of running `wsl -l -v` from Windows?
- What is the output of running `uname -a` from WSL?
- Are you able to access the internet from within WSL? For example does `ping microsoft.com` succeed from WSL?

willzyx-hub, 2021-11-13:
> I haven't seen this before. I have a few questions that may help troubleshoot:
>
> * What is the output of running `wsl -l -v` from Windows?
> * What is the output of running `uname -a` from WSL?
> * Are [...]

sorry for late respond so this is the output

from windows cmd

willi@willy MINGW64 ~
$ wsl -l -v
  NAME      STATE           VERSION
* Ubuntu    Running         1

from wsl 

willy@willy:~$ uname -a
Linux willy 4.4.0-22000-Microsoft #1-Microsoft Fri Jun 04 16:28:00 PST 2021 x86_64 x86_64 x86_64 GNU/Linux
willy@willy:~$ ping microsoft.com
PING microsoft.com (104.215.148.63) 56(84) bytes of data.

willzyx-hub, 2021-11-13:
@benmcmorran what's wrong ?

dorssel, 2021-11-13:
Yes, now it totally makes sense!

Your WSL instance is version 1, but `usbipd-win` requires WSL version 2. We have some detection code for that, but apparently that did not trigger so the error message you got was not very helpful. I will keep this issue open to correct that. It was always intended that the following error was displayed for this case:

```
ReportError($"The specified WSL distribution is using WSL 1, but WSL 2 is required. Learn how to upgrade at {InstallWslUrl}.");
```

It turns out Microsoft has changed the documentation, so the URL that is inserted above isn't even the right one.

The good news for you is: follow the upgrade instructions on: https://docs.microsoft.com/en-us/windows/wsl/basic-commands#set-wsl-version-to-1-or-2 (this *is* the right URL, at least at the time of this writing).

willzyx-hub, 2021-11-13:
> Yes, now it totally makes sense!
>
> Your WSL instance is version 1, but `usbipd-win` requires WSL version 2. We have some detection code for that, but apparently that did not trigger so the error [...]

i told you i already did that now the error is

usbip: error: Attach Request for 3-3 failed - Request Failed

usbipd: error: Failed to attach device with ID '3-3'.

dorssel, 2021-11-13:
You wrote:

> willi@willy MINGW64 ~
> $ wsl -l -v
> NAME STATE VERSION
> Ubuntu Running 1

There it says version is 1. As long as that says version 1, `usbipd-win` is not going to work. Updating from 1 to 2 has to do with WSL, not with `usbipd-win`. All `usbipd-win` can do for version 1 is provide a better error message.

You also write:

> willy@willy:~$ uname -a
> Linux willy 4.4.0-22000-Microsoft #1-Microsoft Fri Jun 04 16:28:00 PST 2021 x86_64 x86_64 x86_64 GNU/Linux

That is indeed the kernel version WSL 1 will report, so it confirms the WSL instance is in fact version 1. So it isn't the case that the version is reported incorrectly. You are, most definitely, running WSL version 1.

willzyx-hub, 2021-11-13:
> You wrote:
>
>
> There it says version is 1. As long as that says version 1, `usbipd-win` is not going to work. Updating from 1 to 2 has to do with WSL, not with `usbipd-win`. All `usbipd-win` can do [...]

Thats old one my WSL is 2 now i updated it recently .... so its not 1 anymore 

and my current problem is 
C:\Windows\system32>usbipd wsl attach -b 3-3
[sudo] password for willy:
usbip: error: Attach Request for 3-3 failed - Request Failed

usbipd: error: Failed to attach device with ID '3-3'.

dorssel, 2021-11-13:
OK, that's progress. Now it is time to get the debug log from the `usbipd-win` service. Please follow the instructions on: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting. That will hopefully provide the cause.

Note to self: change the order of checking the WSL2 switch address and the WSL 1 check.

willzyx-hub, 2021-11-14:
here my log

C:\Windows\system32>usbipd server
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Windows\system32
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client: System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 144
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 155
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 161
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 158

dorssel, 2021-11-14:
Unfortunately, you're now getting the same problem some other people have: `System.IO.FileNotFoundException` #63. 
The original issue you reported had to do with WSL 1 versus 2. I will solve the original issue with better detection and error reporting. Once that is done, I will close this issue. Please monitor #63 for progress on your current problem.

closed 2021-11-15 by dorssel

fnarenji, 2023-03-05:
Ran into a similar issue, after configuring WSL2 in bridge mode. Ensuring the Hyper-V bridge mode switch was named "WSL" instead of whatever I had named it during creation fixed the issue.

PR #92 (2021-11-15): Improve WSL checks
Fixes #84
commit a429c29a (2021-11-15): Improve WSL checks

Fixes #84