dorssel/usbipd-win #85: `usbipd wsl attach` works only with the Debian implementation of `hostname`

sigprof, 2021-11-11:
Whe the `usbipd wsl attach` command is used, `usbipd` tries to run `hostname --all-ip-addresses` in the WSL instance to determine the IP address used by that instance:
https://github.com/dorssel/usbipd-win/blob/1b9d9a3cf53b01d7195f1a82fc26aedf95d1ef8c/UsbIpServer/WslDistributions.cs#L120
However, apparently there are multiple different implementations of the `hostname` utility; the Debian one (which is present in Debian and Ubuntu distros) supports the `--all-ip-addresses` option, but the GNU inetutils implementation (e.g., `inetutils 2.2-1` which is used by Arch Linux) does not, therefore `usbipd wsl attach` fails when the Arch Linux instance is targeted:
```
PS C:\WINDOWS\system32> usbipd wsl attach --busid 4-1 -d Arch
usbipd: error: The specified WSL distribution cannot be reached via the WSL virtual switch; try restarting the WSL distribution.
```
An inconvenient workaround is manually running `usbipd bind` on the Windows side and then `usbip attach` on the Linux side.

I'm not sure what would be the best way to get the list of IP addresses from the WSL instance. One possible option is `ip -j address list`, but the corresponding `iproute2` package might not be installed too, and I'm not sure how common is the `-j` option support (although parsing the `ip -o address list` output could also be an option — `-o` should be supported by all existing versions of `iproute2`).

dorssel, 2021-11-12:
I don't have Arch Linux available right now. Could you test:

```
awk '/32 host/ { print f } {f=$2}' /proc/net/fib_trie | sort | uniq | grep -v 127.0.0.1
```

Does that work?

dorssel, 2021-11-12:
Or just:

```
awk '/32 host/ { print f } {f=$2}' /proc/net/fib_trie
```

We could filter out the duplicates later. Only depends on `awk`. And any distribution without that should be burned...

dorssel, 2021-11-12:
Just created #87, could you give it a try?
Build artifact is as https://github.com/dorssel/usbipd-win/actions/runs/1453661276.
MSI is unsigned, but drivers are.

sigprof, 2021-11-13:
MSI from #87 works for me — I can attach the USB device with `-d Arch` (and attaching to a WSL instance running Debian 10 with the `usbip` package installed still works too).

BTW, another workaround was to attach the device to a supported WSL instance — because all WSL instances actually run as containers in the same VM, the USB device can be accessed from any of them, not just the one that was specified in `usbipd wsl attach`. And the `/dev` filesystem seems to be shared between all those instances (any permission changes done in one instance get propagated to others).

dorssel, 2021-11-13:
Thx for testing. I will merge the change which will close this issue.

And I learned something today as well. Never knew that multiple instances of WSL2 are actually different containers running in a single WSL VM. Thx for sharing that too!

PR #87 (2021-11-13): Improve WSL instance address detection
- no longer rely on 'hostname', which is distribution dependent

Fixes #85
commit 642e3b27 (2021-11-13): Improve WSL instance address detection

- no longer rely on 'hostname', which is distribution dependent

Fixes #85
closed 2021-11-13 by dorssel