dorssel/usbipd-win #91: Mass storage devices not safely closed on Windows

akuropka, 2021-11-13:
Using `usbipd wsl attach` produces an error upon mounting in Linux:

> The disk contains an unclean file system (0, 1).
> The file system wasn't safely closed on Windows. Fixing.

It seems the hardware is not safely removed under Windows but virtually just "plugged out".
It may be a source for file system errors.

**Tested on:**
Microsoft Windows [Version 10.0.22000.346]
WSL Ubuntu 5.10.74.3-microsoft-standard-WSL2+

dorssel, 2021-11-13:
Thanks for the report.

I actually never thought about the option of "safe removal" on the host side. A quick test shows it is indeed possible. If I safely remove a flash drive manually, I can still attach it to a client.

I haven't investigated on exactly how to do this. Probably we would also need to fail the attach if a safe removal fails (for example, if the device is still in use). And then also add a `attach --force` option to get the current behavior: surprise removal.

The other way around ("safe detach") will be more of a problem. But the `unbind` command already states that it acts as a surprise removal on the client side.

All in all, I will mark this as both 'documentation' (for now, to better inform users of the current behavior) and as 'enhancement' (for later). Thanks for the suggestion!

dorssel, 2021-12-08:
@akuropka 
Could you please test the build artifact from PR #153, it is at https://github.com/dorssel/usbipd-win/actions/runs/1554111229.
Does this properly resolve your feature request?
Thanks!

PR #153 (2021-12-10): Refactor attach
- Use restart instead of disable/enable
- Fix unplug detect
- Fix cleanup on startup for non-present devices

Fixes #152 
Fixed #155 
Resolves #91
closed 2021-12-10 by dorssel