dorssel/usbipd-win #94: Can't attach any devices to WSL - "Request Completed Successfully" & "Unknown Op Common Status"

yusufkandemir, 2021-11-14:
## The Problem
```
$ usbipd wsl list
BUSID  DEVICE                                                        STATE
1-5    USB Giriş Aygıtı, WinUsb Aygıtı                               Not attached
2-2    Logitech USB Input Device, USB Giriş Aygıtı                   Not attached
2-4    USB Giriş Aygıtı, WinUsb Aygıtı                               Not attached
6-2    USB Giriş Aygıtı                                              Not attached
6-3    USB Giriş Aygıtı, USB Ses Aygıtı                              Not attached
7-1    HD Pro Webcam C920, USB Ses Aygıtı                            Not attached
7-2    Microsoft Usbccid Akıllı Kart Okuyucusu (WUDF), USB Giriş...  Not attached
7-3    USB Giriş Aygıtı                                              Not attached
8-4    Billboard Cihazı                                              Not attached

$ usbipd wsl attach --busid 2-2
[sudo] password for yusuf:
usbip: error: Attach Request for 2-2 failed - Request Completed Successfully

usbipd: error: Failed to attach device with ID '2-2'.

$ usbipd wsl attach --busid 7-2
[sudo] password for yusuf:
usbip: error: Attach Request for 7-2 failed - Unknown Op Common Status

usbipd: error: Failed to attach device with ID '7-2'.

$ usbipd wsl attach --busid 2-2
[sudo] password for yusuf:
usbip: error: Attach Request for 2-2 failed - Request Completed Successfully

usbipd: error: Failed to attach device with ID '2-2'.

$ usbipd wsl attach --busid 7-2
[sudo] password for yusuf:
usbip: error: Attach Request for 7-2 failed - Request Completed Successfully

usbipd: error: Failed to attach device with ID '7-2'.
```
I can't attach any devices to WSL. I am getting `Unknown Op Common Status` and `Request Completed Successfully`(_doesn't sound like an error but 🤷_) errors. I couldn't figure out why a certain error is thrown under which circumstances. It seems random. For consecutive 13 tries, I got 4 consecutive `Request Completed Successfully` errors, then 4 consecutive `Unknown Op Common Status` errors, then 5 more `Request Completed Successfully` errors.

I tried attaching different devices. I tried attaching a certain device with and without a USB Hub, on different port locations and versions(USB 2, USB 3, etc.). I tried using various releases including `1.1.2` and some still unreleased draft versions such as `1.1.2+27.Branch.master.Sha.e77e634bea91aba546648ab32ae1ce59df831809`. But, no luck 😞
When executing the attach command I don't hear any device disconnect/connect beep sounds, and the device doesn't disappear briefly on the device manager.
(_I saw some of these troubleshooting steps on https://github.com/dorssel/usbipd-win/issues/63#issuecomment-962284719_)

## Device Information
| Bus ID | Friendly Name              | Hardware ID                          |
|--------|----------------------------|--------------------------------------|
| 2-2    | Logitech Unifying Receiver | USB\VID_046D&PID_C52B&REV_2411&MI_00 |
| 7-2    | Yubico Yubikey 5 NFC       | USB\VID_1050&PID_0407&REV_0543&MI_02 |

## System Information
Microsoft Windows 11 Education 21H2 (Build 22000.318)
Ubuntu 20.04.3 LTS on Windows 10 x86_64 (Kernel 5.10.60.1-microsoft-standard-WSL2)
Windows Subsystem for Linux Preview from Microsoft Store

dorssel, 2021-11-14:
Please provide the log of the server following: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

yusufkandemir, 2021-11-15:
Oops, here it is:
```
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.24.181.240
fail: UsbIpServer.RegistryWatcher[4]
      An internal error occurred: Failed to start RegistryWatcher
      System.Management.ManagementException: Not found
         at System.Management.ManagementException.ThrowWithExtendedInfo(ManagementStatus errorCode)
         at System.Management.ManagementEventWatcher.Start()
         at UsbIpServer.RegistryWatcher..ctor(ILogger`1 logger) in D:\a\usbipd-win\usbipd-win\UsbIpServer\RegistryWatcher.cs:line 36
dbug: UsbIpServer.Server[1000]
      connection closed
dbug: UsbIpServer.Server[1000]
      new connection from 172.24.181.240
fail: UsbIpServer.RegistryWatcher[4]
      An internal error occurred: Failed to start RegistryWatcher
      System.Management.ManagementException: Not found
         at System.Management.ManagementException.ThrowWithExtendedInfo(ManagementStatus errorCode)
         at System.Management.ManagementEventWatcher.Start()
         at UsbIpServer.RegistryWatcher..ctor(ILogger`1 logger) in D:\a\usbipd-win\usbipd-win\UsbIpServer\RegistryWatcher.cs:line 36
dbug: UsbIpServer.Server[1000]
      connection closed
```

It looks like it's the same problem as #80. I had the exact same situation explained in https://github.com/dorssel/usbipd-win/issues/80#issuecomment-968190790, I manually terminated the x86 process and it worked, will look for a permanent solution now. So, closing this one. Thanks for all your hard work 💯

A small note: the server logged the same messages for both "Request Completed Successfully" & "Unknown Op Common Status", errors, so there might be a little room for logging improvements.

closed 2021-11-15 by yusufkandemir