dorssel/usbipd-win #96: Segger J-Link issue

DanieleNardi, 2021-11-15:
I'm trying to share my Segger J-Link over USB to my WSL2 instance, but no success right now.  The device appears in lsusb list. JLinkGDBServerCL just replies that the connection to J-Link failed then it asks me `Connected correctly?`.

I'm trying to debug on-chip application from WSL2 using Visual Studio Code.

Environment settings:
Host: Windows 10 Pro 21H1
WSL2 instance: Ubuntu 20.04 LST (5.10.60.1-microsoft-standard-WSL2)
J-link version 7.58a installed in WSL2 instance above.
arm toolchain: arm-none-eabi-8-2018-q4-major

VSCode launch.json configuration:
```
"name": "DVC TopRow Emerald Inventory",
"type": "cortex-debug",
"request": "launch",
"cwd": "${workspaceFolder}",
"armToolchainPath": "/opt/gcc-arm-none-eabi/bin",
"executable": "path/to/elf",
"serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
"servertype": "jlink",
"device": "MK10DX256xxx7",
"interface": "jtag",
"serialNumber": "proper serial number",
"runToMain": true,
"svdFile": "path/to/svd"
```

dorssel, 2021-11-15:
Please provide the log of the server following: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

DanieleNardi, 2021-11-15:
> Please provide the log of the server following: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

```
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Windows\system32
info: UsbIpServer.ConnectedClient[1]
      Client 172.20.52.229 claimed device at 4-3 (USB\VID_1366&PID_0101\000050118073).
```
I get only these ones.

dorssel, 2021-11-15:
Please run with `usbipd server Logging:LogLevel:Default=Trace`

DanieleNardi, 2021-11-15:
After I updated J-Link drivers on host side (j-link drivers now have same version), the connection worked, but the VSCode debugger still doesn't work.
Log in the attached file.
[usbipdlog.txt](https://github.com/dorssel/usbipd-win/files/7539121/usbipdlog.txt)

dorssel, 2021-11-15:
There are no errors in that log. This means that `usbipd-win` was able to forward all USB requests without issue and there were more than 3000 requests. I think the problem is client side. Have a look if `/var/log/syslog` contains any information; you may have to install `rsyslog` first.

RoboGnome, 2021-12-13:
Hi @DanieleNardi,

I am currently working on it and I can debug with JLink + cortex-debug from within VSCode.
Did you copy the udev rules?
  `sudocp /opt/SEGGER/JLink/99-jlink.rules /etc/udev/rules.d/99-jlink.rules` 

I also have some problems with the udev service under WSL2+Ubuntu20.04.
After every reboot I need to restart the udev service with:
 ```
 sudo service udev restart
 sudo udevadm control --reload
```

Also this only works if you didn't already attach the USB device to WSL2. So you need to first restart the udev service and then attach the USB device. At least on my setup I have to do it.
But after that it works from within VSCode until you reboot the system, which is really cool.

DanieleNardi, 2021-12-14:
Very good to know!
Now I'm committed to another project, as soon as I can switch I'll try to get this thing working also for me.

Thank you!

pearson, 2022-02-02:
@RoboGnome :
> I also have some problems with the udev service under WSL2+Ubuntu20.04. After every reboot I need to restart the udev service with:
> 
> ```
> sudo service udev restart
> sudo udevadm control --reload
> ```

Thanks for this! I did not need to copy udev rules, but these two commands got my JLink working.

dorssel, 2022-02-11:
WSL documentation has been updated with udev details. Closing.

closed 2022-02-11 by dorssel