dorssel/usbipd-win #97: Yubikey 5 and gpg works only as root

J0F3, 2021-11-15:
Hi, 

First of all I am very fascinated of the project it awesome and gives the WSL one of the most missing capabilities. So thanks to all involved for all the work already done so far!

I was hopping to use my Yubikey 5  (which I saw is on the list of compatible devices) to gpg sign my git commits in WSL. 
Attaching the Yubikey to WSL worked fine and flawless but when I run `gpg --card-status` in WSL I just get:
```
~$ gpg --card-status
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```

And in the log files of the `scdaemon` I can see the following error:
```
scdaemon[267] pcsc_establish_context failed: no service (0x8010001d)
```

However when I run run the command as root with `sudo gpg --card-status`  I works and gpg can see my Yubikey. 

So I am happy for any ideas or inputs why gpg can see the Yubkikey only when running as root but not as normal user.

Thank!

dorssel, 2021-11-15:
This is not something that can be fixed by `usbipd-win`. This is a problem of the client side OS and/or software.

A quick search on Google gave me: https://github.com/clearlinux/distribution/issues/490. It mentions `udev` settings and the problem appears to be distribution dependent. For example, I use it on a regular Hyper-V VM with Ubuntu 20.04: no problems. I've seen another report here where someone successfully used Yubikey with WSL 2. I think it is related to your specific distribution and/or client software.

benmcmorran, 2021-11-15:
I'd also take a look at the `udev` discussion in #60, especially [this comment](https://github.com/dorssel/usbipd-win/issues/60#issuecomment-962649162). There seem to be some issues with getting udev changes to persist in WSL, but restarting and reloading the service makes the rules take effect.

J0F3, 2021-11-15:
@dorssel Thanks, that what I was afraid of but thought it would be worth to ask here anyway. I am also running Ubuntu 20.04 in the WSL so not really something special here. 

@benmcmorran Thank you. Yes my problem seems to be very similar. I figured out that the problem in my case is ` pcscd`. 
But otherwise it is the same. So basically the service is not started automatically when WSL 2 starts which then leads to the problem. 

I cloud fix it with ` sudo service pcscd start`. However that is needed every time when the WSL2 is re-started.

I have then tested by adding the following to `/etc/wsl.conf` (according to https://docs.microsoft.com/en-us/windows/wsl/wsl-config#user-preview-options):
```
[boot]
command=service pcscd start
```

But this helps also not really because I have noticed that I have to restart the service after the Yubikey is attached to WSL. Only then gpg can access it without root permissions. 

So probably it is for `udev ` the same or at least very similar.

dorssel, 2021-11-17:
Like with many other devices, this looks to be a WSL-only issue w.r.t. `udev`.
Have a look at how others are restarting udev to get things going.
I am marking this as documentation for others to find, but there is nothing else the server side (`usbipd-win`) can do here.

J0F3, 2021-11-18:
Yes, I agree nothing which `usbipd-win` fix.-> closing

closed 2021-11-18 by J0F3

brianddk, 2022-01-12:
> but restarting and reloading the service makes the rules take effect.

@benmcmorran how is that done exactly?

I've tried `sudo service udev restart` as well as `sudo udevadm control --reload-rules`, neither seem to run the rule.  Only way I've gotten the rule to run is with `sudo udevadm test ...` where I name the devpath.