dorssel/usbipd-win #98: Facing issue to bind usb port to wsl2

sudip-mondal-2002, 2021-11-15:
I read this blog to connect my USB device to wsl2 but got error
https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/
[image]

dorssel, 2021-11-15:
Please provide the log of the server following: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

dorssel, 2021-11-16:
The error
```
usbip: error: tcp connect
```
indicates that you may have a firewall issue. Are you running a third party firewall? If so, you need to open TCP port 3240 on your host. `usbipd-win` installs a Windows firewall rule for that, but not all third party firewalls honor those Windows rules.

sudip-mondal-2002, 2021-11-17:
Okay, I turned off the firewall,
Now it shows a different error
[image]

sudip-mondal-2002, 2021-11-17:
the error is only coming for the integrated webcam

sudip-mondal-2002, 2021-11-17:
```pwsh
PS C:\Windows\System32> sc stop usbipd

SERVICE_NAME: usbipd
        TYPE               : 10  WIN32_OWN_PROCESS
        STATE              : 3  STOP_PENDING
                                (STOPPABLE, NOT_PAUSABLE, ACCEPTS_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0
PS C:\Windows\System32> sc stop usbipd
[SC] ControlService FAILED 1062:

The service has not been started.

PS C:\Windows\System32> sc start usbipd

SERVICE_NAME: usbipd
        TYPE               : 10  WIN32_OWN_PROCESS
        STATE              : 2  START_PENDING
                                (NOT_STOPPABLE, NOT_PAUSABLE, IGNORES_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x7d0
        PID                : 6716
        FLAGS              :
PS C:\Windows\System32> usbipd server Logging:LogLevel:Default=Trace
usbipd: error: Another instance is already running.
```

dorssel, 2021-11-17:
That's the wrong way around.
1) Stop service
2) Then run `usbipd server Logging:LogLevel:Default=Trace`
3) Finally, after getting the log, start the service again

sudip-mondal-2002, 2021-11-17:
```pwsh
PS C:\Windows\System32> sc stop usbipd

SERVICE_NAME: usbipd
        TYPE               : 10  WIN32_OWN_PROCESS
        STATE              : 3  STOP_PENDING
                                (STOPPABLE, NOT_PAUSABLE, ACCEPTS_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0
PS C:\Windows\System32> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Windows\System32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
dbug: Microsoft.Extensions.Hosting.Internal.Host[3]
      Hosting stopping
dbug: Microsoft.Extensions.Hosting.Internal.Host[4]
      Hosting stopped
PS C:\Windows\System32> sc start usbipd

SERVICE_NAME: usbipd
        TYPE               : 10  WIN32_OWN_PROCESS
        STATE              : 2  START_PENDING
                                (NOT_STOPPABLE, NOT_PAUSABLE, IGNORES_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x7d0
        PID                : 20968
        FLAGS              :
PS C:\Windows\System32>
```

dorssel, 2021-11-17:
Did you try to 'attach' while you were logging? This log is entirely empty. I need the log from the server while you reproduce the error you reported earlier.

sudip-mondal-2002, 2021-11-18:
```pwsh
PS C:\Windows\System32> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Windows\System32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[0]
      new connection from 172.25.219.240
dbug: UsbIpServer.ConnectedClient[0]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client: System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 144
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 155
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 161
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 158
dbug: UsbIpServer.Server[0]
      connection closed
```

sudip-mondal-2002, 2021-11-18:
Somehow it worked automatically after some tries, but now it's showing the webcam in it's list but can't use it in applications
[image]

dorssel, 2021-11-18:
That error has been reported before, see #63.
The fact that attaches sometimes works, and sometimes does not is new information, thx.

Once the device is attached (and shows up in lsusb) it should work, but the standard WSL kernel probably does not have the Linux drivers. You will have to compile the WSL kernel with the right drivers, then your webcam should work normally.
See: https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel

sudip-mondal-2002, 2021-11-18:
I build the kernel, but the earlier error coming again
```pwsh
PS C:\Users\sudip> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\sudip
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[0]
      new connection from 172.20.39.14
dbug: UsbIpServer.ConnectedClient[0]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client: System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 144
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 155
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 161
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 158
dbug: UsbIpServer.Server[0]
      connection closed
```

dorssel, 2021-11-18:
@sudip-mondal-2002 
Please post the output of the following command

```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

You are now facing issue #63. It can help to simply try attach again, or to unplug/replug the device, or to reboot the host. Unfortunately, this issue is tough to resolve.

sudip-mondal-2002, 2021-11-19:
```pwsh
PS C:\Windows\System32> reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters

ERROR: The system was unable to find the specified registry key or value.
```

dorssel, 2021-11-19:
Too bad. Other users had the `System.IO.FileNotFoundException` because of some incompatible software. But that does not to be your case. The only things I can advise:

- Try attach multiple times, other users (as well as yourself!) did get their device to attach after multiple attempts.
- Try unplugging/replugging the device
- Reboot the host

But it seems your USB controller/drivers somehow seem incompatible with `usbipd-win`. You did state:

> Somehow it worked automatically after some tries

before, so hopefully you can get it to work with a little trial and error.

sudip-mondal-2002, 2021-11-20:
Okay thanks, I'll try

dorssel, 2021-12-01:
@sudip-mondal-2002 
I have made a new attempt on a more aggressive device grabber in PR #134. It may solve the `System.IO.FileNotFoundException` problem. Could you try out the build artifact? It is at https://github.com/dorssel/usbipd-win/actions/runs/1527927708.

sudip-mondal-2002, 2021-12-02:
Thank you very much @dorssel. Will try this.

dorssel, 2021-12-03:
@sudip-mondal-2002 
Multiple users are facing this problem. Hence, more changes to try to solve this. Newest artifact at https://github.com/dorssel/usbipd-win/actions/runs/1535044493.

dorssel, 2021-12-06:
@sudip-mondal-2002
The new code has been released as version 1.3.0.

sudip-mondal-2002, 2021-12-07:
Wow, thank you @dorssel

closed 2021-12-07 by sudip-mondal-2002