dorssel/usbipd-win #101: USB3 Vision Camera is not working

Solairseir, 2021-11-16:
Hello,

Thanks for the usbipd-win, it helps a lot with wsl2 , however I have some problem with USB3.

I have successfully used usbipd to connect multiple USB2 boards to my wsl2. However I am facing difficulty with a Basler USB3 Vision camera.

here is windows powershel output:

```
>usbipd list
Present:
BUSID  DEVICE                                                           STATE
1-2    Billboard Device                                               Not shared
1-3    USB Input Device                                             Not shared
1-4    USB Input Device                                             Not shared
1-5    USB Input Device                                             Not shared
1-8    USB Video Device                                             Not shared
1-10   Synaptics UWP WBDI                                      Not shared
1-14   Intel(R) Wireless Bluetooth(R)                           Not shared
1-17   Basler dart USB3 Vision Camera                       Attached
```

wsl output:
```
$ lsusb
Bus 002 Device 002: ID 2676:ba03 Basler AG daA1920-30uc
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
$ sudo usbip list -r 192.168.1.48
Exportable USB devices
======================
 - 192.168.1.48
       1-17: Basler AG : unknown product (2676:ba03)
           : USB\VID_2676&PID_BA03\23309140
           : Miscellaneous Device / ? / Interface Association (ef/02/01)
           :  0 - Miscellaneous Device / USB3 Vision / unknown protocol (ef/05/00)
           :  1 - Miscellaneous Device / USB3 Vision / unknown protocol (ef/05/02)
```

```
$ arv-tool-0.6 -d all:3
Failed to open USB device
Found 0 USB3Vision device (among 3 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.1.5
No device found
```

I already installed the library for Basler Camera, I used the same one on my ubuntu laptop and it is working, I doubt it would be Basler Camera driver.

Am I missing something here? Is there any extra steps to connect USB3 devices? Is USB3 supported at all? :-D

Thank you for your help in advance.

dorssel, 2021-11-16:
Yes, USB3 is supported. It looks like the device is connected to WSL fine, but no drivers are installed for its function. See https://github.com/dorssel/usbipd-win/wiki/WSL-support on how to compile a kernel with additional USB drivers.

Solairseir, 2021-11-16:
Thank you, I will try it and let you know how it goes

dorssel, 2022-02-11:
Closing due to lack of response.

closed 2022-02-11 by dorssel