dorssel/usbipd-win #102: pyocd probe Access denied, WSL2 udev permissions issue

chadcoonsGB, 2021-11-16:
I'm trying to run pyocd with DAPLink on WSL2 (Ubuntu 20.04). I'm able to attach and identify the DAPLink but there appears to be a udev issue. When I try to list the probes I get a permission error:
```$ pyocd list --probes
0000653:WARNING:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0d28 PID=0204). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help.
0000653:WARNING:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0d28 PID=0204). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help.
No available debug probes are connected
```

I've added the 50-cmsis-dap.rules to /etc/udev/rules.d as specified in pyocd https://github.com/pyocd/pyOCD/tree/main/udev .
After adding the rules, I restarted WSL by running `wsl --shutdown` from command prompt but the issue persists. I am able to see the probe using sudo.

Trace Logs:
```C:\>usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\chadcoons
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[0]
      new connection from 172.19.158.27
dbug: UsbIpServer.ConnectedClient[0]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.19.158.27 claimed device at 2-1 (USB\VID_0D28&PID_0204\000000800E7F79D100000000000000000000000097969902).
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=81, flags=512, length=64, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 256 0 64
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=81, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 18, requested: 64
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=82, flags=512, length=18, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 256 0 18
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=82, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 18, requested: 18
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=83, flags=512, length=5, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 3840 0 5
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=83, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 5, requested: 5
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=84, flags=512, length=57, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 3840 0 57
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=84, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 57, requested: 57
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=85, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 1536 0 10
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=85, pending count = 1
dbug: UsbIpServer.AttachedClient[0]
      USBSUP_XFER_STALL -> EPIPE -> 0
trce: UsbIpServer.AttachedClient[0]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=86, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 1536 0 10
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=86, pending count = 1
dbug: UsbIpServer.AttachedClient[0]
      USBSUP_XFER_STALL -> EPIPE -> 0
trce: UsbIpServer.AttachedClient[0]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=87, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 1536 0 10
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=87, pending count = 1
dbug: UsbIpServer.AttachedClient[0]
      USBSUP_XFER_STALL -> EPIPE -> 0
trce: UsbIpServer.AttachedClient[0]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=88, flags=512, length=9, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 512 0 9
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=88, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 9, requested: 9
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=89, flags=512, length=125, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 512 0 125
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=89, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 125, requested: 125
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=90, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 768 0 255
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=90, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 4, requested: 255
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=91, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 770 1033 255
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=91, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 36, requested: 255
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=92, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 769 1033 255
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=92, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 8, requested: 255
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=93, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 771 1033 255
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=93, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 98, requested: 255
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=94, flags=0, length=0, ep=0
dbug: UsbIpServer.AttachedClient[0]
      Trapped SET_CONFIGURATION: 1
trce: UsbIpServer.AttachedClient[0]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=95, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 772 1033 255
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=95, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 34, requested: 255
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=96, flags=0, length=7, ep=0
trce: UsbIpServer.AttachedClient[0]
      33 32 0 0 7
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=96, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 7, requested: 7
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=97, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 773 1033 255
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=97, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 34, requested: 255
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=98, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 774 1033 255
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=98, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 26, requested: 255
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=99, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 775 1033 255
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=99, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 36, requested: 255
trce: UsbIpServer.AttachedClient[0]
      USBIP_CMD_SUBMIT, seqnum=100, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[0]
      128 6 777 1033 255
trce: UsbIpServer.AttachedClient[0]
      Scheduled seqnum=100, pending count = 1
trce: UsbIpServer.AttachedClient[0]
      actual: 26, requested: 255
info: UsbIpServer.ConnectedClient[2]
      Client 172.19.158.27 released device at 2-1 (USB\VID_0D28&PID_0204\000000800E7F79D100000000000000000000000097969902). 
```

At this point I run `pyocd list --probes` on WSL2 terminal, no new trace logs write out

dorssel, 2021-11-16:
You're in luck!
Some other people have already reported this. Root cause is:
```
USBSUP_XFER_STALL -> EPIPE -> 0
```
This has been fixed in 1.2.0, which was released 1 hour ago! Note that it is not yet on winget, so you will have to install from https://github.com/dorssel/usbipd-win/releases/tag/v1.2.0.

chadcoonsGB, 2021-11-17:
Glad to hear I'm not the only one!

Unfortunately, updating to 1.2.0 didn't appear to fix the issue.
```
usbipd -v
usbipd-win
1.2.0+51.Branch.master.Sha.a1d6f0c6ff27baae9aab9f279a61d196dad3d0b9
```
The trace logs are slightly different
```
usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\chadcoons
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.21.24.233
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.21.24.233 claimed device at 2-1 (USB\VID_0D28&PID_0204\000000800E7F79D100000000000000000000000097969902).
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=21, flags=512, length=64, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 64
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=21, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 64
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=22, flags=512, length=18, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 18
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=22, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 18
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=23, flags=512, length=5, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 3840 0 5
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=23, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 5, requested: 5
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=24, flags=512, length=57, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 3840 0 57
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=24, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 57, requested: 57
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=25, flags=512, length=10, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 1536 0 10
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=25, pending count = 1
dbug: UsbIpServer.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 10
trce: UsbIpServer.AttachedClient[1001]
[... 97 lines cut ...]
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=38, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 774 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=38, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 26, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=39, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 775 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=39, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 36, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=40, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 777 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=40, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 26, requested: 255
```

dorssel, 2021-11-17:
That actually looks OK. Nothing weird on the server side. What is the log on the client side like (/var/log/syslog)?

chadcoonsGB, 2021-11-19:
Sorry for the delay! Here are the client dmesg logs when I run `usbipd wsl attach` on the server.
```
Nov 19 13:53:24 LAPTOP-839K6T2O kernel: [   59.150318] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
Nov 19 13:53:24 LAPTOP-839K6T2O kernel: [   59.150320] vhci_hcd vhci_hcd.0: devid(327681) speed(2) speed_str(full-speed)
Nov 19 13:53:24 LAPTOP-839K6T2O kernel: [   59.150349] vhci_hcd vhci_hcd.0: Device attached
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.432405] vhci_hcd: vhci_device speed not set
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.502483] usb 1-1: new full-speed USB device number 2 using vhci_hcd
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.582470] vhci_hcd: vhci_device speed not set
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.652409] usb 1-1: SetAddress Request (2) to port 0
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.690613] usb 1-1: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.690616] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.690618] usb 1-1: Product: DAPLink CMSIS-DAP
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.690619] usb 1-1: Manufacturer: Arm
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.690620] usb 1-1: SerialNumber: 000000800e7f79d100000000000000000000000097969902
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.692961] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.694834] usbhid 1-1:1.2: couldn't find an input interrupt endpoint
```

When I run `pyocd list --probes` on the client, no new logs appear.

dorssel, 2021-11-19:
That's weird. Your earlier log contains:
```
dbug: UsbIpServer.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 1
```
This is when the device is "turned on" and that call was successful. Is your device (DAPLink CMSIS-DAP) one of those that behaves differently for a few seconds (for example to flash it) and then comes back in some other mode?

stephendpmurphy, 2021-11-22:
Hey @dorssel I work with Chad and am helping him on the backend. Is this SET_CONFIGURATION referring to a USB Descriptor/Configuration on the device? I'm not familiar with how the USB scheme works in DAPLink, but the programmer we are using specifically does not have a bootloader in the Firmware (i.e. No alternate Configuration). I think the best person to help out with this is @mbrossard.

dorssel, 2021-11-22:
@stephendpmurphy 
@chadcoonsGB 
When the device has just been attached, it is in "configuration 0", which is not really a configuration but really a low-power mode. When Linux has identified the device and started a driver for it, that driver will put the device in functional mode by SET_CONFIGURATION. Most devices really have only one configuration (although there could be others). When I see SET_CONFIGURATION in the logs, it tells me that Linux was able to get all the device descriptors (that's already at least 5 URBs handled correctly, as is the case with your logs) and was able to determine the device type, load a driver, etc. It is usually the sign that all is well.

What doesn't look well on your log, is that all URBs go to the control pipe (ep = 0). Most USB devices expose one or more interfaces with actual endpoints (ep > 0).

I also see:

>Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.692961] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
>Nov 19 13:53:25 LAPTOP-839K6T2O kernel: [   59.694834] usbhid 1-1:1.2: couldn't find an input interrupt endpoint

That seems to indicate that the USB device has 2 interfaces. one of which of type HID (usually keyboard emulation). And that the Linux driver complains that it cannot find the endpoint. But I know of some devices that have a "vendor" interface that claims to be HID (perhaps for firmware updates). In that case it makes sense that a true HID driver cannot interact with it.

Anyway, the cdc_acm device seems to be what you are after. Are you sure all the drivers for it are loaded (lsmod / modprobe)?

If you can, please run `usbview.exe` (from the Windows SDK) on the Windows host and post the device descriptors.

mbrossard, 2021-11-22:
The device in question runs the [DAPLink](https://github.com/ARMmbed/DAPLink) Open Source firmware and would typically have multiple interfaces: MSC (not enabled in this case), HID (CMSIS-DAPv2), CDC ACM, WinUSB/WebUSB and Bulk (CMSIS-DAPv2).

chadcoonsGB, 2021-11-23:
@dorssel Here is the device descriptor from usbview.exe
```
          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0210
bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
bMaxPacketSize0:                   0x08 = (8) Bytes
idVendor:                        0x0D28 = ARM Ltd
idProduct:                       0x0204
bcdDevice:                       0x0100
iManufacturer:                     0x01
     English (United States)  "Arm"
iProduct:                          0x02
     English (United States)  "DAPLink CMSIS-DAP"
iSerialNumber:                     0x03
     English (United States)  "000000800e7f79d100000000000000000000000097969902"
bNumConfigurations:                0x01
```

chadcoonsGB, 2021-11-23:
Just in case, here is the entire entry descriptors
```
[Port2]  :  USB Composite Device

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            18
Companion Hub Symbolic Link Name: USB#ROOT_HUB30#4&20bae766&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "DAPLink CMSIS-DAP"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x36
Open Pipes:                           5

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0210
bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
bMaxPacketSize0:                   0x08 = (8) Bytes
idVendor:                        0x0D28 = ARM Ltd
idProduct:                       0x0204
bcdDevice:                       0x0100
iManufacturer:                     0x01
     English (United States)  "Arm"
iProduct:                          0x02
     English (United States)  "DAPLink CMSIS-DAP"
iSerialNumber:                     0x03
     English (United States)  "000000800e7f79d100000000000000000000000097969902"
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x83  -> Direction: IN - EndpointID: 3
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0010 = 0x10 bytes
bInterval:                         0x20

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
[... 167 lines cut ...]
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>BOS Descriptor<===
bLength:                           0x05
bDescriptorType:                   0x0F
wTotalLength:                      0x0039
bNumDeviceCaps:                    0x02

          ===>Platform Capability Descriptor<===
bLength:                           0x18
bDescriptorType:                   0x10
bDevCapabilityType:                0x05
bReserved:                         0x00
Platform Capability UUID:          3408B638-09A9-47A0-8BFD-A0768815B665
00 01 21 00 

          ===>Platform Capability Descriptor<===
bLength:                           0x1C
bDescriptorType:                   0x10
bDevCapabilityType:                0x05
bReserved:                         0x00
Platform Capability UUID:          D8DD60DF-4589-4CC7-9CD2-659D9E648A9F
00 00 03 06 4A 01 20 00 
```

dorssel, 2021-11-23:
The device appears to have 5 interfaces:

- Interface 0+1 are combined (so called IAD) "mbed Serial Port" and map to:

  > cdc_acm 1-1:1.0: ttyACM0: USB ACM device

  This device reports no errors, although the `usbipd-win` log shows that nothing is trying to access it.

- Interface 2 is what I suspected: a proprietary interface "CMSIS-DAP v1" that claims to be HID, but really is not. The weird thing is that this interface does not have any associated endpoint descriptors, so I cannot see how it is to be used. But because it claims to be HID class, Linux tries to load a HID driver for it, which of course complains that it cannot find any endpoints for it:

  > usbhid 1-1:1.2: couldn't find an input interrupt endpoint

- Interface 3 is "WebUSB: CMSIS-DAP", again an interface without any endpoints.

- Interface 4 is "CMSIS-DAP v2", which has endpoints, but isn't claimed by Linux. Probably because the v2 driver isn't available.

All logs look normal and contain no errors. Try using the ACM interface (which seems to be available) from your software, or else try to load/enable a driver for the v2 interface (which seems to be missing on the Linux end).

stephendpmurphy, 2021-11-23:
I actually think the HID interface error where it can't find an endpoint is  "normal". We always get that message. Below is a log capture of the device being plugged into a normal Ubuntu 20.04
``` console
[  532.305894] usb 3-8.2.2.4.1: new full-speed USB device number 15 using xhci_hcd
[  532.390040] usb 3-8.2.2.4.1: device descriptor read/64, error -32
[  532.577927] usb 3-8.2.2.4.1: device descriptor read/64, error -32
[  532.769989] usb 3-8.2.2.4.1: new full-speed USB device number 16 using xhci_hcd
[  532.857974] usb 3-8.2.2.4.1: device descriptor read/64, error -32
[  533.076999] usb 3-8.2.2.4.1: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
[  533.077003] usb 3-8.2.2.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  533.077006] usb 3-8.2.2.4.1: Product: DAPLink CMSIS-DAP
[  533.077007] usb 3-8.2.2.4.1: Manufacturer: Arm
[  533.077009] usb 3-8.2.2.4.1: SerialNumber: 00000080066bff535654868367112151a5a5a5a597969908
[  533.092917] usbhid 3-8.2.2.4.1:1.2: couldn't find an input interrupt endpoint
[  533.705060] cdc_acm 3-8.2.2.4.1:1.0: ttyACM0: USB ACM device
[  533.705766] usbcore: registered new interface driver cdc_acm
[  533.705768] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

@chadcoonsGB is going to verify our serial port is working via the CDC interface.

chadcoonsGB, 2021-11-23:
I was able to confirm that the serial port is working.

stephendpmurphy, 2021-11-23:
@dorssel also, just to recap here. 
- The device enumerates as we expect from our perspective and you see no issues in the log from yours
- The appropriate udev rules have been put in place and the WSL instance has been restarted to ensure those rules are applied
- When we use ```sudo``` before our pyocd command to access the probe. It works. We can see the device. If you do not use sudo, the device can't be found.

dorssel, 2021-11-23:
The udev problems have come up for other users as well, there are several in the current issue list. You may want to look through them, as some have found nice ways to deal with them. However ... all these issues really are WSL related and (as you guessed) are nothing that `usbipd-win` can help with. As soon as the device is detected by the Linux kernel, and identified, and drivers have loaded, then `usbipd-win` is working correctly.

I did notice that, although I started `usbipd-win` for Hyper-V, most users use it in combination with WSL. I have no problem that their WSL issues are resolved here. And I may learn a thing or two from that, so that's good...

dorssel, 2022-02-11:
WSL documentation has been updated with udev details. Closing.

closed 2022-02-11 by dorssel

0Grit, 2022-04-14:
@dorssel I am not finding where the WSL docs address udev beyond mentioning that udev could be an issue. Do you have a link to where they provide details?

dorssel, 2022-04-14:
@0Grit 
https://github.com/dorssel/usbipd-win/wiki/WSL-support#udev