dorssel/usbipd-win #103: Unable to attach an STM Bootloader using 1.2.0

shanselman, 2021-11-18:
I upgraded to usbipd 1.2 (Thanks!) and I'm seeing this:

```
usbipd wsl list
BUSID  DEVICE                                                        STATE
1-13   Realtek Bluetooth Adapter                                     Not attached
1-14   USB Input Device                                              Not attached
1-20   USB Audio Device, USB Input Device, USB Video Device          Not attached
9-2    Logitech BRIO, USB Audio Device                               Not attached
9-3    STM32  BOOTLOADER                                             Not attached
10-2   American Power Conversion USB UPS                             Not attached
11-3   USB Mass Storage Device                                       Not attached
13-2   USB Input Device                                              Not attached
13-3   USB Input Device                                              Not attached
15-1   Logitech USB Input Device, Logitech Unifying USB receiver...  Not attached
15-2   USB Audio Device, USB Input Device                            Not attached
15-3   USB Input Device                                              Not attached
16-4   USB Input Device                                              Not attached
>usbipd wsl attach --busid 9-3
[sudo] password for scott:
usbip: error: Attach Request for 9-3 failed - Request Failed
usbipd: error: Failed to attach device with ID '9-3'.
```
Logs are

```
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\github\usbipd-win\UsbIpServer\bin\x64\Debug
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.19.29.168
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 144
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 155
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 161
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 158
dbug: UsbIpServer.Server[1000]
      connection closed
```

dorssel, 2021-11-18:
`System.IO.FileNotFoundException` is a tough one, it seems. See #63.
Root cause is not known yet, but we have had some luck with:

- trying again (and again); for some users this error is intermittent
- remove other USB capture software, such as USBPCap, UsbDK, etc.
- incompatible USB Root device and/or drivers

The code for this part of `usbipd-win` has not changed between 1.1.1 and 1.2.0. Are you using a different computer than before? I made a preliminary *very aggressive* grabber in PR #93. You could try that instead. Note that you will have to uninstall `usbipd-win` first (the PR reports as 1.1.2).

dorssel, 2021-11-18:
I just updated one of my computers from 1.1.1 to 1.2.0. I could not attach any devices until I rebooted the host. It's strange that MSI did not request a reboot, though. So the good old "have you tried turning it off and on again" applies...

dorssel, 2021-11-18:
I think I have found a root cause for `System.IO.FileNotFoundException` #63.
Please run
```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

If it shows any UpperFilters, those may be incompatible with `usbipd-win`. Remove the conflicting software, or just the registry entry. Then reboot.

shanselman, 2021-11-22:
Ok I tried that, and I don't have ANY UpperFilters. I'm going to run the whole thing in a debugger and let you know more.

dorssel, 2021-11-29:
@shanselman 
I am refactoring SetupAPI to Configuration Manager in PR #120.
The CM API is more capable of restarting and/or resetting devices, making an even more aggressive grabber possible. It's not in #120, but it gives me the tools of adding it later.

dorssel, 2021-12-01:
@shanselman 
I have a new attempt on a more aggressive device grabber in PR #134. The build artifact is at https://github.com/dorssel/usbipd-win/actions/runs/1527927708. Could you give that a try? Thanks.

shanselman, 2021-12-01:
Trying now!

shanselman, 2021-12-02:
Interesting. Initially it's 'friendlier' but I end up in a state where the device disappears from Device Manager (as it should) and VirtualBox USB shows up (I hadn't see that before), but I still get failed to attach.

I will reboot, uninstall, reboot again and clean up, and I'll try with more logging, and if needed, run it under a debugger.

[image]

```
BUSID  DEVICE                                                        STATE
1-13   Realtek Bluetooth Adapter                                     Not attached
1-14   USB Input Device                                              Not attached
1-20   Cam Link 4K, USB Input Device                                 Not attached
8-1    ProFX                                                         Not attached
8-2    American Power Conversion USB UPS                             Not attached
11-3   USB Mass Storage Device                                       Not attached
13-2   USB Input Device                                              Not attached
13-3   USB Input Device                                              Not attached
15-1   Logitech USB Input Device, USB Input Device, Logitech Uni...  Not attached
15-2   GN 2000 USB, USB Input Device                                 Not attached
15-3   USB Input Device                                              Not attached
17-2   Logitech BRIO                                                 Not attached
17-4   STM32  BOOTLOADER                                                 Not attached
19-4   USB Input Device                                                 Not attached
```

dorssel, 2021-12-02:
@shanselman 
The fact that you get the "VirtualBox USB" device identifier is a very important clue!
The "?" at the device is the problem: the device has been "grabbed" by the VBoxMonUsb service, but the VBoxUsb drivers have not loaded. Those drivers are PnP and should have been installed by the `usbipd-win` installer. However, Windows (as you know) is notorious at sometimes requiring a reboot after driver installation.

But the mere fact that you got this far is encouraging! Please try again after a reboot.

shanselman, 2021-12-02:
same after a reboot. Anything I should 'totally uninstall/reinstall?'

[image]

dorssel, 2021-12-02:
@shanselman 
Does the "VirtualBox USB" stay even after failure? That is not supposed to happen either; it should revert to the original device.

But it actually helps us now! Can you try "Update driver" on that node? Should install the VBoxUsb driver automatically. If not, select "Search on my computer" and direct it to the subdirectory of the usbipd-win installation directory.

dorssel, 2021-12-02:
Note to self...

Idea to improve on this programmatically. Instead of looking for the device interface (which only exists if the driver is loaded), look for all USB device with the VBoxUsb hardware ID and fix the driver installation using `UpdateDriverForPlugAndPlayDevicesW `.

dorssel, 2021-12-02:
Note to self...

Also do a disable/enable cycle on the VBoxUsb device on detach/failure, after the VBoxUsbMon filter has been removed, to ensure reverting to the original host device.

shanselman, 2021-12-02:
When I "update driver" it goes BACK to the original bang'ed out version.

[image]

dorssel, 2021-12-02:
@shanselman 
Wait a second. Is this one of those devices that goes into "Bootloader" mode for a while (to inject firmware), and only after a while it "reboots" and then presents itself as a different device? Does the "STM32 BOOTLOADER" become something else after a while?

See #56

shanselman, 2021-12-02:
Yes it is, but in this example I'm FORCING it into Bootloader mode and asking it to STAY there. This worked once before with @craigloewen-msft. We got it to attach ONCE a few weeks ago. I'm going to try to uninstall/reinstall all INFs manually and reboot and see if there's something weird going on.

dorssel, 2021-12-03:
@shanselman
And another attempt: https://github.com/dorssel/usbipd-win/actions/runs/1535044493.
The missing driver for "VirtualBox USB" must be a local installation issue. Maybe remove `usbipd-win` reboot, and then reinstall?

shanselman, 2021-12-04:
That's it. This build worked. :shipit:
[image]
[image]

dorssel, 2021-12-04:
@shanselman 
Great! I also have confirmation for #63 (a very similar issue). I am merging this.
Thanks for all the testing and the useful feedback.

PR #134 (2021-12-04): Add disable/enable during attach
Fixes #63 
Fixes #103
closed 2021-12-04 by dorssel