dorssel/usbipd-win #107: Unable to connect USB to WSL

jramiresbrito, 2021-11-22:
Following the post [https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/](https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/)

I've installed **usbipd-win_1.2.0**.
Successfully listed the devices with PowerShell

```bash
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-5    ADB Device, SM-J500M, USB Serial Device                       Not attached
1-8    Microsoft Hardware USB Keyboard, USB Input Device             Not attached
1-9    Intel(R) Wireless Bluetooth(R)                                Not attached
1-10   USB Input Device, WinUsb Device                               Not attached
1-12   USB Input Device                                              Not attached
1-13   USB Input Device
```

Unfortunatelly, when I try to attach:

```bash
usbipd wsl attach -b 1-5

WARNING: usbip not found for kernel 5.10.74.3-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.10.74.3-microsoft-standard-WSL2
    linux-cloud-tools-5.10.74.3-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2
usbipd: error: Failed to attach device with ID '1-5'.
```

Back to WSL2, I've tried to install the packages, but they are not listed:

```bash
sudo apt install linux-tools-5.10.74.3-microsoft-standard-WSL2

Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-tools-5.10.74.3-microsoft-standard-WSL2
E: Couldn't find any package by glob 'linux-tools-5.10.74.3-microsoft-standard-WSL2'
```

How to proceed?

### Environment
Windows: 21H2 (OS Build: 22000.318)
WSL: WSL Preview (the one from store), Ubuntu 20.04.3 LTS

dorssel, 2021-11-22:
Have a look at https://github.com/dorssel/usbipd-win/wiki/WSL-support.

Depending on your distribution, you will either have to install usbip through the package mentioned there, or you will have to recompile the WSL kernel.

Also have a look at what others did to get their specific distribution to work:
#67 
#99

dorssel, 2021-11-28:
Closing. Already documented on how to resolve.

closed 2021-11-28 by dorssel

plutonium-239, 2022-04-30:
i still get the same message and nothing seems to fix it
its not a matter of distribution because i have ubuntu 20.04 as used everywhere in the tutorial.
but `sudo apt install linux-tools-virtual hwdata` installs `linux-tools-5.4.0-109` `linux-tools-5.4.0-109-generic`
what should i do?

dorssel, 2022-04-30:
@plutonium-239 
Following https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools, you installed `linux-tools-virtual`. But because the version does not exactly match the WSL kernel, the second line is needed to tell Ubuntu that you want this version of `usbip` anyway, even if the version does not match exactly:

> sudo update-alternatives --install /usr/local/bin/usbip usbip /usr/lib/linux-tools/*/usbip 20

After that, `usbip` should work. If it doesn't, can you give the output of `ls /usr/lib/linux-tools/`?

plutonium-239, 2022-05-01:
It just shows one directory: `5.4.0-109-generic`

dorssel, 2022-05-01:
Does the update-alternatives give any errors? What is the output of a plain `usbip` command on WSL?

plutonium-239, 2022-05-01:
The `update-alternatives` runs without any errors.
`usbip` shows the help page of usbip, i.e.
```
usage: usbip [--debug] [--log] [--tcp-port PORT] [version]
             [help] <command> <args>

  attach     Attach a remote USB device
  detach     Detach a remote USB device
  list       List exportable or local USB devices
  bind       Bind device to usbip-host.ko
  unbind     Unbind device from usbip-host.ko
  port       Show imported USB devices
```
However, `usbipd` still doesn't work

dorssel, 2022-05-01:
@plutonium-239 
How many WSL instances do you have? What is the output of (Windows side) `wsl --list`? Are you sure you are using `usbipd wsl attach` on the same instance where you set up usbip? Are you perhaps using docker?

plutonium-239, 2022-05-02:
I only have one wsl instance, `wsl --list` confirms that (just Ubuntu 20.04)
No, I'm not using docker or anything like that, its a plain and fresh wsl install.
I would be able to do `usbipd wsl attach` only when `usbipd` is recognized

dorssel, 2022-05-02:
@plutonium-239 
Ok, let's start from the beginning... Your very first sentence in this conversation:

> i still get the same message and nothing seems to fix it

What exactly is that 'same message'? The OP (from 5 month ago) started with a message from `usbipd wsl attach`. And now you tell me that you cannot even run 'usbipd'. So, that confuses me...

plutonium-239, 2022-05-02:
The "same" message is caused by `usbipd` not being found, even in OP's case (so obviously `usbipd wsl attach` wont work : 
```
 WARNING: usbipd not found for kernel 5.10.102.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.10.102.1-microsoft-standard-WSL2
    linux-cloud-tools-5.10.102.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2
```
However, one difference is that OP's error pertains to `usbip` not being found too, which is not the case for me.

dorssel, 2022-05-02:
@plutonium-239 
Ah, that's very simple (and not the same as the OP). You're supposed to run `usbipd` on the Windows side!

plutonium-239, 2022-05-02:
Oh
well...
Thanks!

staffnetDev, 2024-05-27:
it work 
wsl 2 

sudo apt install linux-tools-virtual hwdata
sudo update-alternatives --install /usr/local/bin/usbip usbip $(command -v ls /usr/lib/linux-tools/*/usbip | tail -n1) 20