dorssel/usbipd-win #110: exception when running usbipd

silabs-JimL, 2021-11-23:
usbipd-win version 1.2.0 which is the latest.
Windows 10 21H1 build 19043

`
PS C:\WINDOWS\system32> usbipd wsl list
Unhandled exception. System.AggregateException: One or more errors occurred. (DEVPKEY_Device_Address (12) does not match DEVPKEY_Device_LocationInfo (3))
 ---> System.NotSupportedException: DEVPKEY_Device_Address (12) does not match DEVPKEY_Device_LocationInfo (3)
   at UsbIpServer.ExportedDevice.GetDevice(SafeDeviceInfoSetHandle deviceInfoSet, SP_DEVINFO_DATA devInfoData, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 158
   at UsbIpServer.ExportedDevice.GetAll(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ExportedDevice.cs:line 239
   at UsbIpServer.Program.<>c.<<Main>b__20_18>d.MoveNext() in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 388
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.<>c__DisplayClass79_0.<OnExecute>b__0()
   at Microsoft.Extensions.CommandLineUtils.CommandLineApplication.Execute(String[] args)
   at UsbIpServer.Program.Main(String[] args) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Program.cs:line 569
`

That used to work on Windows 10 2004, but the error appeared after I upgrade my Windows 10.

silabs-JimL, 2021-11-23:
Seems that only appears when I connected my dock to my laptop. If the dock is not connected, it works well.

dorssel, 2021-11-23:
See issue #82.
You are the second person with this error. We have found the root cause, but some experimentation is needed to get to the right solution. Can you perform a similar analysis on exactly which device(s) are causing the mismatch between Port_# and Address?

See comment:
https://github.com/dorssel/usbipd-win/issues/82#issuecomment-975934794
Please check which (if any) of the 2 resolves your problem; not only listing the devices correctly, but also being able to attach those oddly-behaving ones. Thanks.

dorssel, 2021-11-28:
duplicate of #82

closed 2021-11-28 by dorssel