dorssel/usbipd-win #113: Failed to start RegistryWatcher, again, maybe #80

drazen0, 2021-11-24:
Hi,

This is, potentially, the same issue as #80 Failed to start RegistryWatcher
There, it is explained as registry problem with 32/64 bit ...
But I am using totally fresh Win11 notebook, latest official version of usbipd ...., no hacking with registry, ...

Same story as then: when I try to access usbipd server from WSL2, I receive error:

drazen@MACHINE:/mnt/c/Users/user$ sudo usbip list --remote=172.31.32.1
usbip: error: Exported Device List Request failed - Unknown Op Common Status
usbip: error: failed to get device list from 172.31.32.1

... and on host is the same RegistryWatcher error:

dbug: UsbIpServer.Server[1000]
      new connection from 172.31.44.56
fail: UsbIpServer.RegistryWatcher[4]
      An internal error occurred: Failed to start RegistryWatcher
      System.Management.ManagementException: Not found
         at System.Management.ManagementException.ThrowWithExtendedInfo(ManagementStatus errorCode)
         at System.Management.ManagementEventWatcher.Start()
         at UsbIpServer.RegistryWatcher..ctor(ILogger`1 logger) in D:\a\usbipd-win\usbipd-win\UsbIpServer\RegistryWatcher.cs:line 36
dbug: UsbIpServer.Server[1000]
      connection closed

The same problem is when I try to attach some device from host console:

PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
2-3    Goodix fingerprint                                            Not attached
2-4    USB Mass Storage Device                                       Not attached
2-6    USB Video Device                                              Not attached
4-4    Realtek USB GbE Family Controller                             Not attached
9-5    USB Input Device                                              Not attached
10-2   USB Input Device                                              Not attached
10-3   USB Input Device                                              Not attached
10-5   USB Input Device                                              Not attached
PS C:\WINDOWS\system32>
PS C:\WINDOWS\system32> usbipd wsl attach --busid 2-4
[sudo] password for drazen:
usbip: error: Attach Request for 2-4 failed - Request Completed Successfully

usbipd: error: Failed to attach device with ID '2-4'.

On system I see a lot of WmiPrvSE.exe processes:

[image]

I didn't understand what was the solution: 
a) to try to stop one of these processes, or all of these processes that are 32 bit, ... ?
b) to use your latest version of usbipd that is handling that (I think that I am using such version) ?
c) do something in registry ?
d) something else ?

Thanks in advance for help.

BR
D.

dorssel, 2021-11-24:
In the task manager, go to the "Details" tab.  Right-click on the column headers to enable "User" and "Platform".
Look for a "WMI Provider Host" that is "SYSTEM" and "32-bit".
#80 was resolved by killing that process (I believe, I wasn't able to reproduce the problem)...

I did try to force using the 64-bit provider, and version 1.2.0 has that code, but if your problem is the same I was apparently unsuccessful.

drazen0, 2021-11-25:
Thx.

I kill-ed two 32bit processes (and 6 more 64bit exists), but after that I am not able to start wsl ???

PS C:\User> wsl
The operation timed out because a response was not received from the virtual machine or container.
PS C:\User>

Will try to reboot system ...

BR
D.

drazen0, 2021-11-25:
After reboot:

- after reboot I have 8 64bit versions active and 2 32bit versions of VMI Provider hosts
- start virtual machine (wsl from PowerShell)
- kill these 32 bit version from task manager
- try to usbipd wsl attach --busid x-x
- same error as before ???? (no 32bit version of task in task manager)
- on usbipd list I see that device as Shared, but from virtual machine (lsusb) it is not viewable
- so I made usbipd wsl detach --busid x-x to have correct status on host and on guest machines. After that it is Not shared on host also
- but, starting that command (detach) I fount that new VMI Provider, 32bit version, appear in task manager
- seams that detach command explicitly requested 32bit version ??? - maybe it will help you ...
- then I killed again that 32 bit version
- Try again with attach, and now I succeeded (!?)
- With list on host I see it as attached, and with lsusb (on guest) I see it on list

So, with such "workaround", it is OK, but still seams that initial request to attach is working wrongly.
As you are not able to repeat that, if want, I could test it in place of you (when you will have something to test).
Don't hesitate to ask.

- and one comment more: during my writing of this mail, nobody is working on PC. But now, when I check it, I see again one 32bit process in task manager ??? Maybe it appears with delay, and maybe it is result of your attach request displayed with delay ?

BR
D.

PR #145 (2021-12-04): Refactor RegistryWatcher
- Replace RegistryWatcher
- Remove System.Management
- Re-enable trimmed publishing

Fixes #113
Fixes #80
commit 0acf0efc (2021-12-04): Refactor RegistryWatcher

- Replace RegistryWatcher
- Remove System.Management
- Re-enable trimmed publishing

Fixes #113
Fixes #80
closed 2021-11-26 by drazen0