dorssel/usbipd-win #116: Sharing a device between host and wsl

sjohboh, 2021-11-25:
Hi!
The documentation mention "sharing" of devices, but when I attach a device to WSL2, its no longer available on the host Windows 11.
Is it possible to have a device available on the host as well as shared to WSL?

Regards,
Johan

dorssel, 2021-11-25:
No, a USB device is exclusively available to the guest that claims it. If no guest is claiming the device, it is implicitly claimed by (available to) the host.

In OSI-model terms: the device is handled at level 4 (transport layer). To use a device simultaneously by multiple hosts you would need some application layer protocol that allows that form of sharing.

So it is "sharing", for a certain definition of sharing...

sjohboh, 2021-11-25:
Got it, thanks for the reply!

closed 2021-11-25 by sjohboh