dorssel/usbipd-win #117: SET_CONFIGURATION: 2 fails with ArgumentOutOfRangeException

dorssel, 2021-11-25:
@dorssel  Below is the log when I attach the realtek ethernet adapter on the dock.

```
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-3    USB Input Device                                              Not attached
1-6    Realtek DMFT - IR, USB Video Device, WinUsb Device            Not attached
1-10   Intel(R) Wireless Bluetooth(R)                                Not attached
2-2    Realtek USB GbE Family Controller                             Not attached
3-5    Realtek USB Audio                                             Not attached
7-3    USB Input Device                                              Not attached
8-1    J-Link driver, JLink CDC UART Port                            Not attached
PS C:\WINDOWS\system32> usbipd wsl attach --busid 2-2
[sudo] password for jim:
```

```
PS C:\WINDOWS\system32>
PS C:\WINDOWS\system32> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.22.49.99
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.22.49.99 claimed device at 2-2 (USB\VID_0BDA&PID_8153\000002000000).
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=1, flags=512, length=8, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 8
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=1, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 8, requested: 8
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=2, flags=0, length=0, ep=0
trce: UsbIpServer.AttachedClient[1001]
      0 49 40 0 0
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=2, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=3, flags=512, length=18, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 18
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=3, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 18
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=4, flags=512, length=5, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 3840 0 5
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=4, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 5, requested: 5
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=5, flags=512, length=22, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 3840 0 22
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=5, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 22, requested: 22
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=6, flags=512, length=9, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 512 0 9
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=6, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 9, requested: 9
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=7, flags=512, length=57, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 512 0 57
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=7, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 57, requested: 57
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=8, flags=512, length=9, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 513 0 9
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=8, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 9, requested: 9
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=9, flags=512, length=98, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 513 0 98
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=9, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 98, requested: 98
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=10, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 768 0 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=10, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 4, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=11, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 770 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=11, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 40, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=12, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 769 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=12, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 16, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=13, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 774 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=13, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 26, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=14, flags=0, length=0, ep=0
dbug: UsbIpServer.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 2
info: UsbIpServer.ConnectedClient[2]
      Client 172.22.49.99 released device at 2-2 (USB\VID_0BDA&PID_8153\000002000000).
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values. (Parameter 'configurationValue')
         at UsbIpServer.UsbConfigurationDescriptors.SetConfiguration(Byte configurationValue) in D:\a\usbipd-win\usbipd-win\UsbIpServer\UsbConfigurationDescriptors.cs:line 156
         at UsbIpServer.AttachedClient.HandleSubmitAsync(UsbIpHeaderBasic basic, UsbIpHeaderCmdSubmit submit, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\AttachedClient.cs:line 292
         at UsbIpServer.AttachedClient.RunAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\AttachedClient.cs:line 477
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 184
dbug: UsbIpServer.Server[1000]
      connection closed
```

_Originally posted by @silabs-JimL in https://github.com/dorssel/usbipd-win/issues/82#issuecomment-978843750_

dorssel, 2021-11-25:
@silabs-JimL
Could you run `usbview.exe` (from Windows SDK) and post all configuration descriptors?

silabs-JimL, 2021-11-26:
Here it is.

```
Device Descriptor:
bcdUSB:             0x0300
bDeviceClass:         0x00
bDeviceSubClass:      0x00
bDeviceProtocol:      0x00
bMaxPacketSize0:      0x09 (9)
idVendor:           0x0BDA (Realtek Semiconductor Corp.)
idProduct:          0x8153
bcdDevice:          0x3011
iManufacturer:        0x01
0x0409: "Realtek"
iProduct:             0x02
0x0409: "USB 10/100/1000 LAN"
iSerialNumber:        0x06
0x0409: "000002000000"
bNumConfigurations:   0x02

ConnectionStatus: DeviceConnected
Current Config Value: 0x01
Device Bus Speed:     Unknown
Device Address:       0x04
Open Pipes:              3

Endpoint Descriptor:
bEndpointAddress:     0x81  IN
Transfer Type:        Bulk
wMaxPacketSize:     0x0400 (1024)
bInterval:            0x00

Endpoint Descriptor:
bEndpointAddress:     0x02  OUT
Transfer Type:        Bulk
wMaxPacketSize:     0x0400 (1024)
bInterval:            0x00

Endpoint Descriptor:
bEndpointAddress:     0x83  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0002 (2)
bInterval:            0x08

Configuration Descriptor:
wTotalLength:       0x0039
bNumInterfaces:       0x01
bConfigurationValue:  0x01
iConfiguration:       0x00
bmAttributes:         0xA0 (Bus Powered Remote Wakeup)
MaxPower:             0x20 (64 Ma)

Interface Descriptor:
bInterfaceNumber:     0x00
bAlternateSetting:    0x00
bNumEndpoints:        0x03
bInterfaceClass:      0xFF
bInterfaceSubClass:   0xFF
bInterfaceProtocol:   0x00
iInterface:           0x00

Endpoint Descriptor:
bEndpointAddress:     0x81  IN
Transfer Type:        Bulk
wMaxPacketSize:     0x0400 (1024)
bInterval:            0x00

Unknown Descriptor:
bDescriptorType:      0x30
bLength:              0x06
06 30 03 00 00 00 

Endpoint Descriptor:
bEndpointAddress:     0x02  OUT
Transfer Type:        Bulk
wMaxPacketSize:     0x0400 (1024)
bInterval:            0x00

Unknown Descriptor:
bDescriptorType:      0x30
bLength:              0x06
06 30 03 00 00 00 

Endpoint Descriptor:
bEndpointAddress:     0x83  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0002 (2)
bInterval:            0x08

Unknown Descriptor:
bDescriptorType:      0x30
bLength:              0x06
06 30 00 00 02 00 

```

dorssel, 2021-11-27:
@silabs-JimL 
Thanks a lot for this info. It appears to be yet another problem with Windows handling multiple configurations in a notoriously bad way.

Historical note:
1. At first, usbipd-win crashed on any device with more than one configuration
  We fixed that, so devices with more than one config would just be skipped
2. Next, usbipd-win is able to use devices with multiple configuration, as long as you use configuration 1
  That's where we are now.
3. Future: usbipd-win being able to use any configuration
  That's what we are trying to achieve here.

Your device has multiple configurations (2 to be exact) *and* Linux is trying to use configuration 2. Here is the sequence of setup commands that I decoded from your log:

```
128 6 256 0 8        Get device descriptor
0 49 40 0 0          Set USB_REQUEST_ISOCH_DELAY to 64 ns, USB 3.0
128 6 256 0 18       Get device descriptor (again, different length)
128 6 3840 0 5       Get USB_BOS_DESCRIPTOR_TYPE USB 3.0: 9.6.2 Binary Device Object Store (BOS), Table 9-9. BOS Descriptor (5 bytes)
128 6 3840 0 22      Get USB_BOS_DESCRIPTOR_TYPE (full length 22 this time)
128 6 512 0 9        Get CONFIGURATION descriptor (index 0)
128 6 512 0 57       Get CONFIGURATION descriptor (index 0, full length)
128 6 513 0 9        Get CONFIGURATION descriptor (index 1) <-- This is the second configuration
128 6 513 0 98       Get CONFIGURATION descriptor (index 1, full length)
128 6 768 0 255      Get STRING descriptor (index 0, no language)
128 6 770 1033 255   Get STRING descriptor (index 2, language 1033 en-us)
128 6 769 1033 255   Get STRING descriptor (index 1, language 1033 en-us)
128 6 774 1033 255   Get STRING descriptor (index 6, language 1033 en-us)
Trapped SET_CONFIGURATION: 2
```

So, everything is perfectly in order, except that `usbipd-win` did *not* correctly retrieve the second configuration when the device was claimed. But Linux does, and tries to use it. That's why you get `ArgumentOutOfRangeException`.

I will make a PR that will dump the entire 2nd configuration descriptor, just to see if Windows can get to it at all, and how it should be parsed.

BTW: Windows is so notoriously bad at dealing with more than one configuration, that `usbview.exe` doesn't even bother trying to retrieve it, and they left it out of WinUSB.sys altogether...

dorssel, 2021-11-27:
Owww, that's nasty.
We use https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/usbioctl/ns-usbioctl-_usb_descriptor_request

It says
>On input to the IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION I/O control request, the caller should specify the type of descriptor to retrieve in the high byte of wValue and the descriptor index in the low byte.

But then later it says:
> USB_CONFIGURATION_DESCRIPTOR_TYPE | Instructs the USB stack to return the configuration descriptor and all interface, endpoint, class-specific, and vendor-specific descriptors associated with the current configuration.

Key here is: *with the current configuration*. So, even though you can specify the configuration index, it will always return the information of the current configuration, not of the index you requested!

I will have to think of a way around this...

dorssel, 2021-11-27:
@silabs-JimL 
I think I have a fix. Could you try the artifact at
https://github.com/dorssel/usbipd-win/actions/runs/1511105359

As always with PR builds, you may have to uninstall first. MSI is not signed, but drivers are.

silabs-JimL, 2021-11-29:
@dorssel Thanks a lot for your hard work. Now I can attach this ethernet adapter.
But I didn't test the device in the WSL as the kernel doesn't support this device.

```
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-3    USB Input Device                                              Not attached
1-6    Realtek DMFT - IR, USB Video Device, WinUsb Device            Not attached
1-10   Intel(R) Wireless Bluetooth(R)                                Not attached
2-2    Realtek USB GbE Family Controller                             Not attached
3-5    Realtek USB Audio                                             Not attached
7-3    USB Input Device                                              Not attached
PS C:\WINDOWS\system32> usbipd wsl attach --busid 2-2
[sudo] password for jim:
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-3    USB Input Device                                              Not attached
1-6    Realtek DMFT - IR, USB Video Device, WinUsb Device            Not attached
1-10   Intel(R) Wireless Bluetooth(R)                                Not attached
2-2    Realtek USB GbE Family Controller                             Attached - Ubuntu-20.04
3-5    Realtek USB Audio                                             Not attached
7-3    USB Input Device                                              Not attached
```

```
PS C:\WINDOWS\system32> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.31.23.141
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.31.23.141 claimed device at 2-2 (USB\VID_0BDA&PID_8153\000002000000).
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=16, flags=512, length=8, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 8
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=16, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 8, requested: 8
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=17, flags=0, length=0, ep=0
trce: UsbIpServer.AttachedClient[1001]
      0 49 40 0 0
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=17, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=18, flags=512, length=18, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 256 0 18
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=18, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 18, requested: 18
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=19, flags=512, length=5, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 3840 0 5
[... 66 lines cut ...]
trce: UsbIpServer.AttachedClient[1001]
      actual: 16, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=28, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 774 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=28, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 26, requested: 255
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=29, flags=0, length=0, ep=0
dbug: UsbIpServer.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 2
trce: UsbIpServer.AttachedClient[1001]
      actual: 0, requested: 0
trce: UsbIpServer.AttachedClient[1001]
      USBIP_CMD_SUBMIT, seqnum=30, flags=512, length=255, ep=0
trce: UsbIpServer.AttachedClient[1001]
      128 6 773 1033 255
trce: UsbIpServer.AttachedClient[1001]
      Scheduled seqnum=30, pending count = 1
trce: UsbIpServer.AttachedClient[1001]
      actual: 54, requested: 255
```

dorssel, 2021-11-29:
@silabs-JimL 
Thanks for all the testing. I think we have this "multiple configuration" issue finally solved!

If the device wouldn't work, that would be another issue. This one will be closed once I merge the commit. Got some merge conflicts to solve first, though...

PR #119 (2021-11-29): Fix multiple configurations
Fixes #117
closed 2021-11-29 by dorssel