dorssel/usbipd-win #118: Unable to attach USB Device to WSL2

renegart, 2021-11-27:
Hi,
I've two Windows10 21H2 systems running and on both the same issue.
It's not possible to attach an USB device into WSL2 Distro. I've tested it with Ubuntu 20.04 and Gentoo

I get following error log:
`
usbipd wsl attach --busid 2-4
Password:
usbip: error: Attach Request for 2-4 failed - Request Failed

usbipd: error: Failed to attach device with ID '2-4'.
`

The Server logs looks like following:

`
usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.30.92.220
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 144
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 155
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 161
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 158
dbug: UsbIpServer.Server[1000]
      connection closed
`

dorssel, 2021-11-27:
This may be the same problem as #63 
Could you post the output of:
```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

renegart, 2021-11-28:
@dorssel 

`
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
>>

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}   UpperFilters    REG_MULTI_SZ    TsUsbFlt
`

dorssel, 2021-11-28:
@renegart 
Are you willing to conduct an experiment, just to see if it makes a difference?

- temporarily rename the `UpperFilters` registry key to `UpperFilter_original` (or so)
- reboot
- test if `usbipd attach` works

You can revert the change by naming back to `UpperFilters` and reboot again.
If you want, you can try my 'aggressive grabber' attempt, the artifact is at https://github.com/dorssel/usbipd-win/actions/runs/1464274520. It's a bit older build, so you will probably have to uninstall `usbipd-win` first. But it may make a difference as well.

renegart, 2021-11-30:
@dorssel 
Removing the `UpperFilter` key helped.

Your test build showed no difference.
I get the same error message.

For further analyse: I'm hearing the "Unplug" Sound from Windows.
Error on Service-side:
`
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 148
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 200
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 206
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 133
`

dorssel, 2021-12-03:
@renegart 
I have another test build for you, from PR #134. Could you try the build artifact at https://github.com/dorssel/usbipd-win/actions/runs/1527927708?

Can you tell a bit more about the device you are trying to attach? And the USB hub it is connected to?

dorssel, 2021-12-03:
@renegart
Newest artifact at https://github.com/dorssel/usbipd-win/actions/runs/1535044493.

renegart, 2021-12-06:
Tested with your latest build:

`
PS C:\WINDOWS\system32> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.20.166.114
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 295
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 141
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 151
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 157
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 140
dbug: UsbIpServer.Server[1000]
      connection closed
`

I'm testing with two devices.
* USB-Stick (Kingston Technology Company Inc. : Flash drive (13fe:5500))
* USB-Serial Converter (Future Technology Devices International, Ltd : Bridge(I2C/SPI/UART/FIFO) (0403:6015))

Both are connected on an internal USB-Hub (Intel(R) USB 3.10 eXtensible Host Controller - 1.20 (Microsoft))
The Notebook (Dell Latitude 5520) doesn't have an USB2 Hub

dorssel, 2021-12-06:
@renegart 
Thanks for testing. Was that with the last PR I mentioned, or with the release 1.3.0 which I released earlier today? Because the latter is now the actual latest build...

renegart, 2021-12-06:
@dorssel 
> Newest artifact at https://github.com/dorssel/usbipd-win/actions/runs/1535044493.

This one.

I can also give 1.3.0 form today a try, if you want.

dorssel, 2021-12-06:
@renegart 
Yes please. 1.3.0 is the new baseline.

renegart, 2021-12-09:
@dorssel

I've tested it with version 1.3.0
Same as usual:
`
PS C:\WINDOWS\system32> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.23.49.221
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at UsbIpServer.ConfigurationManager.GetVBoxDevice(BusId busId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 295
         at UsbIpServer.VBoxUsbMon.ClaimDeviceOnce(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 141
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 151
         at UsbIpServer.VBoxUsbMon.ClaimDevice(ExportedDevice device) in D:\a\usbipd-win\usbipd-win\UsbIpServer\VBoxUsbMon.cs:line 157
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 139
dbug: UsbIpServer.Server[1000]
      connection closed
`

dorssel, 2021-12-09:
@renegart 
Your case is intriguing... Most `FileNotFoundException` problems have been solved with 1.3.0. Judging from your hardware (Kingston flash drive and USB-to-serial), those USB devices seem quite normal. I am suspecting an incompatibility with:

> USB-Hub (Intel(R) USB 3.10 eXtensible Host Controller - 1.20 (Microsoft))

It explicitly states USB 3.10. I did test with USB 3.0, but not 3.1. That could be it. Do you have an external USB 2 hub that you can put between the laptop and the devices? Does that make a difference?

renegart, 2021-12-10:
[image]

As you see in the picture, the Host Controller is an USB 3.1 controller. Internally all devices are connected to an USB 3.0 hub. I've put an extra hub between Laptop and the serial converter (and also tested with the USB stick).
You can also see, the hub is really crowded. All internal devices are connected to this hub. Unfortunately the two USB-A connectors too. So it makes no difference, where to put my devices.

Currently I didn't found a way to use the second, empty internal hub. I guess it serves the two USB-C connectors on the other side of the Labtop. I've to find an USB-C hub or docking station. :)

renegart, 2021-12-10:
@dorssel 
[image]

I was able to get an docking station (StarTech TB3DOCK2DPPD). Fun part - It pops up on the 'PCI Express Downstream Switch Port'
Unfortunately it makes no difference. :(

dorssel, 2021-12-12:
@renegart 
Earlier you said:

> Removing the UpperFilter key helped.

Does that mean without UpperFilters and using usbipd-win 1.3.0 you can get this device to work?
If so, we'll just have to conclude that TsUsbFlt and usbipd-win are incompatible...

renegart, 2021-12-13:
> @renegart Earlier you said:
> 
> > Removing the UpperFilter key helped.
> 
> Does that mean without UpperFilters and using usbipd-win 1.3.0 you can get this device to work? If so, we'll just have to conclude that TsUsbFlt and usbipd-win are incompatible...

Yes, it works without this filter driver.

renegart, 2022-01-06:
@dorssel 
Hello,
I've seen you have open an pull request. It there a built available which can I test for you?

dorssel, 2022-01-06:
@renegart 
The PR is in Draft because it isn't complete. In fact, it diverged too much from the current master by now. Once it is actual again I will certainly ask you to test it. Thanks for the offer!

dorssel, 2022-01-17:
@renegart 
The PR is ready. Could you please test? Note that this requires the new WSL instructions, see https://github.com/dorssel/usbipd-win/wiki/WSL-support.
There is a lot more error reporting now. For example, if you have an UpperFilter, it warns you about it. And there is a brand new option "usbipd bind --force" which should work even with UpperFilters present. Also: wsl attach/detach no longer require admin (Windows) nor root (Linux).

renegart, 2022-01-22:
@dorssel 
Hi I've tested the new package. It works when I bind the device with new '--force' parameter manually.
It would be nice if the 'wsl attach' command would add the '--force' parameter automatically if required

Thx for your help

dorssel, 2022-01-23:
@renegart 
I thought about adding the --force option to WSL attach as well, but the "bind" that is included with the *first* attach is for convenience only. If the bind fails, or --force is required, there is really something weird going on. --force is a workaround, not a solution. Hiding that by automagic behavior (that changes the normal sharing semantics) is not a good idea, as users may want to solve the real underlying problem instead. So the best I could think of was an informative error message. Please note that this problem occurs for about 2% of users only (which happens to include you).