dorssel/usbipd-win #121: missing udev rules file

vigilancetech-com, 2021-11-29:
You say:

 As a common example for using embedded devices with openocd copy share/openocd/contrib60-openocd.rules to the /etc/udev/rules.d folder.
 
 but no such file exists in the repo (nor even the "share" folder)

dorssel, 2021-11-29:
@vigilancetech-com 
You opened a new issue quoting un unnamed person, with a quote that is unrelated to `usbipd-win` (it seems it is about Linux/WSL instead), without any further context.

Please elaborate the issue you have with `usbipd-win`.

vigilancetech-com, 2021-11-29:
@dorssel that quote is from the README.md of this repo

vigilancetech-com, 2021-11-29:
@dorssel it is from https://github.com/dorssel/usbipd-win/wiki/WSL-support

About 1/5th the way down:

Note that depending on your application, you may need to configure udev rules to allow non-root users to access the device. Rules to enable a device must be in place before connecting the device. As a common example for using embedded devices with openocd copy share/openocd/contrib60-openocd.rules to the /etc/udev/rules.d folder.

dorssel, 2021-11-29:
Ah, I see.
A quick google gave me:
https://github.com/arduino/OpenOCD/blob/master/contrib/60-openocd.rules

I guess it is an OpenOCD specific thing, maybe even Arduino specific.

Also have a look through the other issues. `udev` pops up every now and then, and some people have made it work for their distro/hardware/tools. I am afraid `usbipd-win` (server-side) cannot help with `udev` (client-side).

vigilancetech-com, 2021-11-29:
@dorssel yes, I'm having a problem with wsl not giving me a drive letter when I hook up my USB external hard drive and I'm suspecting it's a udev rules problem.   udev isn't something I've had to study yet but it looks like I'm going to have to dig in if I can't find this example file

dorssel, 2021-11-29:
@vigilancetech-com 
You could have said you were trying to use a storage device earlier. 

WSL by default has no drivers for USB storage devices. Your problem is more likely that you have to compile a kernel first.

vigilancetech-com, 2021-11-29:
@dorssel oh, thanks!   That's good to know.   That gives me some direction/hope!

eteq, 2022-01-21:
Not sure which distro you're using here, @vigilancetech-com, but I encountered exactly this problem in Ubuntu 20.04 as well as 21.10 in WSL2, and I eventually resolved it by:
1. Compiling a custom kernel with a bunch of filesystems and various other usb drivers enables (just as @dorssel said)
2. running `sudo service udev restart` when wsl starts up (either manually or as part of an automatic script) - for whatever reason, `udev` doesn't start properly in WSL and you have to give it a kick to do so.

dorssel, 2022-02-11:
WSL documentation has been updated with udev details. Closing.

closed 2022-02-11 by dorssel