dorssel/usbipd-win #122: usbipd: error: Server is currently not running.

watson8544, 2021-11-30:
Please run with `usbipd server Logging:LogLevel:Default=Trace` for running usbipd server as windows Administrator mode in cmd.exe

_Originally posted by @dorssel in https://github.com/dorssel/usbipd-win/issues/96#issuecomment-968896352_

watson8544, 2021-11-30:
If you meet the problem "usbipd: error: Server is currently not running", Please run with `usbipd server Logging:LogLevel:Default=Trace`

m4l490n, 2022-01-10:
> If you meet the problem "usbipd: error: Server is currently not running", Please run with `usbipd server Logging:LogLevel:Default=Trace`

Do I have to do this every time? Isn't it supposed to start automatically?

dorssel, 2022-01-11:
@m4l490n 
No, the service should start automatically on Windows boot. If it does not, please check the status in Windows Service manager. It should be set to "Automatic Start". If it is, and the service is not running, then please check the Windows Event Log, it should have an entry with a crash report on what went wrong. Please post it here.

JohnBerg60, 2022-01-12:
Same problem here. Occurs after debugging with a STLINK-V2 device. After starting the ```USBIP Device Host``` in the Service mananger, I am back in business.

```
Log Name:      Application
Source:        usbipd-win
Date:          11-1-2022 18:26:23
Event ID:      3
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      JOHN-LEGION5
Description:
Category: UsbIpServer.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.IOException: Unable to read data from the transport connection: An established connection was aborted by the software in your host machine..
 ---> System.Net.Sockets.SocketException (10053): An established connection was aborted by the software in your host machine.
   --- End of inner exception stack trace ---
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource<System.Int32>.GetResult(Int16 )
   at UsbIpServer.Tools.ReadExactlyAsync(Stream stream, Memory`1 buf, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Tools.cs:line 23
   at UsbIpServer.Interop.UsbIp.ReadUsbIpHeaderAsync(Stream stream, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Interop\UsbIp.cs:line 184
   at UsbIpServer.AttachedClient.RunAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\AttachedClient.cs:line 477
   at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 207
   at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 228

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="usbipd-win" />
    <EventID Qualifiers="0">3</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2022-01-11T17:26:23.8625797Z" />
    <EventRecordID>10710</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>JOHN-LEGION5</Computer>
    <Security />
  </System>
  <EventData>
    <Data>Category: UsbIpServer.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.IOException: Unable to read data from the transport connection: An established connection was aborted by the software in your host machine..
 ---&gt; System.Net.Sockets.SocketException (10053): An established connection was aborted by the software in your host machine.
   --- End of inner exception stack trace ---
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource&lt;System.Int32&gt;.GetResult(Int16 )
   at UsbIpServer.Tools.ReadExactlyAsync(Stream stream, Memory`1 buf, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Tools.cs:line 23
   at UsbIpServer.Interop.UsbIp.ReadUsbIpHeaderAsync(Stream stream, CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\Interop\UsbIp.cs:line 184
   at UsbIpServer.AttachedClient.RunAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\AttachedClient.cs:line 477
   at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 207
   at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 228
</Data>
  </EventData>
</Event>
```

dorssel, 2022-01-12:
@JohnBerg60 
I can reproduce this log when the WSL instance is shut down while a device is attached. It is an error report from `usbipd-win` itself, and (at least for the case I described) is normal. In my test, this does not stop the service itself and is intended behavior.

Can you determine what causes the error in your case? Is it WSL shutdown? Or something else. What I am looking for is a crash report from the Windows Service manager that explains why the service itself stopped.

dorssel, 2022-01-12:
@JohnBerg60 
Ah, I see. Please run the server with verbose debug logging, see https://github.com/dorssel/usbipd-win/wiki/Troubleshooting. That should give you a crash report on the console.

dorssel, 2022-01-17:
@JohnBerg60 
I've just finished a PR that may solve your problem. See #164, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/1705631433. Could you give that a go?

JohnBerg60, 2022-01-21:
I installed the update, and used it for a couple days. So far so good!

Scrappers-glitch, 2022-09-29:
I just started the service manually:

[image]