dorssel/usbipd-win #125: Windows 10 freezes when usbip-win is used with Hyper-V Linux Fedora 34

JohnRoden78, 2021-11-30:
### Discussed in https://github.com/dorssel/usbipd-win/discussions/124

<div type='discussions-op-text'>

<sup>Originally posted by **JohnRoden78** November 30, 2021</sup>
First of all, thank you for such a useful project!  Unfortunately, I've run into a serious problem when trying to use it.  I know that this usage is slightly outside your target (i.e Windows 11 / WSL2) but the information might be useful.

I am attempting to access a USB device (Aeotec Z-Stick) in Linux Fedora 34 running in a virtual machine hosted on Hyper-V running under Windows 10 Pro.  I use usbipd.exe on Windows 10 to bind the USB port and then, on the Fedora 34 VM, use usbip to list and then attach the port.  As soon as the attach command is issued, the Windows 10 machine freezes with a buzzing sound coming from the HDMI audio.  The Fedora 34 VM is also frozen and cannot be accessed via SSH or VNC from another machine.  A forced power off and reboot is required to restart the system.

Strangely, before the recent windows update (Feature Update to Windows 10, 21H2), there was no problem but it did not go away when I reverted the update.

I cannot work out yet if the freeze is due to a crash in the Fedora VM or in Windows 10  but I will try to find another separate Unix machine to test with the usbip protocol.

I would appreciate any insights you could give me.

-- Windows 10 CMD Window (Admin) ---------------------------------------------
`C:\WINDOWS\system32>"\Program Files\usbipd-win\usbipd.exe" -v`
usbipd-win
1.2.0+51.Branch.master.Sha.a1d6f0c6ff27baae9aab9f279a61d196dad3d0b9

`C:\WINDOWS\system32>"\Program Files\usbipd-win\usbipd.exe" list`
Present:
BUSID  DEVICE                                                        STATE
2-1    USB Mass Storage Device                                Shared
2-5    USB Serial Device                                             Shared
2-8    Intel(R) Wireless Bluetooth(R)                          Not shared
2-9    USB Mass Storage Device                                Not shared
4-1    HD Pro Webcam C920, USB Audio Device       Not shared
4-3    USB Input Device                                              Not shared
4-4    USB Input Device                                              Not shared

Persisted:
GUID                                    BUSID  DEVICE
{1c8f598a-1874-47e3-aa6b-a2e3d3e9c191}  1-5    USB Serial Device

-- VM Linux Fedora 34 (root) ----------------------------------------------------------

>$ `usbip version`
usbip (usbip-utils 2.0)

>$ ` usbip list -r 192.168.0.82 `
Exportable USB devices
======================
 - 192.168.0.82
        2-1: Alcor Micro Corp. : Multi Flash Reader (058f:6366)
           : USB\VID_058F&PID_6366\058F63666485
           : (Defined at Interface level) (00/00/00)
           :  0 - Mass Storage / SCSI / Bulk-Only (08/06/50)
        2-5: Sigma Designs, Inc. : Aeotec Z-Stick Gen5 (ZW090) - UZB (0658:0200)
           : USB\VID_0658&PID_0200\5&437BA90&0&5
           : Communications / unknown subclass / unknown protocol (02/00/00)
           :  0 - Communications / Abstract (modem) / AT-commands (v.25ter) (02/02/01)
           :  1 - CDC Data / Unused / unknown protocol (0a/00/00)

>$ ` usbip attach -r 192.168.0.82 -b 2-5`      ####  causes Windows 10 freeze  ####
  
-- Windows 10 System Details:   (from setup > about ) --------------------------

Processor	Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz   3.40 GHz
Installed RAM	16.0 GB (15.9 GB usable)
System type	64-bit operating system, x64-based processor

Edition	Windows 10 Pro
Version	21H1
Installed on	‎15/‎08/‎2020
OS build	19043.1348
Experience	Windows Feature Experience Pack 120.2212.3920.0

</div>

dorssel, 2021-11-30:
@JohnRoden78 
Thanks for the report. This is not at all outside the target. In fact, `usbipd-win` was created initially just for that: Windows 10 + Hyper-V. That's how I use it myself.

A freeze of the host OS is probably driver related. There are 2 drivers involved: the VBoxUsbMon driver that "grabs" the device (by overriding its PID/VID), which in turn causes the VBoxUsb driver to be loaded (as it is registered to handle the new PID/VID).

We have seen problems with VBoxUsbMon when other USB filter software is installed. For example, USBPcap, USBDk, and Windows RemoteFX USB passthrough (enterprise Hyper-V). Can you post the output of:

```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

This will list such software (if any).

JohnRoden78, 2021-11-30:
@dorssel 
The result of the reg command:
-- Windows 10 CMD Window (Admin) ---------------------------------------------
C:\WINDOWS\system32>`reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters`

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}
    UpperFilters    REG_MULTI_SZ    EUsbHubFilter

C:\WINDOWS\system32>

I'm not aware of any USB filters on either OS.

dorssel, 2021-11-30:
See #118.
Maybe EUsbHubFilter is incompatible with the VBox drivers...
A google search tells me it is related to "Flexihub". Is that something that rings a bell?

JohnRoden78, 2021-11-30:
@dorssel 
I've just tracked down a USB driver that should have been deleted and the usbip attach is working again.  Many thanks for your analysis.  I had installed an alternative version of usbip to test something else and it looks as if I didn't remove it properly.

dorssel, 2021-11-30:
@JohnRoden78 
Looks like this issue is resolved? If not, let me know and I will reopen it.

closed 2021-11-30 by dorssel

king-11, 2023-07-29:
@JohnRoden78 can you help me out with the fedora packages that are needed I checked [WSL Support](https://github.com/dorssel/usbipd-win/wiki/WSL-support) doesn't contain anything for fedora

JohnRoden78, 2023-07-30:
@king-11 
All you need on Fedora is the usbip package.  `$ dnf info usbip`  will give you more information.
I no longer use the package as I now use a dedicated Fedora machine with local USB ports.

king-11, 2023-07-31:
Thanks a lot that does work I can maybe add these instructions in README @dorssel