dorssel/usbipd-win #128: Failed to attach

julianx4, 2021-11-30:
Windows 11 Pro
Ubuntu 20.04 WSL2

Trying to attach an intel realsense camera. I get the following error:

`usbipd wsl attach --busid 2-3
[sudo] password for ubuntu:
usbip: error: Attach Request for 2-3 failed - Request Completed Successfully

usbipd: error: Failed to attach device with ID '2-3'.`

I already checked if the "UpperFilters" issue applies to me, but there is no such entry in the registry.
I also tried a different USB ports.
I rebooted a few times as well, but there is no change.

Here is the log:
`usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\julia
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: UsbIpServer.Server[1000]
      new connection from 172.28.173.187
fail: UsbIpServer.RegistryWatcher[4]
      An internal error occurred: Failed to start RegistryWatcher
      System.Management.ManagementException: Not found
         at System.Management.ManagementException.ThrowWithExtendedInfo(ManagementStatus errorCode)
         at System.Management.ManagementEventWatcher.Start()
         at UsbIpServer.RegistryWatcher..ctor(ILogger`1 logger) in D:\a\usbipd-win\usbipd-win\UsbIpServer\RegistryWatcher.cs:line 36
dbug: UsbIpServer.Server[1000]
      connection closed`

dorssel, 2021-11-30:
Your problem is known: `Failed to start RegistryWatcher`
There is a fix underway, but it is not ready for release yet (ETA about a week).
In the mean time, you may have success following the troubleshooting steps in: #113.

julianx4, 2021-11-30:
Unfortunately it didn't work, now I get the following error:

libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver

dorssel, 2021-11-30:
@julianx4 
That problem is Linux side, not `usbipd-win`. Did you update the WSL kernel with `wsl --update`, see https://github.com/dorssel/usbipd-win/wiki/WSL-support? Because it looks like your WSL kernel does not have USBIP support yet...

kuochuwon, 2021-11-30:
> @julianx4 That problem is Linux side, not `usbipd-win`. Did you update the WSL kernel with `wsl --update`, see https://github.com/dorssel/usbipd-win/wiki/WSL-support? Because it looks like your WSL kernel does not have USBIP support yet...

I have the same error message.
The solution that update the WSL works for me, 
others that ever re-compile the WSL2 kernel like me can try this method.

julianx4, 2021-11-30:
I probably made a mistake when following the instruction here: https://github.com/dorssel/usbipd-win/wiki/WSL-support
I guess I forgot the last part to add "alias usbip=/usr/lib/linux-tools/5.4.0-77-generic/usbip" to .bashrc. (noob here :))

Now with the workaround of killing the 32bit process of WmiPrvSE it kind of works.
I have 2 Realsense cameras, a D435 and a T265. The T265 works, with the D435 I get the error message "fail: UsbIpServer.ConnectedClient[3]" same as issue #63
Let's see, but I think this issue can be closed.

closed 2021-11-30 by julianx4

Robert3141, 2021-12-17:
``` bash
PS C:\Users\robert> usbipd wsl list
BUSID  DEVICE
    STATE
2-1    Intel(R) Wireless Bluetooth(R)
    Not attached
2-3    Integrated Webcam
    Not attached
2-8    USB Input Device
    Not attached
2-9    Goodix fingerprint
    Not attached
3-1    USBasp
    Not attached
3-4    Realtek USB FE Family Controller
    Not attached
PS C:\Users\robert> usbipd wsl attach --busid 3-1
[sudo] password for robert:
libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver
usbip: error: query
usbipd: error: Failed to attach device with ID '3-1'.
```
I get the same error trying to connect my AVR microcontroller over usb. I understand it is a software issue but am wondering the cause of the problem (I would much prefer to do a WSL passthrough and handle it in linux then have to do a horrible windows install).

dorssel, 2021-12-18:
@Robert3141 
Your WSL kernel is outdated and does not support USBIP yet (missing vhci_driver). Follow the instructions on https://github.com/dorssel/usbipd-win/wiki/WSL-support (`wsl --update` should be enough).

PS: commenting on a closed issue is not the fastest way to get a reply...