dorssel/usbipd-win #131: no bzImage found after compiling kernel

vigilancetech-com, 2021-11-30:
I followed the instructions in this:

[https://github.com/dorssel/usbipd-win/wiki/WSL-support](url)

and after doing the 

`sudo make -j 8 && sudo make modules_install -j 8 && sudo make install -j 8
` 

there was no error but the command:

`cp arch/x86/boot/bzImage /mnt/c/Users/<user>/usbip-bzImage`

can't find bzImage.   In fact, there's no bzImage anywhere under the repo folder.  

in fact, inside wsl:

`kevin@topsail:/$ find . -mount -type f -name bzImage 2>/dev/null
kevin@topsail:/$ echo $?
1`

Kernel version: 5.10.60.1

vigilancetech-com, 2021-12-02:
when I did the make commands independently it worked.

No idea why.

closed 2021-12-02 by vigilancetech-com

Serverraum, 2024-01-03:
I have the same problem, but found the information, that it's better to use the home directory (~)
https://github.com/microsoft/WSL/issues/8045#issuecomment-1037465681

Had to learn the hard way that the kernel has 2 files, which only differ in upper and lower case.
Windows (sometimes?) doesn't like files, which only differ in upper and lower case, therefore the git will throw away the second needed file. The files are: net/netfilter/xt_TCPMSS.c and net/netfilter/xt_tcpmss.c.