dorssel/usbipd-win #135: usbipd: error: No default WSL distribution exists.

thierryfischer, 2021-12-02:
Hello,

I have an issue similar to #77, `usbipd wsl attach` does not work, I have tested with latest build
 `1.2.1-PullRequest0132.19+Branch.pull-132-merge.Sha.a23268970ddeed852ecf2e2b19774e1138d2c870`.
With **admin rights** I have the following output:
```
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-2    STM32 STLink                                                  Not attached
PS C:\WINDOWS\system32> usbipd wsl attach --busid=1-2
usbipd: error: No default WSL distribution exists.
PS C:\WINDOWS\system32> wsl --list
Access is denied.
```
With **user rights** the status is a little different but does not work either:
```
PS C:\Users\tfischer> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-2    STM32 STLink                                                  Not attached
PS C:\Users\tfischer> usbipd wsl attach --busid=1-2
usbipd: error: Access denied.
PS C:\Users\tfischer> wsl --list
Windows Subsystem for Linux Distributions:
Ubuntu-18.04 (Default)
```

I have tried this workaround:
@tunguskar 
In the mean time, you can use `usbipd-win` without the `wsl` subcommand by using `usbipd bind` and then manually attach from within WSL with `usbip attach`. In fact, `usbipd wsl attach` is really a macro that does exactly that,

_Originally posted by @dorssel in https://github.com/dorssel/usbipd-win/issues/77#issuecomment-978142698_

I can bind the usb device with admin rights:
```
PS C:\WINDOWS\system32> usbipd bind --busid=1-2
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
1-2    STM32 STLink                                                  Not attached
PS C:\WINDOWS\system32> usbipd list
Present:
BUSID  DEVICE                                                        STATE
1-2    STM32 STLink                                                  Shared
```
But on the WSL side the attach command does not work:
```
tfischer@MYCOMPUTER:~$ sudo usbip attach -b 1-2 -r 192.168.116.33
libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver
```

thierryfischer, 2021-12-02:
I think I have found the solution to my issue.
It really came from the lack of vhci-hcd module. Since I did not have the courage to recompile the kernel with this module, I tried installing the latest WSL update from there https://www.catalog.update.microsoft.com/Search.aspx?q=wsl and selected `Windows Subsystem for Linux Update - 5.10.60.1`

After shutting down and restart WSL I now can attach the device properly:
```
tfischer@MYCOMPUTER:~$ usbip attach -d 1-2 -r 192.168.116.33
tfischer@MYCOMPUTER:~$ sudo usbip port
Imported USB devices
====================
Port 00: <Port in Use> at Full Speed(12Mbps)
       STMicroelectronics : ST-LINK/V2 (0483:3748)
       1-1 -> usbip://192.168.116.33:3240/1-2
           -> remote bus/dev 001/002

tfischer@MYCOMPUTER:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0483:3748 STMicroelectronics ST-LINK/V2
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

tfischer@MYCOMPUTER:~$ sudo usbip detach -p 00
usbip: info: Port 0 is now detached!

tfischer@MYCOMPUTER:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

dorssel, 2021-12-02:
Yep, this was a duplicate of #128 
Glad you worked it out.

closed 2021-12-02 by dorssel