dorssel/usbipd-win #136: The specified WSL distribution cannot be reached via the WSL virtual switch

SJ123SJ, 2021-12-02:
I'm running Windows 10 20H2, Docker Desktop 4.3.0 (71786), usbipd-win_1.2.0, WSL2 5.10.60.1
C:\WINDOWS\system32>usbipd  list
Present:
BUSID  DEVICE                                                                                   STATE
1-1    USB Serial Converter                                                                Not shared
1-2    USB Input Device                                                                     Not shared
1-10   Dell ControlVault w/ Fingerprint Touch Sensor, Microsoft ...  Not shared
1-14   Intel(R) Wireless Bluetooth(R)                                                 Not shared
3-4    Realtek USB GbE Family Controller                                          Not shared
5-5    USB Input Device                                                                     Not shared
6-4    Realtek USB Audio                                                                   Not shared
6-5    USB Input Device                                                                     Not shared

C:\WINDOWS\system32>usbipd wsl attach --busid 1-1
**usbipd: error:** The specified WSL distribution cannot be reached via the WSL virtual switch; try restarting the WSL distribution.

Restarting WSL2 doesn't remedy the error.

In WSL2
root@2365b39e6896:/# uname -a
Linux 2365b39e6896 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

root@2365b39e6896:/# lsusb
Bus 002 Device 001: ID 1d6b:0003
Bus 001 Device 001: ID 1d6b:0002

Any clues regarding this virtual switch would be helpful.

dorssel, 2021-12-03:
@SJ123SJ 
Can you post the output of the following commands under Windows:
```
wsl --list --all --verbose
ipconfig
```

And from within WSL:
```
cat /proc/net/fib_trie
```

SJ123SJ, 2021-12-03:
**Windows**
wsl --list --all --verbose
  NAME                   STATE           VERSION
* docker-desktop-data    Running         2
  docker-desktop         Running         2

Windows IP Configuration

Wireless LAN adapter Wi-Fi:

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::c980:3a0f:1116:f7c1%27
   IPv4 Address. . . . . . . . . . . : 192.168.0.227
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : 192.168.0.1

Ethernet adapter Ethernet:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Ethernet adapter Ethernet 3:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Ethernet adapter vEthernet (Default Switch):

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::816c:35c8:a918:6f94%32
   IPv4 Address. . . . . . . . . . . : 172.27.48.1
   Subnet Mask . . . . . . . . . . . : 255.255.240.0
   Default Gateway . . . . . . . . . :

Ethernet adapter Ethernet 6:

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::d908:200f:d7dc:d97f%14
   IPv4 Address. . . . . . . . . . . : 192.168.56.1
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . :

Wireless LAN adapter Local Area Connection* 2:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Wireless LAN adapter Local Area Connection* 4:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Ethernet adapter Ethernet 4:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Ethernet adapter Bluetooth Network Connection:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Ethernet adapter vEthernet (WSL):

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::e147:f777:91ba:92b1%84
   IPv4 Address. . . . . . . . . . . : 172.17.240.1
   Subnet Mask . . . . . . . . . . . : 255.255.240.0
   Default Gateway . . . . . . . . . :

**WSL**
 cat /proc/net/fib_trie
Main:
  +-- 0.0.0.0/0 3 0 5
     |-- 0.0.0.0
        /0 universe UNICAST
     +-- 127.0.0.0/8 2 0 2
        +-- 127.0.0.0/31 1 0 0
           |-- 127.0.0.0
              /32 link BROADCAST
              /8 host LOCAL
           |-- 127.0.0.1
              /32 host LOCAL
        |-- 127.255.255.255
           /32 link BROADCAST
     +-- 172.17.0.0/16 2 0 2
        +-- 172.17.0.0/30 2 0 2
           |-- 172.17.0.0
              /32 link BROADCAST
              /16 link UNICAST
           |-- 172.17.0.3
              /32 host LOCAL
        |-- 172.17.255.255
           /32 link BROADCAST
Local:
  +-- 0.0.0.0/0 3 0 5
     |-- 0.0.0.0
        /0 universe UNICAST
     +-- 127.0.0.0/8 2 0 2
        +-- 127.0.0.0/31 1 0 0
           |-- 127.0.0.0
              /32 link BROADCAST
              /8 host LOCAL
           |-- 127.0.0.1
              /32 host LOCAL
        |-- 127.255.255.255
           /32 link BROADCAST
     +-- 172.17.0.0/16 2 0 2
        +-- 172.17.0.0/30 2 0 2
           |-- 172.17.0.0
              /32 link BROADCAST
              /16 link UNICAST
           |-- 172.17.0.3
              /32 host LOCAL
        |-- 172.17.255.255
           /32 link BROADCAST

SJ123SJ, 2021-12-06:
@dorssel This issue was closed, but v1.3 did not resolve the problem on my system. Are there some new instructions to make t his work?

C:\WINDOWS\system32>usbipd --version
usbipd-win
1.3.0+37.Branch.master.Sha.89a4e59f384d08d953380913735f59821e63fa1f

C:\WINDOWS\system32>usbipd wsl attach --busid 1-1
usbipd: error: The specified WSL distribution cannot be reached via the WSL virtual switch; try restarting the WSL distribution.

dorssel, 2021-12-06:
@SJ123SJ 
Yes, you closed this issue yourself.
Looking back at your logs, there is something wrong:

1) The switch has address 172.17.240.1 with mask 255.255.240.0, in CIDR notation that is 172.17.240.1/28.
2) The WSL instance has address 172.17.0.3.

It looks like the error "The specified WSL distribution cannot be reached via the WSL virtual switch" is correct. Maybe there is a special route to the instance. Can you run the same commands again, and additionally:

```
route print
```

Can you ping the instance at its address from the host (note that the addresses may have changed by now)?

SJ123SJ, 2021-12-06:
C:\WINDOWS\system32>route print
'''
===========================================================================
Interface List
 13...00 05 9a 3c 7a 00 ......Cisco AnyConnect Secure Mobility Client Virtual Miniport Adapter for Windows x64
 27...4c 1d 96 38 93 25 ......Intel(R) Wi-Fi 6 AX200 160MHz
  3...98 e7 43 83 21 5d ......Intel(R) Ethernet Connection (7) I219-LM
 17...98 e7 43 d9 97 07 ......Realtek USB GbE Family Controller
 32...00 15 5d 9e 2a 62 ......Hyper-V Virtual Ethernet Adapter
 14...0a 00 27 00 00 0e ......VirtualBox Host-Only Ethernet Adapter
 11...4c 1d 96 38 93 26 ......Microsoft Wi-Fi Direct Virtual Adapter #3
  5...4e 1d 96 38 93 25 ......Microsoft Wi-Fi Direct Virtual Adapter #4
  7...00 ff 21 ea 5e 0e ......Zscaler Network Adapter 1.0.2.0
 23...4c 1d 96 38 93 29 ......Bluetooth Device (Personal Area Network)
  1...........................Software Loopback Interface 1
 84...00 15 5d 8c b5 89 ......Hyper-V Virtual Ethernet Adapter #2
===========================================================================

IPv4 Route Table
===========================================================================
Active Routes:
Network Destination        Netmask          Gateway       Interface  Metric
          0.0.0.0          0.0.0.0      192.168.0.1    192.168.0.227      2
         10.0.0.0        255.0.0.0       10.14.55.1      10.14.55.15      2
       10.14.55.0    255.255.255.0         On-link       10.14.55.15    257
      10.14.55.15  255.255.255.255         On-link       10.14.55.15    257
     10.14.55.255  255.255.255.255         On-link       10.14.55.15    257
     10.14.128.10  255.255.255.255       10.14.55.1      10.14.55.15      2
    10.192.128.10  255.255.255.255       10.14.55.1      10.14.55.15      2
   12.126.140.174  255.255.255.255      192.168.0.1    192.168.0.227      2
       64.181.0.0      255.255.0.0       10.14.55.1      10.14.55.15      2
        127.0.0.0        255.0.0.0         On-link         127.0.0.1    331
        127.0.0.1  255.255.255.255         On-link         127.0.0.1    331
  127.255.255.255  255.255.255.255         On-link         127.0.0.1    331
      140.87.60.0    255.255.254.0       10.14.55.1      10.14.55.15      2
       172.16.0.0      255.240.0.0       10.14.55.1      10.14.55.15      2
     172.17.240.0    255.255.240.0         On-link      172.17.240.1   5256
     172.17.240.0    255.255.240.0       10.14.55.1      10.14.55.15      2
     172.17.240.1  255.255.255.255         On-link      172.17.240.1   5256
   172.17.255.255  255.255.255.255         On-link      172.17.240.1   5256
      172.27.48.0    255.255.240.0         On-link       172.27.48.1    271
      172.27.48.0    255.255.240.0       10.14.55.1      10.14.55.15      2
      172.27.48.1  255.255.255.255         On-link       172.27.48.1    271
    172.27.63.255  255.255.255.255         On-link       172.27.48.1    271
      192.168.0.0      255.255.0.0       10.14.55.1      10.14.55.15      2
      192.168.0.0    255.255.255.0         On-link     192.168.0.227    257
      192.168.0.0    255.255.255.0       10.14.55.1      10.14.55.15      2
      192.168.0.1  255.255.255.255         On-link     192.168.0.227      2
    192.168.0.227  255.255.255.255         On-link     192.168.0.227    257
    192.168.0.255  255.255.255.255         On-link     192.168.0.227    257
     192.168.56.0    255.255.255.0         On-link      192.168.56.1    281
     192.168.56.0    255.255.255.0       10.14.55.1      10.14.55.15      2
     192.168.56.1  255.255.255.255         On-link      192.168.56.1    281
   192.168.56.255  255.255.255.255         On-link      192.168.56.1    281
        224.0.0.0        240.0.0.0         On-link         127.0.0.1    331
        224.0.0.0        240.0.0.0         On-link      192.168.56.1    281
        224.0.0.0        240.0.0.0         On-link     192.168.0.227    257
        224.0.0.0        240.0.0.0         On-link      172.17.240.1   5256
        224.0.0.0        240.0.0.0         On-link       172.27.48.1    271
        224.0.0.0        240.0.0.0         On-link       10.14.55.15  10000
  255.255.255.255  255.255.255.255         On-link         127.0.0.1    331
  255.255.255.255  255.255.255.255         On-link      192.168.56.1    281
  255.255.255.255  255.255.255.255         On-link     192.168.0.227    257
  255.255.255.255  255.255.255.255         On-link      172.17.240.1   5256
  255.255.255.255  255.255.255.255         On-link       172.27.48.1    271
  255.255.255.255  255.255.255.255         On-link       10.14.55.15  10000
===========================================================================
Persistent Routes:
  None

[... 32 lines cut ...]
'''
C:\WINDOWS\system32>ping 172.17.240.1

Pinging 172.17.240.1 with 32 bytes of data:
Reply from 172.17.240.1: bytes=32 time<1ms TTL=128
Reply from 172.17.240.1: bytes=32 time<1ms TTL=128
Reply from 172.17.240.1: bytes=32 time<1ms TTL=128
Reply from 172.17.240.1: bytes=32 time<1ms TTL=128

Ping statistics for 172.17.240.1:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 0ms, Maximum = 0ms, Average = 0ms

C:\WINDOWS\system32>ping 172.17.0.3

Pinging 172.17.0.3 with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 172.17.0.3:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),
'''

SJ123SJ, 2021-12-06:
Perhaps Docker is blocking the WSL instance from being pinged??

dorssel, 2021-12-06:
Maybe... is that docker on Windows (host side) or do you run docker within WSL?

But in any case, it is really weird that the host WSL switch and the WSL IP-address are not on the same subnet in the first place. As far as I know, you are the only person experiencing this. Have you rebooted since the original issue? I think the WSL switch gets re-initialized every time you start the first WSL instance after reboot (not sure, though).

SJ123SJ, 2021-12-06:
Docker is on Windows (host side).
I have rebooted since the original issue, and the above info is from after the reboot. I haven't rebooted since installing v1.3 of usbipd.  I will try rebooting again to see if anything fixes itself.

christanaka, 2021-12-11:
@SJ123SJ If you install a linux distro with WSL and enable integration with the distro in Docker Desktop you'll be able to connect. You'll also want to run through the [WSL Support Wiki](https://github.com/dorssel/usbipd-win/wiki/WSL-support) to set up your distro.
The `docker-desktop` and `docker-desktop-data` distros are created by docker for it's own infrastructure, container images, configs, etc., so I'm not so sure you can set up usbipd with it.

dorssel, 2022-02-11:
Version 2.0.0 has been released which has better error reporting for network/firewall issues. Closing.

looninho, 2023-02-18:
Hi, I have the same issue as @SJ123SJ but I can ping to the switch and the WSL instance.

umie0128, 2023-07-21:
@dorssel 
window
[image]

docker 
[image]

dzid26, 2023-10-09:
It used to work, but now I have the same problem.  `usbipd error no default wsl distribution cannot be reached via the wsl virtual switch`  The only thing I can think of that has changed between earlier when it worked and now is that I updated Ubuntu (apt upgrade).  

```
wsl --list --all --verbose
  NAME                   STATE           VERSION
* Ubuntu-20.04           Running         2
  Ubuntu-16.04           Stopped         2
  docker-desktop         Stopped         2
  Ubuntu-22.04           Stopped         1
  docker-desktop-data    Stopped         2
```

```
Ethernet adapter vEthernet (WSLCore):

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::7f9d:4e71:dd44:5bfb%16
   IPv4 Address. . . . . . . . . . . : 192.168.144.1
   Subnet Mask . . . . . . . . . . . : 255.255.240.0
   Default Gateway . . . . . . . . . :

Ethernet adapter vEthernet (WSL):

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::a5e2:d1bc:4829:b30e%59
   IPv4 Address. . . . . . . . . . . : 172.17.112.1
   Subnet Mask . . . . . . . . . . . : 255.255.240.0
   Default Gateway . . . . . . . . . :
```
```
cat /proc/net/fib_trie
Main:
  +-- 0.0.0.0/0 3 0 5
     |-- 0.0.0.0
        /0 universe UNICAST
     +-- 127.0.0.0/8 2 0 2
        +-- 127.0.0.0/31 1 0 0
           |-- 127.0.0.0
              /8 host LOCAL
           |-- 127.0.0.1
              /32 host LOCAL
        |-- 127.255.255.255
           /32 link BROADCAST
     +-- 172.17.112.0/20 2 0 1
        |-- 172.17.112.0
           /20 link UNICAST
        |-- 172.17.122.33
           /32 host LOCAL
        |-- 172.17.127.255
           /32 link BROADCAST
Local:
  +-- 0.0.0.0/0 3 0 5
     |-- 0.0.0.0
        /0 universe UNICAST
     +-- 127.0.0.0/8 2 0 2
        +-- 127.0.0.0/31 1 0 0
           |-- 127.0.0.0
              /8 host LOCAL
           |-- 127.0.0.1
              /32 host LOCAL
        |-- 127.255.255.255
           /32 link BROADCAST
     +-- 172.17.112.0/20 2 0 1
        |-- 172.17.112.0
           /20 link UNICAST
        |-- 172.17.122.33
           /32 host LOCAL
        |-- 172.17.127.255
           /32 link BROADCAST
```
```
 ping 172.17.122.33

Pinging 172.17.122.33 with 32 bytes of data:
Reply from 172.17.122.33: bytes=32 time<1ms TTL=64
Reply from 172.17.122.33: bytes=32 time<1ms TTL=64
```
```
 route print
===========================================================================
Interface List
 19...f ec dc ......Microsoft Wi-Fi Direct Virtual Adapter #4
  5.... ec db ......Microsoft Wi-Fi Direct Virtual Adapter #5
 27....ec db ......Intel(R) Wi-Fi 6 AX201 160MHz
 12... ec df ......Bluetooth Device (Personal Area Network) #4
  1...........................Software Loopback Interface 1
 18...00 00 00 00 00 00 00 e0 Microsoft Teredo Tunneling Adapter
 16...00 15 5d 96 3d 55 ......Hyper-V Virtual Ethernet Adapter
 59...00 15 5d 14 eb fa ......Hyper-V Virtual Ethernet Adapter #2
===========================================================================

IPv4 Route Table
===========================================================================
Active Routes:
Network Destination        Netmask          Gateway       Interface  Metric
          0.0.0.0          0.0.0.0      192.168.2.1     192.168.2.70     35
        127.0.0.0        255.0.0.0         On-link         127.0.0.1    331
        127.0.0.1  255.255.255.255         On-link         127.0.0.1    331
  127.255.255.255  255.255.255.255         On-link         127.0.0.1    331
     172.17.112.0    255.255.240.0         On-link      172.17.112.1   5256
     172.17.112.1  255.255.255.255         On-link      172.17.112.1   5256
   172.17.127.255  255.255.255.255         On-link      172.17.112.1   5256
      192.168.2.0    255.255.255.0         On-link      192.168.2.70    291
     192.168.2.70  255.255.255.255         On-link      192.168.2.70    291
    192.168.2.255  255.255.255.255         On-link      192.168.2.70    291
    192.168.144.0    255.255.240.0         On-link     192.168.144.1   5256
    192.168.144.1  255.255.255.255         On-link     192.168.144.1   5256
  192.168.159.255  255.255.255.255         On-link     192.168.144.1   5256
        224.0.0.0        240.0.0.0         On-link         127.0.0.1    331
        224.0.0.0        240.0.0.0         On-link      192.168.2.70    291
        224.0.0.0        240.0.0.0         On-link     192.168.144.1   5256
        224.0.0.0        240.0.0.0         On-link      172.17.112.1   5256
  255.255.255.255  255.255.255.255         On-link         127.0.0.1    331
  255.255.255.255  255.255.255.255         On-link      192.168.2.70    291
  255.255.255.255  255.255.255.255         On-link     192.168.144.1   5256
  255.255.255.255  255.255.255.255         On-link      172.17.112.1   5256
===========================================================================
Persistent Routes:
  None

```

dorssel, 2023-10-09:
The logic was made by Microsoft. It looks for a (single) switch with WSL in the name. You have two, the first having address 192.168.144.1, which is indeed unreachable from Ubuntu (it uses the other one).

How did you get two WSL switches? Is that something docker related?

I think I can just revise the logic to find _any_ switch that has a matching address...

dzid26, 2023-10-09:
> How did you get two WSL switches? Is that something docker related?

I don't know where wslcore came from. Maybe it's Android Subsystem or wslg related. I haven't done much with docker. 
I don't know if I had this switch earlier this week when usbpid-win worked.  I would assume so. Maybe the order of switches has changed after the update.

dorssel, 2023-10-19:
Related (and now confirmed it is from the Android Subsystem), see: https://github.com/dorssel/usbipd-win/issues/588#issuecomment-1770293347

dorssel, 2023-10-20:
@dzid26 
Closing this again, as this (old) issue had a different root cause, please follow #588.

closed 2023-10-20 by dorssel

geosk, 2024-10-09:
https://github.com/dorssel/usbipd-win/issues/588 
winget install usbipd 
solved mine problem with 2 eth