dorssel/usbipd-win #137: usbip: error: import device

nicoborghi, 2021-12-03:
Windows 11 Pro
Ubuntu 20.04.3 LTS
5.10.74.3-microsoft-standard-WSL2
usbipd-win 1.2.0

Trying to attach exFAT-formatted SSD
```
C:\Windows\system32>usbipd wsl list
BUSID  DEVICE                                                        STATE
1-2    USB Attached SCSI (UAS) Mass Storage Device                   Not attached
2-6    USB Video Device                                              Not attached
2-10   Intel(R) Wireless Bluetooth(R)                                Not attached            

C:\Windows\system32>usbipd wsl attach --busid 1-2
[sudo] password for nic:
usbip: error: import device
usbipd: error: Failed to attach device with ID '1-2'. 
connection closed      

```

Log:
```
dbug: UsbIpServer.Server[1000]
      new connection from 172.29.26.159
dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: UsbIpServer.ConnectedClient[1]
      Client 172.29.26.159 claimed device at 1-2 (USB\VID_0781&PID_55AE\MSFT30323133384839343031373632).
info: UsbIpServer.ConnectedClient[2]
      Client 172.29.26.159 released device at 1-2 (USB\VID_0781&PID_55AE\MSFT30323133384839343031373632).
dbug: UsbIpServer.Server[1000]
      connection closed      
```

In WSL I am seeing a folder in /mnt/, but it cannot be accessed (`ls: cannot open directory '.': No such device`). 

Can the exFAT format have a role in this?
The only way to make it work is to `wsl --shutdown` every time and restart it with the SSD attached.

Thank you for your help!

dorssel, 2021-12-03:
@nicoborghi 
Could you post the log of the WSL side (/var/log/syslog)? You may need to install and/or start rsyslog first.

nicoborghi, 2021-12-03:
@dorssel 
`Dec  3 11:15:46 Surface kernel: [72942.208178] vhci_hcd: Failed attach request for unsupported USB speed: super-speed-plus`

dorssel, 2021-12-03:
@nicoborghi 
Ow, that's a new one.
Did you compile your own kernel? See https://github.com/dorssel/usbipd-win/wiki/WSL-support.
If not, I would report this to WSL (https://github.com/microsoft/WSL2-Linux-Kernel) and kindly ask them to include support.

dorssel, 2021-12-03:
@nicoborghi 
Maybe as a workaround, you could connect this to a USB 2.0 port. `usbipd-win` supports USB 3 alright, but the WSL kernel you use obviously does not.

dorssel, 2022-03-12:
Closing due to lack of response.

closed 2022-03-12 by dorssel