dorssel/usbipd-win #139: Yubikey 4 OTP+U2F+CCID (1050:0407) not working after attachment to WSL

rgabdrakhmanov, 2021-12-03:
Steps performed in PS:
```
PS C:\Windows\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
2-1    Microsoft Usbccid Smartcard Reader (WUDF), USB Input Device   Not attached
2-3    Synaptics UWP WBDI                                            Not attached
2-4    Integrated Camera, Realtek DMFT - IR, WinUsb Device           Not attached
2-9    Alcorlink USB Smart Card Reader                               Not attached
2-10   Intel(R) Wireless Bluetooth(R)                                Not attached
3-2    USB Input Device                                              Not attached
3-4    USB Input Device                                              Not attached
PS C:\Windows\system32> usbipd wsl attach --busid 2-1
```
in WSL:
```
❯ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 1050:0407 Yubico.com Yubikey 4 OTP+U2F+CCID
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
❯ sudo usbip port
Imported USB devices
====================
Port 00: <Port in Use> at Full Speed(12Mbps)
       Yubico.com : Yubikey 4 OTP+U2F+CCID (1050:0407)
       1-1 -> usbip://172.24.208.1:3240/2-1
           -> remote bus/dev 002/001
```
Still can't use device to auth in Firefox (running in WSL). Also, device not visible in yubikey-personalization-gui (running in WSL). Also getting empty output in ykman CLI:
```
❯ ykman list

```
I see that "1050:0407 Yubikey 5" is among [tested devices](https://github.com/dorssel/usbipd-win/wiki/Tested-Devices) with same VID/PID as my device. Could you please tell if I am missing something here?

dorssel, 2021-12-03:
Same as discussed in #127.
Problem is with WSL drivers, not usbipd-win.

Seldaek, 2022-07-12:
Just in case it helps anyone..

For me on Windows 11 with latest kernel (`wsl --update`) I only needed to run `sudo service pcscd start` to fix things. That service was needed and without it `ykman list` was outputting: `WARNING: PC/SC not available. Smart card protocols will not function.`

Still can't really do what I wanted to do with ykman tho, but at least the key is visible to it.