dorssel/usbipd-win #148: Error to attach USB device

FloFerrier, 2021-12-05:
When I try to attach my USB device (Nucleo board on Mass storage) on WSL, I have this error :

`PS C:\WINDOWS\system32> usbipd wsl list
BUSID  DEVICE                                                        STATE
2-3    Périphérique d'entrée USB                                     Not attached
2-4    Périphérique d'entrée USB                                     Not attached
2-11   Périphérique vidéo USB                                        Not attached
2-14   Intel(R) Wireless Bluetooth(R)                                Not attached
2-22   Périphérique USB inconnu (échec de réinitialisation de port)  Not attached
3-4    Dispositif de stockage de masse USB                           Not attached
4-3    Dispositif de stockage de masse USB, STMicroelectronics S...  Not attached
PS C:\WINDOWS\system32> usbipd wsl attach -b 4-3
[sudo] password for flfer:
usbip: error: Attach Request for 4-3 failed - Request Completed Successfully

usbipd: error: Failed to attach device with ID '4-3'.`

My PC configuration : 
- 5.10.60.1-microsoft-standard-WSL2
- Windows11 Version 21H2 build 22000.348

dorssel, 2021-12-05:
@FloFerrier 
Before asking you for debug information: could you please test the latest master build first? A lot has changed since the last release, and in fact I am about to issue a new release the coming days. There is a good chance that your issue is actually fixed already.
You can get the installer at https://github.com/dorssel/usbipd-win/actions/runs/1542136092.

FloFerrier, 2021-12-06:
@dorssel Yes the latest release solves my problem, thank you !

closed 2021-12-06 by FloFerrier