dorssel/usbipd-win #150: Unable to attach an iPhone to wsl2

YahiaAngelo, 2021-12-06:
Trying to attach my iPhone to Ubuntu 20.04 with WSL2 but no luck.
Log:
`dbug: UsbIpServer.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      UsbIpServer.ConfigurationManagerException (1003): CM_Disable_DevNode returned CR_REMOVE_VETOED
         at UsbIpServer.ConfigurationManager.ThrowOnError(CONFIGRET configRet, String function) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 42
         at UsbIpServer.ConfigurationManager.TemporarilyDisabledDevice..ctor(UInt32 deviceNode) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 323
         at UsbIpServer.ConfigurationManager.TemporarilyDisabledDevice..ctor(String instanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 316
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 136
dbug: UsbIpServer.Server[1000]
      connection closed`
      
Tested attaching an android phone and it worked, but no luck with my iPhone.

dorssel, 2021-12-06:
@YahiaAngelo 

> CR_REMOVE_VETOED

Some program or service on the host has exclusive access to the device, so it cannot be grabbed. Maybe you need to do "Safely remove hardware" or stop iTune or something... Are you able to attach with version 1.2.0? The way device grabbing works has changed between 1.2.0 and 1.3.0. This could be a regression bug.

YahiaAngelo, 2021-12-06:
You were absolutely right!
I uninstalled all itunes/icloud stuff and it worked right away!
[image]

Cheers.

closed 2021-12-06 by YahiaAngelo

khromov, 2022-01-07:
Had the same issue and for me it was that Android Studio was running in the background! 😅