dorssel/usbipd-win #152: Devices that doesn't support disabling can no longer be attached (CM_Disable_DevNode returned CR_NOT_DISABLEABLE)

yusufkandemir, 2021-12-08:
## Logs

```
$ usbipd wsl attach --busid 6-2
[sudo] password for yusuf:
usbip: error: Attach Request for 6-2 failed - Request Failed

usbipd: error: Failed to attach device with ID '6-2'.
```
```
$ usbipd server
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\Yusuf
fail: UsbIpServer.ConnectedClient[3]
      An exception occurred while communicating with the client:
      UsbIpServer.ConfigurationManagerException (1003): CM_Disable_DevNode returned CR_NOT_DISABLEABLE
         at UsbIpServer.ConfigurationManager.ThrowOnError(CONFIGRET configRet, String function) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 42
         at UsbIpServer.ConfigurationManager.TemporarilyDisabledDevice..ctor(UInt32 deviceNode) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 323
         at UsbIpServer.ConfigurationManager.TemporarilyDisabledDevice..ctor(String instanceId) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConfigurationManager.cs:line 316
         at UsbIpServer.ConnectedClient.HandleRequestImportAsync(CancellationToken cancellationToken) in D:\a\usbipd-win\usbipd-win\UsbIpServer\ConnectedClient.cs:line 136
```

## USB Device Information
| Friendly Name              | Hardware ID                          |
|----------------------------|--------------------------------------|
| Yubico Yubikey 5 NFC       | USB\VID_1050&PID_0407&REV_0543&MI_02 |

## System Information
Microsoft Windows 11 Education 21H2 (Build 22000.348)
Ubuntu 20.04.3 LTS on Windows 10 x86_64 (Kernel 5.10.60.1-microsoft-standard-WSL2)
Windows Subsystem for Linux Preview from Microsoft Store

## Context
It was working fine with v1.2.0, it got broken with v1.3.0. It might be related to #134.

dorssel, 2021-12-08:
@yusufkandemir 
Thanks for the report.
You are right, this is a regression bug. I will try to make a workaround. Would you be willing to test later?

yusufkandemir, 2021-12-08:
@dorssel thanks for confirming, and yes, I am willing to test.

dorssel, 2021-12-08:
@yusufkandemir 
Could you please test the build artifact from PR #153, it is at https://github.com/dorssel/usbipd-win/actions/runs/1554111229.
Thanks!

yusufkandemir, 2021-12-08:
@dorssel thanks for the quick fix, it works!

matrixes, 2021-12-09:
I can also confirm that the artifact in #153 works for attaching my `ID 1050:0407 Yubico.com Yubikey 4 OTP+U2F+CCID` where it had the same issue as originally reported here.

closed 2021-12-10 by dorssel

Creepios, 2022-01-18:
Is this bug planned to be fixed in a major release? and if so, when? @dorssel

dorssel, 2022-01-18:
@Creepios 
It is already fixed in master, so it will be in the next release. Probably in the coming week, but we need more testing on version 2.0.0, because a lot has changed. If you want to help out, try the build from #164, it is at https://github.com/dorssel/usbipd-win/actions/runs/1710391286.

dorssel, 2022-01-20:
@Creepios 
Sometimes (not often!), things go quicker than expected. Version 2.0.0 is out!

Creepios, 2022-01-20:
@dorssel Ok that was faster than I expected :D
I actually wanted to play around with the builds a bit today, but that's probably done now xD

PR #153 (2021-12-10): Refactor attach
- Use restart instead of disable/enable
- Fix unplug detect
- Fix cleanup on startup for non-present devices

Fixes #152 
Fixed #155 
Resolves #91